home-assistant/operating-system #1494: 2021.6.2 vs 2021.8.0

ushroff, 2021-08-05:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [x] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ x] Updated from version 2021.7.4 to 2021.6.2 as provided on 8/2 under Update Pending in Supervisor tab
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

drthanwho, 2021-08-05:
That shouldn't happen. https://version.home-assistant.io/stable.json is correct so try `Reload Supervisor`, refresh your browser and see if it shows 2021.8.1 now. 

Regardless, it's likely related to Supervisor and has nothing to do with the OS itself. So head over to that repo (after closing this issue)

agners, 2021-08-05:
As @drthanwho says, if you can reproduce it even after "Reload Supervisor", please create a new issue with complete information in github.com/home-assistant/supervisor.

closed 2021-08-05 by agners