home-assistant/operating-system #1499: Cannot get the IP on DSM 7.0 with haos_ova-6.2.ova

griffinqiu, 2021-08-06:
[image]

agners, 2021-08-06:
Not familiar with DSM, what is that? What kind of Hypervisor are we talking about? What settings (specifically for network) are you using in your Hypervisor configuration?

Giga-Pudding, 2021-08-17:
I have the same Problem, using ESXi Hypervisor. I am not able to configure IP Settings.
Also, the command `--ipv4-method static` causes an error. Something is broken here.

[image]

Giga-Pudding, 2021-08-17:
> Not familiar with DSM, what is that? What kind of Hypervisor are we talking about? What settings (specifically for network) are you using in your Hypervisor configuration?

When @griffinqiu wrote "DSM 7.0" i think he meant "DiskStation Manager" which is the name of the Operating System of Synology NAS devices. DSM 7.0 is the latest release. It also contains a hypervisor (KVM, as far as i know).

47780, 2021-08-30:
I am running into the exact same issue, I am on 6.2 and when I try to set the IP to static I get the same error. My hypervisor is proxmox.

47780, 2021-08-30:
> I am running into the exact same issue, I am on 6.2 and when I try to set the IP to static I get the same error. My hypervisor is proxmox.

I did get the network working by following the nmcli instructions [here](https://github.com/home-assistant/operating-system/blob/rel-6/Documentation/network.md) to set a static IP. The only difference is that I made the changes to the supervisor config instead of the "Home Assistant OS default" config. Changes to "Home Assistant OS default" didn't work.

agners, 2021-09-03:
@griffinqiu which version of Supervisor have you been using? It seems to work here today:
[image]

47780, 2021-09-04:
KVM, I'm gonna try again and see if it happens again.

closed 2021-12-10 by github-actions[bot]