home-assistant/operating-system #1500: Fresh Install never boots

damusmf, 2021-08-08:
**Hardware Environment**

- [ ] Generic x86-64 (like Intel NUC)
- [ ] Dell Optiplex 5090 Ultra
- [ ]  Intel Core i5- 1145G7
- [ ] 16GB RAM
- [ ] M.2 512GB PCIe NVMe

**Home Assistant OS release:**
- [ ] Fresh installation of release 6.2

- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

Not entirely sure how to provide useful info here.  Had a new-ish unused optiplex 3090 ultra laying around and wanted to test out HA OS.  Booted to a ubuntu live cd and burned the image onto the internal drive using Etcher.  Reboot and it just hangs,  currently sitting at "waiting for root device PARTUUID=SomeLongString"

Attempted reburning multiple times but it never finishes the boot process.  If i plug in/unplug USB devices i do get messages on the screen so the whole thing isn't just locked up.  Can also type into the console but it does nothing.  Attaching pictures of where it gets stuck and then what happens when i plug in and unplug a device.  My guess is something in this computer just is not compatible with HA OS currently and i will have to try a different machine. 

[image]
[image]
[image]

damusmf, 2021-08-08:
Can close this.  In BIOS needed to switch drive type from the default RAID to the AHCI/NVME option.

closed 2021-08-08 by damusmf

agners, 2021-08-10:
Thanks for the update!