home-assistant/operating-system #1501: hass constantly crashes, general instability

zolstarym, 2021-08-09:
**Hardware Environment**

- [x ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x ] Updated from version 21.8.0
- Additional information (if accessible):

logs unavailable due to crash. Hass crashes before I could get any logs

**Description of problem:**

When using hass OS 21.8.4 on raspberry pi 4, it becomes unstable quickly after plugging it in. Unstable on new installs as well.

agners, 2021-08-10:
> [x ] Updated from version 21.8.0

21.8.0 is not the version number of the OS.  Which version of the Operating System image did you install? (You can find the current and older release images here: https://github.com/home-assistant/operating-system/releases).

There is 32-bit and 64-bit version, which version have you been using?

Does the same hardware work stable when using Raspberry Pi OS?

zolstarym, 2021-08-12:
6.2 64 bit
The pi previously ran ubuntu before and seemed stable for a few days, but I never tested for more than that.

lepeli, 2021-08-20:
I had issues before 6.0 where my raspberry pi would crash randomly every day. It didn't really happen that much in 6.0 but after updating to 6.2 I had this issue again. 

I did downgrade again to 6.0 and it fixed this issue. 

if you want to downgrade you can probably use the SSH addon, and then type 
```bash
ha os update --version 6.0 
```
And it should downgrade the OS. I just don't know if it works using the SSH addons tho.

If a maintainer sees this, how can we provide you data when something like this happens to us ?

zolstarym, 2021-11-18:
still happening

closed 2022-02-23 by github-actions[bot]