home-assistant/operating-system #1503: HA OS update not possible in VM

TheDK, 2021-08-10:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [x] OVA (Open Virtualization Appliance, on Oracle VirtualBox on Win2019 x86 PC)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 5.12

**Supervisor logs:**
```
21-08-10 15:45:42 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/6.0/hassos_ova-6.0.raucb
21-08-10 15:45:42 ERROR (MainThread) [supervisor.hassos] Error raised from OTA Webserver: 404
```
**Journal logs:**

**Kernel logs:**

**Description of problem:**
HA OS Update to 6.x is not offered by supervisor. When trying to update via SSH the error above occurs.
Apparently the CLI is looking at the wrong spot, the correct path seems to be:
https://github.com/home-assistant/operating-system/releases/download/6.2/haos_ova-6.2.raucb

I had the same problem with updating to v5.x, gave up and re-installed the VM - but that shouldn't be the norm?

closed 2021-11-22 by github-actions[bot]

agners, 2021-11-28:
If you install a old version first, make sure to update to the latest Supervisor first before updating OS.

TheDK, 2021-11-30:
I now have core on 2021.11.5, supervisor on 2021.10.8 - and Home Assistant OS 5.12

Fun Fact: It was on 5.13 - after updating supervisor to the above it detected 5.12 as most recent version. When running update from the CLI it downgraded to 5.12

TheDK, 2021-11-30:
If I tell it to upgrade to a specific version it cannot find it:

```
21-11-30 15:32:54 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/6.2/hassos_ova-6.2.raucb
21-11-30 15:32:54 ERROR (MainThread) [supervisor.os.manager] Error raised from OTA Webserver: 404
21-11-30 15:33:00 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/6.0/hassos_ova-6.0.raucb
21-11-30 15:33:00 ERROR (MainThread) [supervisor.os.manager] Error raised from OTA Webserver: 404
```

agners, 2021-11-30:
You are using the stable channel?

The URL is derived from the version file from https://version.home-assistant.io/stable.json. The update URL is generated depending on version number: If the requested OS version is equal or higher than 6.0, a different OS name is used, see:
https://github.com/home-assistant/supervisor/blob/2021.10.8/supervisor/os/manager.py#L84-L91

Are you using proxy or some other kind of firewall? Is the stable.json maybe veeeery outdated on your end?

You can try `ha su reload`, to force the Supervisor to fetch the update file.