home-assistant/operating-system #1505: WIFI not supported for NUC6

eriknn, 2021-08-15:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [X ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [X ] Fresh installation of release 6.1
- [X ] Updated from version 6.1
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

eriknn, 2021-08-17:
Actually I see the following:
BR2_PACKAGE_LINUX_FIRMWARE_IWLWIFI_8000=y
in buildroot-external/configs/generic_x86_64_defconfig

I would expect this to be the correct option - so why isn't the driver in the OS then?

agners, 2021-08-17:
The package is actually named `BR2_PACKAGE_LINUX_FIRMWARE_IWLWIFI_8000C`. Just created a PR to fix the name and add some other firmwares.

agners, 2021-09-03:
The firmware will be part of OS release 6.3 onward.

closed 2021-09-03 by agners