home-assistant/operating-system #1507: usb device disconnects after certain time.

malosaa, 2021-08-18:
### The problem

Usb gets disconnected, for me its zigbee2mqtt usb device.

```
[19747.492579] ohci-pci 0000:00:06.0: frame counter not updating; disabled
[19747.493188] ohci-pci 0000:00:06.0: HC died; cleaning up
[19747.554385] usb 2-1: USB disconnect, device number 2
[19749.086739] usb 2-2: USB disconnect, device number 3
```

it did work before last core update, now after a few hours i get an error that the com port is not found in the zigbee2mqtt log...
when i go in to homeassistant hardware tab the usb is not there..

before i started, it showed up:

```
    2.568765] usb 2-2: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
[    2.570915] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.573000] usb 2-2: Product: TI CC2531 USB CDC
[    2.577167] usb 2-2: Manufacturer: Texas Instruments
[    2.581745] usb 2-2: SerialNumber: __0X001XXXXXXXCD44275
```

when i just de-select this usb in virtualbox and i run zigbee2mqtt in shellcommand in windows without disconnecting the usb device it starts up normal... it keeps working for a couple of days... 

### What is version of Home Assistant Core has the issue?

core-2021.8.7

### What was the last working version of Home Assistant Core?

core-2021.8.6

### What type of installation are you running?

Home Assistant OS

agners, 2021-08-19:
This is unlikely to be caused by Home Assistant Core, more likely by hardware or maybe by an OS update. Did you also update the operating system maybe?

What version of the OS are you using? Which hardware is this running on? How is the USB device connected exactly (directly/via USB hub?)

malosaa, 2021-08-19:
> This is unlikely to be caused by Home Assistant Core, more likely by hardware or maybe by an OS update. Did you also update the operating system maybe?
> 
> What version of the OS are you using? Which hardware is this running on? How is the USB device connected exactly (directly/via USB hub?)

hi,

Well its tunning on my server windows 10 using virtualbox version of the os 6.2.ova
I never had an issue before, but after i updated the core version i'm getting this issue.

the usb is on a hub, igo to virtual box and select the usb to import it to homeassistant os.
than it appears and i use it with zigbee2mqtt..
Than after a day it says cannot find usb tty0 no port or directory.. in the logs of zigbee2mqtt..

when i check the homeassistant os hardware log i see the above message... so nothing showing up under usb tty0
but when i go go again to virtualbox and de-select the usb and run it on the windows version of zigbee2mqtt it works fine.. without even disconnecting the usb.

regards

gimi87, 2021-08-22:
I have the same issue on 6.1 and 6.2. I am on raspberry pi 3b.

agners, 2021-08-23:
@malosaa can you check the virtual box log (right click on the Virtual Machine, choose "Show Log...", then upload the complete log file here.

agners, 2021-08-23:
> I have the same issue on 6.1 and 6.2. I am on raspberry pi 3b.

@gimi87 so your problems also started recently with the latest Core update?

gimi87, 2021-08-23:
May be. I did a core and os upgrade at the same time. So it is hard to decide which product broke the USB.
I was looking for a solution here as USB is more hardware/os related than HA core. But... I also have an SSD connected via USB and no problem.

So it could be related to the HA core or even to the Zigbee2Mqtt component.

gimi87, 2021-09-02:
I am on:
```
Home Assistant OS 6.2
supervisor-2021.08.1
core-2021.9.0
```

The issue is still present :( Device died after 24 hrs.

agners, 2021-09-03:
This seems different, as there are always USB bus errors reported before the disconnect:

```
[1626945.681402] ftdi_sio ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[1626945.688348] ftdi_sio ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
```

This errors point to some kind of low level USB problem: Clocks issues, power issues, maybe EMI issues or devices not behaving according to spec.

gimi87, 2021-09-08:
I switched to different hardware. Topic I guess can be closed now. :)

agners, 2021-09-09:
Thanks for the update!

malosaa, 2021-09-16:
I still get that error, Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'

So when i use this usb not on virtualbox but on windows itself it keeps working for weeks.
I'm using last version homeassistant OS on virtualbox.

When i put out usb and insert again it works again for a day,  don't know why

malosaa, 2021-09-16:
@agners 

here is the log,

[hassio-2021-09-15-21-49-54.log](https://github.com/home-assistant/operating-system/files/7179483/hassio-2021-09-15-21-49-54.log)

agners, 2021-09-16:
Hm, maybe its VirtualBox which does that? Are you using the Extension Pack?

Which Home Assistant OS version are you running?

malosaa, 2021-09-17:
@agners 

Home Assistant 2021.9.6
No i'm not using the extensionpack, it did happen again now it has been almost 24h 

this is in the console:

[122311.043700}ohci-pci 000:00:06.0: HC died; cleaning up

malosaa, 2021-09-17:
i will try now firstto install the extensionpack, and select usb2.0 controller and see what happens

malosaa, 2021-09-20:
@agners 

Its running for 2 days now. 
fingers crossed.

i report back in a week

malosaa, 2021-11-18:
@agners 

I'm having this issue now every 24h.i have 2 x USB to n it, one in the PC itself 1 on a hub, they both died at same time and I need to reboot system... 

Is there are work around?
It's really annoying.

malosaa, 2022-01-31:
> Is this still a problem with HAOS 7.1? Can you share the logs with USB 2.0 enabled? What version of VirtualBox are you using?

sorry for the super lat reply,
Well after the last firmware upgrade it works perfect without an issue,

agners, 2022-02-02:
Thanks for the update. With firmware update you mean and update to the TI CC2531 correct?

closed 2022-02-02 by agners

malosaa, 2022-02-02:
@

> Thanks for the update. With firmware update you mean and update to the TI CC2531 correct?

> Thanks for the update. With firmware update you mean and update to the TI CC2531 correct?

yes