home-assistant/operating-system #1508: Home Assistant iOS Companion App does no longer work when two NIC's are connected to system running HA

ronaldvdmeer, 2021-08-18:
**Hardware Environment**
Host system: Proxmox hypervisor
Running method: Debian Buster on a Virtual Machine with USB passthrough (USB full port passthrough)

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [X] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [X] Fresh installation of release 6.2
- [ ] Updated from version x.y
- Additional information (if accessible):

## System Health

version | core-2021.8.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.6
os_name | Linux
os_version | 5.10.53
arch | x86_64
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4963
Installed Version | 1.13.2
Stage | running
Available Repositories | 867
Installed Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.2
-- | --
update_channel | stable
supervisor_version | supervisor-2021.06.8
docker_version | 20.10.6
disk_total | 30.8 GB
disk_used | 4.4 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Visual Studio Code (3.6.1), Terminal & SSH (9.1.3)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 0
views | 4
mode | storage

</details>

**Supervisor logs:**
```
21-08-17 15:53:03 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-08-17 15:59:42 INFO (MainThread) [supervisor.host.info] Updating local host information
21-08-17 15:59:43 INFO (MainThread) [supervisor.host.services] Updating service information
21-08-17 15:59:43 INFO (MainThread) [supervisor.host.network] Updating local network information
21-08-17 15:59:43 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-08-17 15:59:43 INFO (MainThread) [supervisor.host] Host information reload completed
21-08-17 16:23:03 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-08-17 16:52:40 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
21-08-17 16:52:40 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
21-08-17 16:52:40 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
21-08-17 16:52:40 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
21-08-17 16:52:40 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-08-17 16:52:40 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-08-17 16:52:40 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-08-17 16:52:40 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-08-17 16:52:40 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-08-17 16:53:02 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
21-08-17 16:53:02 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
21-08-17 16:53:03 INFO (MainThread) [supervisor.store] Loading add-ons from store: 62 all - 0 new - 0 remove
21-08-17 16:53:03 INFO (MainThread) [supervisor.store] Loading add-ons from store: 62 all - 0 new - 0 remove
21-08-17 16:53:03 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-08-17 17:23:04 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-08-17 17:52:40 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
21-08-17 17:52:40 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
21-08-17 17:52:40 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
21-08-17 17:52:40 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
21-08-17 17:52:40 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-08-17 17:52:40 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-08-17 17:52:40 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-08-17 17:52:40 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-08-17 17:52:40 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-08-17 17:53:02 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-08-17 17:53:04 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-08-17 18:06:23 INFO (MainThread) [supervisor.host.info] Updating local host information
21-08-17 18:06:23 INFO (MainThread) [supervisor.host.services] Updating service information
21-08-17 18:06:23 INFO (MainThread) [supervisor.host.network] Updating local network information
21-08-17 18:06:23 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-08-17 18:06:23 INFO (MainThread) [supervisor.host] Host information reload completed
21-08-17 18:23:04 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-08-17 18:52:40 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
21-08-17 18:52:40 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
21-08-17 18:52:40 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
21-08-17 18:52:40 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
21-08-17 18:52:40 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-08-17 18:52:40 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-08-17 18:52:40 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-08-17 18:52:40 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-08-17 18:52:40 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-08-17 18:53:04 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-08-17 19:02:44 INFO (SyncWorker_4) [supervisor.docker.interface] Restarting ghcr.io/home-assistant/qemux86-64-homeassistant
21-08-17 19:02:51 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
21-08-17 19:02:52 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-08-17 19:02:52 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-08-17 19:03:01 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
```

**Journal logs:**
Sadly I'm unable to get access to SSH on OS level. 
Tried placing authorized_keys file on ext4 formatted USB and get it working but the port 22222 is simple not responding.

**Kernel logs:**

**Description of problem:**
I have a network with multiple VLAN's. My IoT devices are in one VLAN but HA is connected to another. Normally the boundary between VLAN's is controlled by firewall rules. However the API to connect with Xiaomi devices (atleast the Air Purifier) requires me to be within the same subnet/VLAN. To solve this i've connected another NIC to my HA setup which is connected to the IoT VLAN.

Although the website version of HA still works I can no longer use the app on my iPhone. I expect something goes wrong with the local routing but because i'm using HA OS i'm not able to debug it the way I would normally do.

When I disconnect the second NIC the Companion App works again.

ronaldvdmeer, 2021-08-18:
I'm creating this issue here as well because this might be a better place.
Got another ticket open here: https://github.com/home-assistant/core/issues/54756

agners, 2021-08-23:
> Tried placing authorized_keys file on ext4 formatted USB and get it working but the port 22222 is simple not responding.

I think ext4 should work too, but just to be sure, can you try using FAT? Important is that the file system label needs to be `CONFIG` for it to work.

I am not 100% sure how the iPhone App finds/connects to HA, but I think its using mDNS/Zeroconf. Maybe your setup causes Home Assistant to use the wrong network interface or similar. There is the [Network integration](https://www.home-assistant.io/integrations/network/) with configuration under Configuration -> General, maybe setting the correct interface manually helps?

ronaldvdmeer, 2021-08-24:
> I think ext4 should work too, but just to be sure, can you try using FAT? Important is that the file system label needs to be CONFIG for it to work.

Works! I forgot to label it

Just manually configured both interfaces but with that I can no longer connect with app nor can i use the webinterface. When I configure the main interface to be manual and the iot interface to be automatic it will take a couple of minutes before the app goes unresponsive. It feels like when there is a connection with a device on the iot interface the default route changes.

closed 2021-11-29 by github-actions[bot]