home-assistant/operating-system #1518: OS 6.2 RPi4 SSD - console spits crond[473]

drthanwho, 2021-08-29:
**Hardware Environment**

- [x ] Raspberry Pi 4 (SSD boot)

**Home Assistant OS release:**
- [x] Updated from version 6.1 to 6.2

## System Health

version | core-2021.9.0b3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.6
os_name | Linux
os_version | 5.10.17-v8
arch | aarch64
timezone | Europe/Athens

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.14.1
Stage | running
Available Repositories | 892
Installed Repositories | 68

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.2
-- | --
update_channel | beta
supervisor_version | supervisor-2021.08.0
docker_version | 20.10.6
disk_total | 109.3 GB
disk_used | 36.2 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), AdGuard Home (4.1.9), ESPHome (2021.8.2), Glances (0.13.0), Let's Encrypt (4.11.0), Mosquitto broker (6.0.1), Node-RED (10.0.0), Portainer (1.5.2), Spotify Connect (0.9.1), Terminal & SSH (9.1.3), WireGuard (0.5.1), deCONZ (6.10.0), Visual Studio Code (3.6.2), Grafana (7.1.0), Home Assistant Google Drive Backup (0.104.3), MariaDB (2.4.0), Assistant Relay (0.7.4), InfluxDB (4.2.1), VSCode Remote (6541b8e), Network UPS Tools (0.8.0), SSH & Web Terminal (9.0.1), File editor (5.3.3), JupyterLab (0.5.0), chrony (2.2.0), Nginx Proxy Manager (0.11.0), TasmoAdmin (0.15.0), Scrutiny (fd4f0429), Zigbee2mqtt (1.21.0-4)

</details>

<details><summary>Lovelace</summary>

dashboards | 8
-- | --
resources | 46
views | 58
mode | storage
[... 1781 lines cut ...]
[   51.219866] audit: type=1400 audit(1630231841.476:92): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="db21ed7f_scrutiny" pid=1800 comm="apparmor_parser"
[   51.242120] audit: type=1400 audit(1630231841.492:93): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="hassio-supervisor" pid=1803 comm="apparmor_parser"
[   51.242150] audit: type=1400 audit(1630231841.492:93): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="hassio-supervisor///usr/bin/gdbus" pid=1803 comm="apparmor_parser"
[   51.242171] audit: type=1400 audit(1630231841.492:93): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="hassio-supervisor///usr/bin/git" pid=1803 comm="apparmor_parser"
[   51.486343] hassio: port 3(veth1cdc978) entered blocking state
[   51.486359] hassio: port 3(veth1cdc978) entered disabled state
[   51.486953] device veth1cdc978 entered promiscuous mode
[   51.487053] audit: type=1700 audit(1630231841.740:94): dev=veth1cdc978 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[   52.006481] audit: type=1325 audit(1630231842.248:95): table=nat family=2 entries=0 op=xt_register pid=1866 subj==unconfined comm="iptables"
[   52.325765] eth0: renamed from veth9574f12
[   52.355328] IPv6: ADDRCONF(NETDEV_CHANGE): veth1cdc978: link becomes ready
[   52.355460] hassio: port 3(veth1cdc978) entered blocking state
[   52.355469] hassio: port 3(veth1cdc978) entered forwarding state
[   52.859034] hassio: port 4(vethf920ddf) entered blocking state
[   52.859048] hassio: port 4(vethf920ddf) entered disabled state
[   52.859251] device vethf920ddf entered promiscuous mode
[   53.658501] kauditd_printk_skb: 18 callbacks suppressed
[   53.658515] audit: type=1325 audit(1630231843.892:114): table=nat family=2 entries=13 op=xt_replace pid=2206 subj==unconfined comm="iptables"
[   53.690534] audit: type=1325 audit(1630231843.924:115): table=nat family=2 entries=14 op=xt_replace pid=2207 subj==unconfined comm="iptables"
[   53.739926] eth0: renamed from vethf83b865
[   53.768286] IPv6: ADDRCONF(NETDEV_CHANGE): vethf920ddf: link becomes ready
[   53.768511] hassio: port 4(vethf920ddf) entered blocking state
[   53.768528] hassio: port 4(vethf920ddf) entered forwarding state
[   54.138228] hassio: port 5(vethb53d153) entered blocking state
[   54.138243] hassio: port 5(vethb53d153) entered disabl

agners, 2021-09-03:
Hm, not sure where this is coming from, definitely not from OS: There is no `crond` on OS.... This is likely caused by one of the addons.

drthanwho, 2021-09-03:
Damn! It's going to be a pain to work out which one. Likely it doesn't interfere with anything

agners, 2021-09-04:
Since this is most likely an add-on thing I'll close this for now. You can still reopen if it turns out that it needs some fix/change on OS level.

closed 2021-09-04 by agners