home-assistant/operating-system #1524: DHCP fails to renew using KVM - Drops from network

davealger64, 2021-09-02:
**Hardware Environment**
Running KVM under Ubuntu 21.04 (GNU/Linux 5.11.0-25-generic x86_64)
I used the KVM image to install: Linux homeassistant 5.10.53 #1 SMP Wed Jul 28 20:42:41 UTC 2021 x86_64 HAOS

# ha info
arch: amd64
channel: stable
docker: 20.10.6
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- hassos
hassos: "6.2"
homeassistant: 2021.8.8
hostname: homeassistant
logging: info
machine: qemux86-64
operating_system: Home Assistant OS 6.2
state: running
supervisor: 2021.06.8
supported: true
supported_arch:
- amd64
- i386
timezone: America/New_York

top - 11:40:59 up 11 days, 21:45,  0 users,  load average: 0.00, 0.02, 0.00

As reported by other users in the forums, this install of HomeAssistant OS seems to lose its network connection periodically (On mine every 24 hours). I believe this is specifically tied to the DHCP renewal failing to renew its IP address. I do not see dhclient or dnsmasq running (I'm not sure what is initially making the DHCP request and setting the IP Address. Many others have reported "fixing" this by either setting a static ip (which removes the DCHP issue) or "rebooting the host" which in turns reboots the guest. 

However you can see my virtual NIC currently does not have an IP address on it.
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop qlen 1000
    link/ether ba:d0:70:82:16:69 brd ff:ff:ff:ff:ff:ff
3: enp0s3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 52:54:00:27:28:65 brd ff:ff:ff:ff:ff:ff
    inet6 2601:19b:4500:150::6c/128 scope global dynamic noprefixroute
       valid_lft 3245sec preferred_lft 1445sec
    inet6 fe80::4501:7972:f41f:6e3e/64 scope link noprefixroute

If I simply bring the virtual NIC down and back up, the DHCP server recieves a new request and assigns to proper IP Address.
# ip link set dev enp0s3 down
# ip link set dev enp0s3 up
# ip addr show enp0s3
3: enp0s3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 52:54:00:27:28:65 brd ff:ff:ff:ff:ff:ff
    inet 10.5.1.26/24 brd 10.5.1.255 scope global dynamic noprefixroute enp0s3
       **valid_lft 86386sec** preferred_lft 86386sec
    inet6 2601:19b:4500:150::6c/128 scope global dynamic noprefixroute
       valid_lft 3590sec preferred_lft 1790sec
    inet6 fe80::4501:7972:f41f:6e3e/64 scope link noprefixroute
       valid_lft forever preferred_lft forever

It has now obtained its lease again for 1 day. I'd be happy to help troubleshoot. Right now I cant rely on HA to run any routines as its failing off the network.   Point me in the direction of whats requesting DHCP (assuming thru NetworkManager) on the base OS of HA and I'd be happy to troubleshoot further. I'm not familiar enough with what tools are installed on the base image that is running docker.

See others having similar issues:
https://community.home-assistant.io/t/losing-internet-connection-on-virtual-box/318922/4
https://github.com/home-assistant/operating-system/issues/545
https://community.home-assistant.io/t/hassio-unstable-needs-to-be-sent-to-an-asylum/158247

agners, 2021-09-02:
I am assuming you are using the OVA/qcow2 image?

I would expect that NetworkManager is making sure that things are setup correctly. Is maybe NetworkManager log saying something? 

```
journalctl -u NetworkManager
```

davealger64, 2021-09-03:
Yes - using the OVA/qcow2 image to install. 

NetworkManager is failing to renew the IP address for some reason. Note there are literally dozens of other devices using DHCP working perfectly fine. 

Snippet from the log: 
Sep 03 10:38:29 homeassistant NetworkManager[300]: <info>  [1630665509.5043] device (enp0s3): DHCPv4: grace period expired
Sep 03 10:39:06 homeassistant NetworkManager[300]: <warn>  [1630665546.6597] dhcp4 (enp0s3): selecting lease failed: -131

I've attached a longer version of the log. You can see where HA was able to initially get a lease, but then renewing the lease is failing after that.  From the behavior - this appears to be the same issue others have that have "solved" by setting to a static IP address. Unfortunately Google doesnt have much for what error "131" means.

[HA-NetworkMAnagerLog.txt](https://github.com/home-assistant/operating-system/files/7105860/HA-NetworkMAnagerLog.txt)

agners, 2021-09-03:
Hm, this seems to be the same issue: https://www.reddit.com/r/archlinux/comments/g65sj0/arch_networkmanager_keeps_going_haywire/

The linked NetworkManager merge requests points to this RedHat issue: https://bugzilla.redhat.com/show_bug.cgi?id=1829178. There they claim the problem is solved in 1.22.12, we are using 1.22.16 (the last version of the 1.22 branch). So it appears not to be that exact problem :(

Is it always the same error number -131?

davealger64, 2021-09-03:
Yes - its always the same error message.  Is there anyway to update
NetworkManager.conf?

Right now I'm getting that the /etc/NetworkManager/NetworkManager.conf is
read only when trying to edit. One of the suggestions was to modify the
dhcp line in the config file.

This appears to be by default
# mount
/dev/sda3 on / type squashfs (ro,relatime)
/dev/sda7 on /etc/NetworkManager/system-connections type ext4 (rw,relatime)

dhclient is installed, but complains ifconfig is missing when I run it
manually as well:

# dhclient enp0s3
/sbin/dhclient-script: line 83: ifconfig: not found
/sbin/dhclient-script: line 1: hostname: not found
/sbin/dhclient-script: line 102: hostname: not found
/sbin/dhclient-script: line 120: ifconfig: not found
/sbin/dhclient-script: line 126: route: not found
/sbin/dhclient-script: line 8: can't create /etc/resolv.conf.dhclient:
Read-only file system
chmod: /etc/resolv.conf.dhclient: No such file or directory
/sbin/dhclient-script: line 19: can't create /etc/resolv.conf.dhclient:
Read-only file system
/sbin/dhclient-script: line 19: can't create /etc/resolv.conf.dhclient:
Read-only file system
cat: can't open '/etc/resolv.conf.*': No such file or directory
#

agners, 2021-09-03:
> Is there anyway to update NetworkManager.conf?

Yes, I think you can add configs into a file in `/run/NetworkManager/conf.d/` e.g. `/run/NetworkManager/conf.d/custom.conf`, it should get parsed by network manager. Make sure to restart NetworkManager after editing.

Yeah not sure how to run dhclient manually to reflect well what NetworkManager is doing.

closed 2021-12-09 by github-actions[bot]

ultrafez, 2023-02-19:
I seem to be having the same issue. Did you ever find a solution, or more info on the cause of the problem @davealger64 ?