home-assistant/operating-system #1530: Allow setting WiFi regulatory domain

agners, 2021-09-03:
```
$ ha os info
board: rpi4
boot: A
update_available: false
version: "6.2"
version_latest: "6.2"
```

```
$ ha network scan wlan0 | grep -i frequency
- frequency: 2437
- frequency: 2437
- frequency: 2412
- frequency: 2412
- frequency: 2412
- frequency: 2462
- frequency: 2462
```

I have a Unifi AP right beside it broadcasting on channel 153 / VHT80.

_Originally posted by @deviantintegral in https://github.com/home-assistant/operating-system/issues/1441#issuecomment-912669650_

agners, 2021-09-03:
Hm, using `iw list` shows that only some frequencies are enabled:

```
        Band 1:
...
                Frequencies:                                                                                                                                                                                                                                                                                                  
                        * 2412 MHz [1] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2417 MHz [2] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2422 MHz [3] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2427 MHz [4] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2432 MHz [5] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2437 MHz [6] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2442 MHz [7] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2447 MHz [8] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2452 MHz [9] (20.0 dBm)                                                                                                                                                                                                                                                                             
                        * 2457 MHz [10] (20.0 dBm)                                                                                                                                                                                                                                                                            
                        * 2462 MHz [11] (20.0 dBm)                                                                                                                                                                                                                                                                            
                        * 2467 MHz [12] (20.0 dBm)                                                                                                                                                                                                                                                                            
                        * 2472 MHz [13] (20.0 dBm)                                                                                                                                                                                                                                                                            
                        * 2484 MHz [14] (disabled)
        Band 2:
....
                Frequencies:
                        * 5170 MHz [34] (disabled)
                        * 5180 MHz [36] (20.0 dBm)
                        * 5190 MHz [38] (disabled)
                        * 5200 MHz [40] (20.0 dBm)
                        * 5210 MHz [42] (disabled)
                        * 5220 MHz [44] (20.0 dBm)
                        * 5230 MHz [46] (disabled)
                        * 5240 MHz [48] (20.0 dBm)
                        * 5260 MHz [52] (20.0 dBm) (no IR, radar detection)
                        * 5280 MHz [56] (20.0 dBm) (no IR, radar detection)
                        * 5300 MHz [60] (20.0 dBm) (no IR, radar detection)
                        * 5320 MHz [64] (20.0 dBm) (no IR, radar detection)
                        * 5500 MHz [100] (20.0 dBm) (no IR, radar detection)
[... cut ...]
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

You can set the default regulatory domain used by the kernel by adding `cfg80211.ieee80211_regdom=<COUNTRY_CODE>` e.g. `cfg80211.ieee80211_regdom=US` to `cmdline.txt` (present in the first partition of your SD card, or in `/mnt/boot/cmdline.txt` through the OS shell).

Ideally we should set the regulatory domain through OS Agent using information the user entered during onboarding.

regevbr, 2021-10-13:
After some investigation, it seems that you need to have it as a one liner to make it work:
```sh
# cat /mnt/boot/cmdline.txt 
dwc_otg.lpm_enable=0 console=tty1 cfg80211.ieee80211_regdom=US
```

Vlad-Star, 2021-10-17:
> After some investigation, it seems that you need to have it as a one liner to make it work:
> 
> ```shell
> # cat /mnt/boot/cmdline.txt 
> dwc_otg.lpm_enable=0 console=tty1 cfg80211.ieee80211_regdom=US
> ```

Correct, these are the options supplied to Linux kernel at boot, must be one-liner.

deviantintegral, 2022-07-30:
This would still be a nice addition to the OS. If it's unlikely to be implemented, would a docs update be OK instead?

Once set, I had to re-join the Wifi network. Otherwise, it still preferred the 2.4GHz band.

TonsiTT, 2023-09-19:
Coming from issue #2755,  whenever I create the file /mnt/boot/cmdline.txt and set the option, it gets deleted on reboot.
Any idea

tyeth, 2024-05-02:
Bumping this issue, as a new user to Home-Assistant (RPi) I felt it was unreasonable that I couldn't configure wifi from the CLI (I don't have a spare usb stick and didn't spot that technique in Getting Started guide), and even then my network is on a UK specific band which I'd like to obviously use as part of setting up network connectivity.
Editing cmdline.txt is fine for me, I have other linux devices, but I wouldn't expect a normal user to accomplish this.