home-assistant/operating-system #1531: writing the image results in a not bootable disk

increddibelly, 2021-09-04:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [X] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [X] Fresh installation of release 6.2
- [ ] Updated from version x.y
- Additional information (if accessible):

N/A

**Supervisor logs:**

N/A

**Journal logs:**

N/A

**Kernel logs:**

N/A

**Description of problem:**

when I write the HAOS image from https://github.com/home-assistant/operating-system/releases/download/6.2/haos_generic-x86-64-6.2.img.xz I expect to stick the drive in a target machine and be able to boot. 
Burning that image resulted in a non-bootable disk.

I followed the steps [here](https://www.home-assistant.io/installation/generic-x86-64)
In contrast,  burning the latest ubuntu from their release URL _did_ result in a bootable disk. same balena machine, same disk, same target machine, no changes except disk image.

Burning the HAOS image afterwards again to the same drive again resulted in a non bootable system.

Could it be that the installation image partition wasn't set as bootable?

agners, 2021-09-05:
What kind of system are you trying to install it on? Is your system UEFI capable (and is EFI boot enabled?)

increddibelly, 2021-09-15:
the system bios is spamming UEFI on every single page, almost like a cheap commercial. However, when I stuck the drive in yet another different PC, it instantly booted up HassOS. 
I guess I got fooled into thinking my UEFI-shouting PC would  actually handle UEFI.
( ASRock B85M Pro4 Bios 2.30 )

closed 2021-09-15 by increddibelly