home-assistant/operating-system #1532: POPP ZB-Stick stopped working in Home Assistant OS 6.3

TabakM4n, 2021-09-05:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [x] OVA (Proxmox and virtual box)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [x] Fresh installation of release 6.3
- [x] Updated from version 6.2
- Additional information (if accessible):
## System Health

version | core-2021.9.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.6
os_name | Linux
os_version | 5.10.61
arch | x86_64
timezone | UTC

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.3
-- | --
update_channel | stable
supervisor_version | supervisor-2021.09.0
docker_version | 20.10.7
disk_total | 30.8 GB
disk_used | 2.7 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Visual Studio Code (3.6.2), Log Viewer (0.11.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

**Supervisor logs:**
```
[services.d] done.
[15:11:34] INFO: Starting local supervisor watchdog...
[... 120 lines cut ...]
```
 File "/usr/local/lib/python3.9/site-packages/bellows/zigbee/application.py", line 151, in startup
    status, node_type, nwk_params = await ezsp.getNetworkParameters()
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
2021-09-05 18:16:29 ERROR (MainThread) [zigpy.application] Couldn't start application
2021-09-05 18:16:29 ERROR (MainThread) [homeassistant.components.zha.core.gateway] Couldn't start EZSP = Silicon Labs EmberZNet protocol: Elelabs, HUSBZB-1, Telegesis coordinator
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 152, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 69, in new
    await app.startup(auto_form)
  File "/usr/local/lib/python3.9/site-packages/bellows/zigbee/application.py", line 151, in startup
    status, node_type, nwk_params = await ezsp.getNetworkParameters()
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
......
```

agners, 2021-09-05:
Hm, other than the update to the latest stable Linux kernel I don't see a change which might influence ZB Stick behavior.

Is your OS 6.2 installation fully up-to-date (same Core/Supervisor version as the 6.3 installation?).

Can you also check the Host log (Supervisor -> System -> Host) and see if you see a difference?

TabakM4n, 2021-09-05:
Everything is updated with the latest version. I see some differences in the host logs, but I do not know what I am searching for. 

The logs from the same VM just switching between HA OS versions:
<details>
  <summary>Logs 6.2</summary>
  
  ```
[    0.000000] Linux version 5.10.53 (builder@8b47b16fd37b) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -gf358c8ba-dirty) 9.3.0, GNU ld (GNU Binutils) 2.35.2) #1 SMP Wed Jul 28 20:42:41 UTC 2021
[    0.000000] Command line: apparmor=1 security=apparmor console=ttyS0 console=tty1 bootchooser.active=A  root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rootwait systemd.machine_id=f1b18bd3ca2b40d391fa721de534405b zram.enabled=1 zram.num_devices=3
[    0.000000] [Firmware Bug]: TSC doesn't count with P0 frequency!
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007d9b3fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007d9b4000-0x000000007d9b4fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007d9b5000-0x000000007de90fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007de91000-0x000000007df1afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007df1b000-0x000000007ef9afff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ef9b000-0x000000007eff2fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007eff3000-0x000000007effafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007effb000-0x000000007effefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007efff000-0x000000007f46afff] usable
[    0.000000] BIOS-e820: [mem 0x000000007f46b000-0x000000007ffeffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi: ACPI=0x7effa000 ACPI 2.0=0x7effa014 SMBIOS=0x7effd000 
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 2450d001, primary cpu clock
[    0.000000] kvm-clock: using sched offset of 304453424651 cycles
[    0.000003] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000005] tsc: Detected 3593.254 MHz processor
[    0.002135] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.002137] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.002141] last_pfn = 0x7f46b max_arch_pfn = 0x400000000
[    0.002276] MTRR default type: uncachable
[    0.002277] MTRR fixed ranges enabled:
[    0.002278]   00000-9FFFF write-back
[    0.002279]   A0000-FFFFF uncachable
[    0.002279] MTRR variable ranges enabled:
[    0.002280]   0 base 000000000000 mask FFFF80000000 write-back
[    0.002281]   1 base 00007FFF0000 mask FFFFFFFF0000 uncachable
[    0.002281]   2 disabled
[    0.002282]   3 disabled
[    0.002282]   4 disabled
[    0.002282]   5 disabled
[    0.002282]   6 disabled
[    0.002283]   7 disabled
[    0.002291] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.114110] check: Scanning 1 areas for low memory corruption
[    0.114444] Secure boot disabled
[    0.114449] ACPI: Early table checksum verification disabled
[    0.114454] ACPI: RSDP 0x000000007EFFA014 000024 (v02 VBOX  )
[    0.114457] ACPI: XSDT 0x000000007EFF90E8 000044 (v01 VBOX   VBOXFACP 00000001      01000013)
[    0.114463] ACPI: FACP 0x000000007EFF7000 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[    0.114469] ACPI: DSDT 0x000000007EFF4000 002325 (v02 VBOX   VBOXBIOS 00000002 INTL 20100528)
[    0.114472] ACPI: FACS 0x000000007EFFE000 000040
[    0.114474] ACPI: APIC 0x000000007EFF8000 00005C (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
[    0.114477] ACPI: SSDT 0x000000007EFF3000 0001CC (v01 VBOX   VBOXCPUT 00000002 INTL 20100528)
[    0.114480] ACPI: BGRT 0x000000007D9B4000 000038 (v01 INTEL  EDK2     00000002      01000013)
[    0.114483] ACPI: Reserving FACP table memory at [mem 0x7eff7000-0x7eff70f3]
[    0.114485] ACPI: Reserving DSDT table memory at [mem 0x7eff4000-0x7eff6324]
[    0.114485] ACPI: Reserving FACS table memory at [mem 0x7effe000-0x7effe03f]
[    0.114486] ACPI: Reserving APIC table memory at [mem 0x7eff8000-0x7eff805b]
[    0.114487] ACPI: Reserving SSDT table memory at [mem 0x7eff3000-0x7eff31cb]
[    0.114488] ACPI: Reserving BGRT table memory at [mem 0x7d9b4000-0x7d9b4037]
[    0.114537] ACPI: Local APIC address 0xfee00000
[... 1311 lines cut ...]
[   74.058622] audit: type=1325 audit(1630864882.248:141): table=nat family=2 entries=8 op=xt_replace pid=3869 subj==unconfined comm="iptables"
[   74.059633] audit: type=1325 audit(1630864882.249:142): table=nat family=2 entries=10 op=xt_replace pid=3870 subj==unconfined comm="iptables"
[   74.060853] audit: type=1325 audit(1630864882.250:143): table=nat family=2 entries=11 op=xt_replace pid=3871 subj==unconfined comm="iptables"
[   74.062200] audit: type=1325 audit(1630864882.252:144): table=nat family=2 entries=12 op=xt_replace pid=3872 subj==unconfined comm="iptables"
[   74.064129] audit: type=1325 audit(1630864882.254:145): table=nat family=2 entries=13 op=xt_replace pid=3873 subj==unconfined comm="iptables"
[   74.073369] eth0: renamed from veth02444fb
[... cut ...]
[   79.236801] audit: type=1334 audit(1630864887.427:153): prog-id=17 op=LOAD
[   79.237273] audit: type=1334 audit(1630864887.428:154): prog-id=18 op=LOAD
[  109.258256] audit: type=1334 audit(1630864917.450:155): prog-id=16 op=UNLOAD
[  109.258259] audit: type=1334 audit(1630864917.450:156): prog-id=15 op=UNLOAD
[  109.315814] audit: type=1334 audit(1630864917.507:157): prog-id=18 op=UNLOAD
[  109.315817] audit: type=1334 audit(1630864917.507:158): prog-id=17 op=UNLOAD

  ```
</details>

Let me know if you need anything more.

tjharman, 2021-09-05:
I found with 6.3 that presses of my Zigbee Button (An Xiaomi Aqura button) sometimes doesn't work, sometimes is very delayed (I'll press it and between 0-10 seconds will laps before it works)

I have rolled back to 6.2 and all it working as expected again.  My other Zigbee devices seemed to function OK (Xiaomi power plugs)

I use a Zig a Zig Ah stick.

I realise this is a pretty usless post with zero logs, only offered as another datapoint for the "something's changed in 6.3 that's impacting Zigbee"

agners, 2021-09-05:
Make sure to check CPU load.

There is really nothing special with 6.3, just a regular kernel update. Make sure to downgrade to 6.2 to double check if it is really related to the OS.

agners, 2021-09-05:
@TabakM4n

The 6.2 log shows a network related stack trace, I am surprised that this does not lead to issues on 6.2 :sweat_smile: 

In both 6.2 and 6.3 the Linux kernel seems to detect a ch341 based USB to serial adapter which I guess is the POPP ZB-Stick. It is attached as ttyUSB0 in both cases. What device are you referencing in the ZHA integration?

TabakM4n, 2021-09-05:
@agners 

In both cases:

1. Enter Manually
2. Radio type `EZSP = Silicon Labs EmberZNet protocol: Elelabs, HUSBZB-1, Telegesis`
3. Serial device path `/dev/ttyUSB0` 
4. port speed `115200`
5. data flow control `hardware`

Exactly as it is specified in the PDF, except `5` because probably this option is added to home assistant after this tutorial was made.
https://elelabs.com/wp-content/uploads/2020/07/ELU013_UG_11_HomeAssistant_Hassio_0.112.4.pdf

I see the device in Supervisor -> System -> Host -> Hardware

It connects in both versions and correctly pick up the `/dev/serial/by-id/.....` path but in `6.3` I get errors and I cannot add devices.

What is strange about it - is that when I add the integration in `6.3` and then downgrade to `6.2`  -> the integration starts on boot without any problems.

pedrolamas, 2021-09-05:
Kernel update is the reason for this problem, as described here: https://github.com/Koenkk/zigbee2mqtt/issues/8623#issuecomment-913202614

I experienced this issue myself with zigbee2mqtt and managed to confirm that previous kernels are working fine, but 5.10 and above can repro the problem with some Aqara devices (such as the door contact sensor)

tjharman, 2021-09-05:
@pedrolamas Thank you so much.  I shall bow out of this thread, sorry for the noise.

puddly, 2021-09-05:
The kernel update seems to only affect the CH340 driver, but the Elelabs stick (and I think the POPP Zb-Stick) uses a Silicon Labs CP210x chip, which has its own driver. I don't think this was changed as well, was it?

puddly, 2021-09-05:
Ignore my last comment, the Elelabs ELU013 / POPP ZB-Stick sticks definitely use the CH340 according to their docs:

[image]

agners, 2021-09-06:
Linux 5.10.62 indeed fixes a CH340/CH341 related issue, specifically:
https://bugzilla.kernel.org/show_bug.cgi?id=214131

I'll update the kernel to the latest version for 6.4.

agners, 2021-09-06:
Only now reading @pedrolamas comment :see_no_evil: . But yeah that seems to be the culprit.

agners, 2021-09-06:
For now, if you experience issues with CH340 devices, please revert back to 6.2 until 6.4 is out:

```
ha os update --version 6.2
```

bossgos, 2021-09-07:
> For now, if you experience issues with CH340 devices, please revert back to 6.2 until 6.4 is out:
> 
> ```
> ha os update --version 6.2
> ```

Today HASS OS 6.4 has been released. Is it still affected with this bug?

tjharman, 2021-09-07:
Where do you see 6.4?

https://github.com/home-assistant/operating-system/releases

bossgos, 2021-09-07:
> Where do you see 6.4?
> 
> https://github.com/home-assistant/operating-system/releases

https://www.home-assistant.io/blog/2021/09/01/release-20219/

bossgos, 2021-09-07:
> Where do you see 6.4?
> 
> https://github.com/home-assistant/operating-system/releases

aww sorry my mistake... I was check Home Assistant Core :( with 4 at the end...

TabakM4n, 2021-09-09:
I can confirm that after upgrading to `HASS OS 6.4`, the integration started working.

agners, 2021-09-09:
This is addressed in HAOS 6.4 and newer (via kernel update PR #1536).

closed 2021-09-09 by agners