home-assistant/operating-system #1533: Enocean Dongle USB 300 not recognized by odroid N2+

pschyrembel258, 2021-09-05:
**Hardware Environment**
I've baught the HA blue (odroid n2+) and moved my installation from a VMM to the new odroid HA installation

**Home Assistant OS release:**

-## System Health

version | core-2021.9.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.6
os_name | Linux
os_version | 5.10.61
arch | aarch64
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.15.2
Stage | running
Available Repositories | 880
Installed Repositories | 36

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.3
-- | --
update_channel | stable
supervisor_version | supervisor-2021.09.0
docker_version | 20.10.7
disk_total | 113.9 GB
disk_used | 10.9 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.13.0), File editor (5.3.3), Mosquitto broker (6.0.1), NGINX Home Assistant SSL proxy (3.0.2), Samba share (9.5.1), AirSonos (3.1.1), FTP (4.2.1), Log Viewer (0.11.1), Node-RED (10.0.0), Vaultwarden (Bitwarden) (0.13.0), php-nginx Docker Image (latest), rest980 Docker Image (20200205), evcc (0.62), Tor (3.0.5), Z-Wave JS to MQTT (0.24.1), AdGuard Home (4.1.9), SSH & Web Terminal (9.0.1), Portainer (2.0.0), UniFi Controller (1.0.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 25
views | 11
mode | storage

</details>

**Supervisor logs:**
21-09-05 18:55:00 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE1963477-if00
21-09-05 18:55:00 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/011 - None
21-09-05 18:55:03 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/012 - None
21-09-05 18:55:03 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE1963477-if00
21-09-05 18:55:08 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE1963477-if00
21-09-05 18:55:08 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/012 - None
21-09-05 18:55:11 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/013 - None
21-09-05 18:55:11 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE1963477-if00

**Description of problem:**

I've an USB 300 Enocean Dongle which worked perfectly on my VM, but my new odroid n2+ HA Blue doesn't recognize the stick. ZWAVE  and ZIGBEE both were recognized and working perfectly.

agners, 2021-09-05:
It seems that the EnOcean USB 300 Dongle uses the FTDI chip to emulate a serial port. The driver is enabled on ODROID-N2..

```
CONFIG_USB_SERIAL_FTDI_SIO=m
```

Are you using a USB Hub? Can you plug it into another USB port? Can you unplug all other USB devices and post a screenshot of the Hardware screen when searching for "USB" devices?

pschyrembel258, 2021-09-06:
I've unplugged all usb sticks and tried all usb ports with no success. I made each time a restart of the odroid.How can I search for usb devices?

agners, 2021-09-06:
Under Supervisor -> System -> Thee dots on the bottom right of the Host section -> Hardware.  In that dialog you see all hardware detected. You can use `USB` to filter by USB devices.

pschyrembel258, 2021-09-06:
OK, that's what I did all the time. But there are no USB Devices listed.
[image]

pschyrembel258, 2021-09-08:
I've ordered a second Usb300 stick, to see if it is only a problem with my current stick. Or do you have other suggestions, to solve the problem?

pschyrembel258, 2021-09-08:
Second stick is working. Everything is fine. Thank you for your help.

agners, 2021-09-09:
I see so it was a hardware issue. Thanks for the update!

closed 2021-09-09 by agners