home-assistant/operating-system #1534: Home Assistant Supervisor ERRORS

Berzell, 2021-09-05:
After update
Home Assistant Supervisor ERRORS
Could not load the Supervisor panel!
Z-Wave JS doesn't work after update

- [X ] Updated from version Home Assistant 2021.9.1 (?) to 2021.9.2

- [ X] ODROID [C2/C4/N2(+)/XU4]

----------
Home Assistant observer
Supervisor: Connected
----------

## System Health

error | failed to load: unknown
-- | --

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 25 september 2021 2:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

error | failed to load: unknown
-- | --

</details>
2021-09-05 21:58:02 ERROR (MainThread) [homeassistant.components.hassio.handler] /homeassistant/options return code 403
2021-09-05 21:58:02 ERROR (MainThread) [homeassistant.components.hassio.handler] /supervisor/options return code 403
2021-09-05 21:58:02 ERROR (MainThread) [homeassistant.components.hassio.handler] /info return code 403
2021-09-05 21:58:02 WARNING (MainThread) [homeassistant.components.hassio] Can't read last version:
2021-09-05 21:58:02 ERROR (MainThread) [homeassistant.components.hassio.handler] /ingress/panels return code 403
2021-09-05 21:58:02 ERROR (MainThread) [homeassistant.components.hassio.addon_panel] Can't read panel info:
2021-09-05 21:58:02 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Supervisor for hassio
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 304, in async_setup
result = await component.async_setup_entry(hass, self) # type: ignore
File "/usr/src/homeassistant/homeassistant/components/hassio/__init__.py", line 600, in async_setup_entry
coordinator = HassioDataUpdateCoordinator(hass, entry, dev_reg)
File "/usr/src/homeassistant/homeassistant/components/hassio/__init__.py", line 679, in __init__
self.is_hass_os = "hassos" in get_info(self.hass)
TypeError: argument of type 'NoneType' is not iterable
2021-09-05 21:58:04 ERROR (MainThread) [homeassistant.setup] Error during setup of component usb
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/setup.py", line 255, in _async_setup_component
result = await task
File "/usr/src/homeassistant/homeassistant/components/usb/__init__.py", line 69, in async_setup
await usb_discovery.async_setup()
File "/usr/src/homeassistant/homeassistant/components/usb/__init__.py", line 102, in async_setup
await self._async_start_monitor()
File "/usr/src/homeassistant/homeassistant/components/usb/__init__.py", line 114, in _async_start_monitor
info = await system_info.async_get_system_info(self.hass)
File "/usr/src/homeassistant/homeassistant/helpers/system_info.py", line 52, in async_get_system_info
info_object["supervisor"] = info.get("supervisor")
AttributeError: 'NoneType' object has no attribute 'get'
2021-09-05 21:58:04 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of zwave_js. Setup failed for dependencies: usb
2021-09-05 21:58:04 ERROR (MainThread) [homeassistant.setup] Setup failed for zwave_js: Could not set up all dependencies.
2021-09-05 21:58:04 ERROR (MainThread) [homeassistant.components.light] The rfxtrx platform for the light integration does not support platform setup. Please remove it from your config.
2021-09-05 21:58:05 ERROR (MainThread) [homeassistant.components.updater] Unexpected error fetching Home Assistant update data: 'NoneType' object is not subscriptable
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 187, in _async_refresh
self.data = await self._async_update_data()
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 147, in _async_update_data
return await self.update_method()
File "/usr/src/homeassistant/homeassistant/components/updater/__init__.py", line 82, in check_new_version
newest = core_info["version_latest"]
TypeError: 'NoneType' object is not subscriptable
2021-09-05 21:58:09 ERROR (MainThread) [homeassistant.components.hassio.handler] /discovery return code 403
2021-09-05 21:58:09 ERROR (MainThread) [homeassistant.components.hassio.discovery] Can't read discover info:
2021-09-05 21:58:11 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/hass_nabucasa/iot.py", line 113, in _async_handle_handler_message
await self.client.send_json(response)
File "/usr/local/lib/python3.9/site-packages/aiohttp/client_ws.py", line 164, in send_json
await self.send_str(dumps(data), compress=compress)
File "/usr/local/lib/python3.9/site-packages/aiohttp/client_ws.py", line 150, in send_str
await self._writer.send(data, binary=False, compress=compress)
File "/usr/local/lib/python3.9/site-packages/aiohttp/http_websocket.py", line 687, in send
await self._send_frame(message, WSMsgType.TEXT, compress)
File "/usr/local/lib/python3.9/site-packages/aiohttp/http_websocket.py", line 656, in _send_frame
await self.protocol._drain_helper()
File "/usr/local/lib/python3.9/site-packages/aiohttp/base_protocol.py", line 84, in _drain_helper
assert waiter is None or waiter.cancelled()
AssertionError
2021-09-05 21:58:11 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/hass_nabucasa/iot.py", line 113, in _async_handle_handler_message
await self.client.send_json(response)
File "/usr/local/lib/python3.9/site-packages/aiohttp/client_ws.py", line 164, in send_json
await self.send_str(dumps(data), compress=compress)
File "/usr/local/lib/python3.9/site-packages/aiohttp/client_ws.py", line 150, in send_str
await self._writer.send(data, binary=False, compress=compress)
File "/usr/local/lib/python3.9/site-packages/aiohttp/http_websocket.py", line 687, in send
await self._send_frame(message, WSMsgType.TEXT, compress)
File "/usr/local/lib/python3.9/site-packages/aiohttp/http_websocket.py", line 656, in _send_frame
await self.protocol._drain_helper()
File "/usr/local/lib/python3.9/site-packages/aiohttp/base_protocol.py", line 84, in _drain_helper
assert waiter is None or waiter.cancelled()
AssertionError

agners, 2021-09-05:
This is hardly Operating System related: You a 2021.9.1 and 2021.9.2 are Home Assistant Core versions. Did you try restart the system?

Berzell, 2021-09-05:
Several times

closed 2021-09-05 by Berzell