home-assistant/operating-system #1535: After OS Upgrade 6.3 Host time is set to date: 2021 Aug 31 and time: 11:00 after start

Zoriontsu, 2021-09-06:
### The problem

I had a robust system working properly, until I upgraded to OS 6.3
Now my system time  is set to date: 2021 Aug 31 and time: 11:00 after every reboot.
My RPi4 boots OK and I can ssh into it. I am able to access locally but not via Nabucasa (probably due to time issue).
I have tested my NTP server and it works OK. I tried loading a previous backup, but the issue continues.
I have spent hours looking a possible similar issues from the community, but nothing seems to fix the issue.

### What is version of Home Assistant Core has the issue?

core-2021.9.3

### What was the last working version of Home Assistant Core?

core-2021.9.1

### What type of installation are you running?

Home Assistant OS

### Anything in the logs that might be useful for us?

```txt
This is the only interesting entry in the logs.

[    1.974731] systemd[1]: System time before build time, advancing clock.
```

agners, 2021-09-06:
Some OS components have been upgraded, but for the most part this upgrades were patch/security updates.

> I have tested my NTP server and it works OK. 

How do you set the NTP server? Home Assistant OS does not (and did also not) honor DHCP provided NTP so far (there is an open issue for this #689).

Can you try downgrade to 6.2 and reconfirm that it indeed used to work in OS 6.2?

```
ha os update --version 6.2
```

Zoriontsu, 2021-09-06:
Thank you for the prompt reply. I solved the issue overnight.
- To troubleshoot my ntp server functionality, I booted the RPi4 with the latest Raspbian and it acquired the time OK. 
- I then created a fresh HASS image on a different SD and I noticed in the console that it would get stuck on:
      [OK] Started Docker Application Container Engine
      [*]  A start job is running for Wait Kernel Time Synchronized...
- Started inspecting OS, Core, and Supervisor info and noticed the timezone was empty for some reason.
- Typed ha su options -t America/Chicago and it populated the timezone. 
- Shutdown the fresh install, replaced with my production ssd, booted RPi and upon boot the time was OK.
I truly don't understand why this happened, but I am happy it is working now (after sweating bullets for two days). 
Any idea what would cause the Supervisor not to aquire timezone info?
Thanks again.

agners, 2021-09-06:
Hm, so it seems that in your setup HAOS cannot access the time server? Since you also get `A start job is running for Wait Kernel Time Synchronized...`.

Zoriontsu, 2021-09-07:
That seems like the case. 
Now I have the correct date and able to reach host remotely, but my host time is 30 minutes behind.
I spoke too soon. Again, booting the RPi4 with Raspbian the host retrieves the correct time/date.
With Hass OS, it times out retrieving it. 
I took a few rudimentary pictures of the screen during boot as well as the config.
Notice it always pauses waiting for Kernel Time Sync.
[Boot Screen Images.zip](https://github.com/home-assistant/operating-system/files/7118534/Boot.Screen.Images.zip)

Zoriontsu, 2021-09-07:
I believe I found the problem this time!
My RPi worked fine for a long time without the host being turned off. After the last OS update, I turned off the Pi to clean  the dust.
Puzzled by the fact that only HASS OS did not acquire NTP I played with my Unifi Security Gateway for a couple of hours, changing settings and rebooting RPi host. 
Turns out that the Unifi USG GeoIP filtering was the issue. I have only allowed traffic to/from the Americas and Europe in my LAN for 10+ months.I became little paranoid about Russian/Chinese/Indian hackers since my Unifi Threat Management was constantly blocking threats from those geos.
Somehow, HASS OS does not like that filtering. Once I turned off GeoIP filtering, my host acquires time.
Not a huge deal, but very curious why that affects only Hass OS. Thanks for your help.

closed 2021-09-07 by Zoriontsu