home-assistant/operating-system #1538: Raspberry Pi 4 stuck after updating to 6.3 when USB Keyboard is connected

agners, 2021-09-06:
**Hardware Environment**

- [x] Raspberry Pi 4
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 6.2
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

After updating to HAOS 6.3 booting can get stuck.  It seems that this is related to a USB Keyboard.

```
U-Boot 2021.04 (Sep 04 2021 - 00:27:03 +0000)

DRAM:  1.9 GiB
RPI 4 Model B (0xb03112)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Register 5000420 NbrPorts 5
Starting the controller
USB XHCI 1.00
scanning bus xhci_pci for devices... 3 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
2413 bytes read in 12 ms (196.3 KiB/s)
## Executing script at 02400000
switch to partitions #0, OK
mmc0 is current device
loading env...

MMC read: dev # 0, block # 1214464, count 32 ... 32 blocks read: OK
Trying to boot slot B, 0 attempts remaining. Loading kernel ...
8664336 bytes read in 487 ms (17 MiB/s)
storing env...

MMC write: dev # 0, block # 1214464, count 32 ... 32 blocks written: OK
Starting kernel
Kernel image @ 0x080000 [ 0x000000 - 0x843510 ]
## Flattened Device Tree blob at 2eff3b00
   Booting using the fdt blob at 0x2eff3b00
<freeze>
```

Unplugging the USB keyboard doesn't show the issue anymore.

antro31, 2021-09-06:
Hi agners, looks like 6.3 has overall issues with USB. I had a similar issue [#1539 ] where RPi refuses to boot with an RFLink connected on USB. 

If possible for you, I suggest that you ssh and downgrade ha os to 6.2
`ha os update --version 6.2`

Hope this will help.

closed 2021-09-09 by agners

agners, 2021-09-13:
This most likely has been uncovered since #1203 has been resolved.