home-assistant/operating-system #1539: RPi4 Home Assistant OS 6.3 won't boot if RFLink is plugged in a USB port

antro31, 2021-09-06:
**Hardware Environment**

- [X] Raspberry Pi4
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [X] Updated from version 6.2
- Additional information (if accessible):  Not available (System does not boot with 6.3)

**Supervisor logs:**
Not Available

**Journal logs:**
Not Available

**Kernel logs:**
Not Available.

**Description of problem:**
Using RPi4 with Homeassistant OS 6.2 (32 bits) with RFLink connected to RPi USB Port (no external power on RFLink)
When upgrading to 6.3, RPI does not boot anymore.  
RFLink connected to RPi has been identified has being the issue since unplugging it allows RPi to boot. 

Problem is reproducible. 
Booting with 6.2 with RFLink plugged in USB worked before. So this is a regression.

Had to rever to 6.2 to have everything working.

agners, 2021-09-06:
It seems that this is also a CH340 device. There is isse #1532. which tracks that problem.

closed 2021-09-06 by agners

antro31, 2021-09-09:
Hi agners, 
Tried just now with HomeAssistant OS 6.4, issue is still there. Should we re-open?