home-assistant/operating-system #1545: Home Assistant OS is not responsive to modprobe.d blacklist configuration

asdfghjkai, 2021-09-13:
**Hardware Environment**

- [X] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [X] Fresh installation of release 6.3 (6.4 released during investigation)
- [ ] Updated from version x.y
- Additional information (if accessible):
- ## System Health

version | core-2021.9.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.6
os_name | Linux
os_version | 5.10.17-v8
arch | aarch64
timezone | Europe/London

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4923
Installed Version | 1.14.1
Stage | running
Available Repositories | 953
Installed Repositories | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.3
-- | --
update_channel | stable
supervisor_version | supervisor-2021.09.0
docker_version | 20.10.7
disk_total | 219.4 GB
disk_used | 27.2 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Check Home Assistant configuration (3.8.0), AirCast (3.1.1), Samba share (9.5.1), Mosquitto broker (6.0.1), Nextcloud Backup (0.16.2), SSH & Web Terminal (9.0.1), Home Panel (2.2.0), Portainer (2.0.0), zigbee2mqttassistant (0.3.157), Zigbee2mqtt (1.21.1-1), Git pull (7.13.1), MariaDB (2.4.0), Grafana (7.0.2), InfluxDB (4.1.1), Wmbusmeters (W-MBus to MQTT) (0.2.19)

</details>

<details><summary>Lovelace</summary>
[... 790 lines cut ...]
[  195.549445] audit: type=1334 audit(1631560904.009:249): prog-id=18 op=LOAD
[  225.372373] audit: type=1334 audit(1631560933.833:250): prog-id=16 op=UNLOAD
[  225.372412] audit: type=1334 audit(1631560933.833:251): prog-id=15 op=UNLOAD
[  226.324472] audit: type=1334 audit(1631560934.785:252): prog-id=18 op=UNLOAD
[  226.324507] audit: type=1334 audit(1631560934.785:253): prog-id=17 op=UNLOAD
[  390.059282] veth920d42b: renamed from eth0
[  390.079318] hassio: port 8(veth622c5f4) entered disabled state
[  390.189929] hassio: port 8(veth622c5f4) entered disabled state
[  390.191570] device veth622c5f4 left promiscuous mode
[  390.191628] audit: type=1700 audit(1631561098.586:254): dev=veth622c5f4 prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
[  390.191923] hassio: port 8(veth622c5f4) entered disabled state
[  390.380614] audit: type=1325 audit(1631561098.838:255): table=nat family=2 entries=15 op=xt_unregister pid=377 subj==unconfined comm="kworker/u8:2"
[  390.380655] audit: type=1325 audit(1631561098.838:256): table=filter family=2 entries=4 op=xt_unregister pid=377 subj==unconfined comm="kworker/u8:2"
# 

**Description of problem:**
I'm trying to use a dvb tuner as an SDR to communicate (or listen) to responses from my home water meter. The details do not matter too much, the main issue is that the tuner is (correctly) identified by the OS as a DVB tuner and as such loads the appropriate module. Because I do not want it to do this, I followed the instructions [here](https://github.com/home-assistant/operating-system/blob/dev/Documentation/configuration.md) to add an entry to modprobe.d, to blacklist the kernel module in question. For what it's worth, it's dvb_usb_rtl28xxu.

This upon restart had no effect, and I started digging down towards the os build, and busybox, which has a number of configurations around modprobe [here](https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/busybox.config#L565) including toggles to enable blacklisting. I confirmed I would need to enable CONFIG_MODPROBE as well as CONFIG_FEATURE_MODPROBE_BLACKLIST which I found within the source [here](https://git.busybox.net/busybox/tree/modutils/modprobe.c)

I built a 6.3 image (at the time) from source using the instructions provided, and with this subtle tweak, and then tested on my RPi - the same symptoms.

Question is, what else needs to be changed, is there anyway to integrate this as part of the standard build, or is the idea bust and I should just purchase a proper radio dongle.

I was hoping the above would provide a solution and I could raise a PR, but unfortunately not. Happy to try alternative options and build from source to confirm/deny if any proposed fixes work.

agners, 2021-09-13:
The `modprobe` binary used in HAOS is not from the busybox package but from the kmod package (`BR2_PACKAGE_KMOD`). Afaict this gets selected implicitly somehow. This is the tool which comes from [kernel.org](https://git.kernel.org/pub/scm/utils/kernel/kmod/kmod.git) and I think blacklist is a feature enabled by default.

Are you sure you blacklist the module correctly? From what I can tell the module filename is `dvb-usb-rtl28xxu.ko`, so I'd guess you need to use `blacklist dvb-usb-rtl28xxu` to blacklist that module. Although, I read on the Arch Linux Wiki that `-` and `_` can be used interchangeably, so I guess its not that.

agners, 2021-09-13:
Maybe another module is depending on it, which causes it to get loaded? There is a work around using a `install` instruction documented on the Arch Linux wiki: https://wiki.archlinux.org/title/Kernel_module#Blacklisting

asdfghjkai, 2021-09-14:
Thanks for letting me know I was going down the rabbit hole! I think the blacklist 'should' have worked, given it looks for the name of the loaded module (Which uses underscores versus the filename (Which likely uses hyphens to stand out) - also to stick with convention.

Let me take a look. I think I found the module it that depends on it (And blocked that also) but looks like no others.
I'll try the arch work around and report back

`
# cat /etc/modprobe.d/blacklist-dvb-2.conf 
blacklist dvb_usb_rtl28xxu
blacklist dvb_usb_v2# 
# 
# lsmod | grep dvb
dvb_usb_rtl28xxu       32768  1
dvb_usb_v2             32768  1 dvb_usb_rtl28xxu
# 
`

Cheers @agners

asdfghjkai, 2021-09-14:
Apologes, still no good. Still loading?!? Restarted betweenb

`cat /etc/modprobe.d/blacklist-dvb-2.conf`
install dvb_usb_rtl28xxu /bin/true

`lsmod | grep dvb`
dvb_usb_rtl28xxu       32768  1
dvb_usb_v2             32768  1 dvb_usb_rtl28xxu

`ls /dev | grep dvb`
dvb

agners, 2021-09-14:
@asdfghjkai how do you access the OS shell?

asdfghjkai, 2021-09-14:
@agners One of two methods
1. Interactively by typing login on the device, and accessing like that
2. Remotely (After creating an SSH key pair and placing it on my CONFIG usb). Using ssh root@10.0.100.9 -p 22222

In this case, not using the SSH addon because I wanted to access the underlying OS before it passes anything through to the HA container.

agners, 2021-09-14:
@asdfghjkai [this Reddit post](https://www.reddit.com/r/RTLSDR/wiki/blacklist_dvb_usb_rtl28xxu) lists multiple `blacklist` lines, maybe that helps?

```
blacklist dvb_usb_rtl28xxu
blacklist rtl2832
blacklist rtl2830
```

Maybe `journalctl` is saying something?

```
journalctl -b 1 | grep modprobe
```

Can you also try the following: Have the blacklist setup the way you think it should work. Plug out the DVB tuner. Boot Home Assistant completely, make sure the modules are not loaded, plug-in the DVB tuner, check if the modules are loaded now (it might be that the blacklist directory gets mounted too late, bascially modprobe loading the modules too early).

asdfghjkai, 2021-09-14:
@agners Just have that ago, same symptoms

cat /etc/modprobe.d/blacklist-dvb-2.conf 
blacklist dvb_usb_rtl28xxu
blacklist rtl2832
blacklist rtl2830

dvb_usb_rtl28xxu       32768  1
dvb_usb_v2             32768  1 dvb_usb_rtl28xxu

Also nothing from journalctl :(

Just tried the second method
Restarted not connected, lsmod | grep dvb returns nothing, good, no sign of it in /dev also. Good so far

Plug in the device and it doesn't load any drivers. This I think will suffice for now, the integration is able to sense the raw device and work with it. Still not quite right, but I think it will have to do

agners, 2021-09-14:
> Plug in the device and it doesn't load any drivers. This I think will suffice for now, the integration is able to sense the raw device and work with it. Still not quite right, but I think it will have to do

So it did not load the drivers correct?

Hm, that means that our mount is too late for blacklist to work. I'll have a closer look at that, but it will take me some time to debug.

agners, 2021-10-06:
Indeed it seems that blacklist module is currently not working. The reason is that the `/etc/modprobe.d` mount point is loaded after `systemd-udevd`, hence the module is loaded before `systemd-udevd` has a chance to read the config files from `/etc/modprobe.d`. I tried to reverse the start order (start `systemd-udevd` after `/etc/modprobe.d` is mounted), but that seems not to work. From what I can tell `systemd-udevd` is really essential to get anything going in terms of mounting etc. so basically we cannot mount before `systemd-udevd` is running, a chicken-egg problem.

From [this comment](https://github.com/systemd/systemd/issues/13947#issuecomment-549935626) it seems that mount `/etc` as part of the regular system boot seems not officially supported by systemd. I'd guess bind mount parts of `/etc` is somewhat an edge case.

I think the only proper way to fix this would be to mount `/etc` in full (bind mount all sub directories) in a initramfs is probably the proper solution for this. But this would mean a major change for HAOS...

However, there is good news also: There is a work around: You can add `module_blacklist=dvb_usb_rtl28xxu` to `/mnt/boot/cmdline.txt`, which tells the kernel on a low level to ignore loading a module with that name.

@asdfghjkai can you try this approach?

Harryjholmes, 2022-05-06:
> However, there is good news also: There is a work around: You can add `module_blacklist=dvb_usb_rtl28xxu` to `/mnt/boot/cmdline.txt`, which tells the kernel on a low level to ignore loading a module with that name.

That doesn't seem to work. Do you know of any other workarounds?

agners, 2022-05-06:
It does work, see official Linux kernel documentation of `module_blacklist` at https://www.kernel.org/doc/html/v4.14/admin-guide/kernel-parameters.html.

Did you reboot after setting that parameter? Do you have that driver/card in question? Can you share your host logs (dmesg)?

sairon, 2025-03-28:
This has been pinned int the past but there are no follow ups for several years. It doesn't seem to go anywhere and there is no obvious bug.

closed 2025-03-28 by sairon