home-assistant/operating-system #1546: You are running an unsupported installation (Network Manager)

drfinn15, 2021-09-14:
**Hardware Environment**

- [x] Raspberry Pi [4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version ?
- Additional information (if accessible): Updated HA Core from core-2021.9.3 to core-2021.9.6

## System Health

version | core-2021.9.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.6
os_name | Linux
os_version | 5.10.17-v8
arch | aarch64
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.15.2
Stage | running
Available Repositories | 944
Installed Repositories | 13

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | September 20, 2021, 2:00 AM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.3
-- | --
update_channel | stable
supervisor_version | supervisor-2021.09.0
docker_version | 20.10.7
disk_total | 28.6 GB
disk_used | 17.0 GB
healthy | true
supported | failed to load: Unsupported
board | rpi4-64
[... 78 lines cut ...]
21-09-14 14:11:14 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
21-09-14 14:11:14 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
21-09-14 14:11:14 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-09-14 14:11:14 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-09-14 14:11:14 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-09-14 14:11:14 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-09-14 14:11:14 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-09-14 14:19:22 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-09-14 14:49:22 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-09-14 15:07:44 INFO (MainThread) [supervisor.api.middleware.security] /snapshots access from cebe7a76_hassio_google_drive_backup
**Journal logs:**

not possible

**Kernel logs:**

not possible

**Description of problem:**

Since today morning I see a warning message on the supervisor tab telling me "You are running an unsupported installation". The "Learn more" button shows that this is related to the network manager and directs we to this page:
. I am running up-to-date versions of everything and can't explain myself neither where it is coming from, nor how to resolve it. Only red line in the supervisor logs is
`21-09-14 09:10:30 ERROR (MainThread) [supervisor.utils] Can't execute check_trust while a task is in progress`
[image]
This button leads me to [https://www.home-assistant.io/more-info/unsupported/network_manager]

agners, 2021-09-14:
Hm, it seems that the Supervisor somehow deems HAOS's Network Manager as unsupported.

Does that issue persists after reboot?

drfinn15, 2021-09-14:
> Hm, it seems that the Supervisor somehow deems HAOS's Network Manager as unsupported.
> 
> Does that issue persists after reboot? 

yes, it does

agners, 2021-09-15:
What kind of network configuration do you use and how did you set it?

Does Restarting the Supervisor helps? And just to be sure the UI doesn't hold some stale state, refresh the page after restarting with Ctrl+F5.

drfinn15, 2021-09-15:
> What kind of network configuration do you use and how did you set it?
> 
> Does Restarting the Supervisor helps? And just to be sure the UI doesn't hold some stale state, refresh the page after restarting with Ctrl+F5.

Actually I cant tell you a lot about my network config - haven't changed a lot here. Basically I just plugged my Raspberry via LAN cable into my router. Also, I haven't changed anything in regards to hardware at all before the warning came up. 
I just restarted supervisor and HA as well again, also refreshed page as you recommended, but nothing changed so far. On the mobile app I do see the message too. 
Not sure if it provides anything to you, but here are my logs after the HA restart:
[image]

drfinn15, 2021-09-16:
So I've just updated my OS from version 6.3 to 6.4 and the warning message disappeared. Thanks for your commitment!

closed 2021-09-16 by drfinn15