home-assistant/operating-system #1549: Building HAOS for qemuarm-64

adamfeldman, 2021-09-16:
**I want to run HAOS in a VM on Raspberry Pi 4. I recognize it's likely an unsupported deployment method for HAOS. Has anyone done this before, for development purposes? Does anyone have KVM or QEMU commands or scripts to share?**

I see some references to [`qemuarm-64`](https://github.com/home-assistant/operating-system/blob/dev/Documentation/development.md) in the HAOS development docs but have been unable to get a build to complete.

I was unable to figure-out how to get the standard Pi `u-boot`-based Pi 4 OS image to boot in QEMU or KVM. I tried command invocations to get QEMU and KVM to "skip" over the bootloader and boot the kernel image directly.

I spent a day tinkering, using a large cloud VM as a kernel build box. I tried building my own OS image by creating a new board with `barebox` instead of `u-boot`, but could not get a build to complete after ~5 iterations.

agners, 2021-09-17:
Related to that, there was an issue about running Home Assistant OS on Arm using ESXi: #1050

> I tried command invocations to get QEMU and KVM to "skip" over the bootloader and boot the kernel image directly.

Its probably quick if you get it running, but it won't allow for updates etc. I would try to get the whole boot flow working.

That is a bit the problem in the ARM world: The boot firmware's are not well standardized and every machine does its own thing basically. U-Boot is probably still the most popular and customizable firmware for Arm machines. There is also a push for UEFI based boot flows especially on Aarch64. It seems that ESXi comes with UEFI firmware, and expects a virtual image to use the UEFI boot flow. I'd guess that using UEFI boot flow for an Aarch64 virtual image is likely the most versatile solution. But at least at this point seems to require quiet a bit of work especially using KVM/Qemu (see http://cdn.kernel.org/pub/linux/kernel/people/will/docs/qemu/qemu-arm64-howto.html).

I think for the KVM/Qemu use case work using `qemu-system-aarch64 --machine virt` and the `qemu_arm64_defconfig` U-Boot configuration is probably the quickest way to success.

There is some information here:
https://github.com/u-boot/u-boot/blob/master/doc/develop/driver-model/virtio.rst
https://github.com/u-boot/u-boot/blob/master/doc/board/emulation/qemu-arm.rst

Once you have U-Boot working, you can integrate that into a new machine, say qemu-aarch64 or similar.

As for the kernel, I'd try to use the mainline kernel (as opposed to use the Raspberry Pi kernel). For the buildroot machine config I would use ODROID-N2 as base since its mainline kernel based and an Aarch64 platform.

Would be cool to have something going for Arm virtualization!

neingeist, 2021-12-29:
I was trying to boot a stock haos_rpi4-64-7.0.img on a Pi 4 b, using qemu and got u-boot to load the kernel from the image, the kernel does however does not find the root partition.

* I built u-boot v2021.10 on the Pi 4 b itself, with `qemu_arm64_defconfig` + SquashFS + LZO decompression support
* `qemu-system-aarch64 -machine virt -cpu cortex-a72 -bios u-boot.bin -nographic -drive if=none,file=haos_rpi4-64-7.0.img,format=raw,id=hd0 -device virtio-blk-pci,drive=hd0` (no KVM yet)

This gets me until:
~~~
...
[    4.782396] uart-pl011 9000000.pl011: no DMA platform data
[    4.797543] Waiting for root device PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd...
~~~

So this stock kernel probably has no support for the VirtIO device, AFAICT

agners, 2021-12-30:
> So this stock kernel probably has no support for the VirtIO device, AFAICT

Quite possible, its based on `bcm2711_defconfig` from the Raspberry Pi Kernel, which is optimized by the RPi team to run on RPi.

I think this really calls for a new Buildroot config/board. It probably should be named `ova-arm64` (so `buildroot-external/configs/ova_arm64_defconfig`), and be stored in `buildroot-external/board/arm64/ova`. Building HAOS using Buildroot is documented here if you are interested to further dig into it: https://developers.home-assistant.io/docs/operating-system

neingeist, 2021-12-30:
I was experimenting a bit with the stock image to see how far I could get - as I am somewhat experienced with QEMU but totally new to the ARM platform, buildroot, the bootloaders etc.

Now I'll try buildroot :+1:

neingeist, 2022-01-02:
Update: I managed to build *something* but it does not boot. My virtual UEFI does not boot the built barebox (from the qcow2 image I build using a custom buildroot config). As I have no idea what I am doing actually: @agners Do you know any IRC channel (or similar) to ask? It's purely for fun, I just want a virtual HAOS for my private use.

agners, 2022-01-03:
If its really Barebox which doesn't come up probably [their](https://www.barebox.org/) mailing list/IRC channel is the best place to ask.

Does Barebox support UEFI on Aarch64? It doesn't look like:
https://www.barebox.org/doc/latest/boards/efi.html

Also, there is only `arch/x86/configs/efi_defconfig` but nothing similar in the `arm` architecture folder.

Similarly, it seems that U-Boot also only supports x86 currently:
https://u-boot.readthedocs.io/en/latest/develop/uefi/u-boot_on_efi.html#future-work

Buildroot itself has a [aarch64_efi_defconfig](https://github.com/analogdevicesinc/buildroot/blob/master/configs/aarch64_efi_defconfig) configuration, which seems to use Grub. It seems that our update system [RAUC supports Grub](https://rauc.readthedocs.io/en/latest/integration.html#grub). I guess that is the only/best option at this point.

neingeist, 2022-01-03:
barebox seems to support ARM64 https://www.barebox.org/doc/latest/boards/aarch64-qemu-virt.html, but as I had more success with U-Boot earlier, and as I am somewhat lost with debugging anything UEFI, I think I'll try that first.

Doridian, 2022-02-16:
I just tried to make HAOS compile for aarch64/EFI with some changes and hit that issue with barebox not supporting this combination.

Sadly I don't seem to have enough knowledge of the configs to make this work. I keep getting weird linker errors in GRUB.

Some of the errors I get (I get a lot more, but they pretty much complain about the same things)
```
/build/output/host/lib/gcc/aarch64-buildroot-linux-gnu/9.4.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: libgrubmods.a(libgrubmods_a-libgrub_a_init.o): in function `grub_init_all':
libgrub_a_init.c:(.text+0x48): undefined reference to `grub_f2fs_init'
/build/output/host/lib/gcc/aarch64-buildroot-linux-gnu/9.4.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: libgrub_a_init.c:(.text+0xfc): undefined reference to `grub_romfs_init'
/build/output/host/lib/gcc/aarch64-buildroot-linux-gnu/9.4.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: libgrubmods.a(libgrubmods_a-libgrub_a_init.o): in function `grub_fini_all':
libgrub_a_init.c:(.text+0x184): undefined reference to `grub_f2fs_fini'
/build/output/host/lib/gcc/aarch64-buildroot-linux-gnu/9.4.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: libgrub_a_init.c:(.text+0x238): undefined reference to `grub_romfs_fini'
```

Here's a link to my fork in case I am doing something stupid: https://github.com/Doridian/operating-system/tree/feat/aarch64-efi
(I named my thing generic-aarch64, as my intention is for this to also boot on bare metal ARM64 EFI boards, but mainly as an ARM64 EFI VM for various systems, like M1 MacBooks, or my Traverse Ten64 with muVirt)

agners, 2022-02-16:
Hm, seem to be some linking errors when trying to link Grub. I'd say this is likely a Buildroot Grub packaging problem, but hard to tell. I'd check the latest version of Buildroot and maybe also check the mailing list.

Doridian, 2022-02-17:
After realizing a `make clean` can fix a lot of weird errors while you're digging through configs, I got it to compile.
I also got further by fixing a few more things and got it actually booting into grub now.
Problem is the A/B scheme of HAOS does (obviously) not seem to ship grub configs for itself, yet, so it doesn't actually boot anything at all, just has a "buildroot" grub option which points to a file (while HAOS ships a kernel partition)

Manually providing grub with the necessary `(hd0,gpt2)/Image` and `root=/dev/vda3` arguments (which should be taken from current boot state) makes it work.
I will have to do some research to figure out the best method to implement the A/B scheme. Unless someone else has any ideas of course.

Doridian, 2022-02-17:
I have gotten it to build a booting image without A/B booting, but just hardcoded the 0 partitions in essentially.

The problem is GRUB's search feature can only search by "contains file", "filesystem UUID", "filesystem label".
It cannot search by partition label or partition UUID, which from my understanding makes A/B booting essentially useless, except if you hardcoded `(hd0,gptX)` in, but then if GRUB's device name changes? So you'd have to "okay, check your own device, then from that device, go into the X'th partition". This seems horribly flimsy if I'm honest.

Doridian, 2022-02-17:
I have now, with some regex module help, actually managed to get grub to recognize the correct things in a somewhat sane way.
Which means it can now boot both A and B, and it can source which it should boot from a config file.

Failed boot counting is missing, as is changing the file after a flash. Haven't figured out how to make GRUB actually write a counter file just yet.
Also haven't figured out how HAOS actually resets the failed boot counter.

agners, 2022-02-19:
> Also haven't figured out how HAOS actually resets the failed boot counter.

That depends on the case. On success, its rauc which resets it, on failure, its the bootloader (boot script) which switches A<->B and resets the counter.

rauc supports various bootloader backends, see https://rauc.readthedocs.io/en/v1.5.1/integration.html#id4. Grub is also in the list. Set `BOOTLOADER` in the meta file of your buildroot board and rauc should set the variables accordingly.

agners, 2022-02-19:
> I have now, with some regex module help, actually managed to get grub to recognize the correct things in a somewhat sane way. Which means it can now boot both A and B, and it can source which it should boot from a config file.

I see you went the UUID path. Since we hardcode UUID (just like partition label) its kinda equivalent. Also we use this approach for platforms using GPT and U-Boot (Raspberry Pi/ASUS Tinker, see e.g. `buildroot-external/board/raspberrypi/uboot-boot64.ush`).

From what I can tell you only need to integrate your script with the variables rauc writes when using the Grub backend (see https://rauc.readthedocs.io/en/v1.5.1/integration.html#grub), and things should work end-to-end! Great work!

Doridian, 2022-02-19:
> I see you went the UUID path. Since we hardcode UUID (just like partition label) its kinda equivalent. Also we use this approach for platforms using GPT and U-Boot (Raspberry Pi/ASUS Tinker, see [...]

I went the UUID path where possible (root=PARTUUID=).
But if you look at where grub pulls the Image (aka kernel) from, that is done by partition index because GRUB cannot read partition labels or UUIDs at all, only filesystem labels and UUIDs (which are obviously identical for A/B slots)

Also thanks for your pointers, I have something I think should work now. What would be the easiest way to actually test it? Is there like a "RAUC, just write slot B as nextboot" debug command? I can't seem to see one.

Doridian, 2022-02-21:
Opened a PR here: https://github.com/home-assistant/operating-system/pull/1757/files
From the testing I could do without a RAUC PKI setup, everything seems working well.
Of course, making it boot from slot B, which is unflashed, causes an immediate boot failure.

agners, 2022-02-23:
Thanks @Doridian, merged your PR today! :tada: 

This means HAOS 8.0 will come with a new machine called `generic-aarch64`. It supports UEFI boot and has been tested with QEMU so far. Make sure to pass QEMU suitable UEFI firmware images.

```sh
dd if=/dev/zero of=flash1.img bs=1M count=64
dd if=/dev/zero of=flash0.img bs=1M count=64
dd if=/usr/share/qemu-efi-aarch64/QEMU_EFI.fd of=flash0.img conv=notrunc
```

```sh
qemu-system-aarch64 -vga std -machine virt,gic-version=max -m 1G -cpu max -smp 4 \
    -netdev user,id=vnet,net=192.168.76.0/24,dhcpstart=192.168.76.9,hostfwd=tcp::8123-:8123,hostfwd=tcp::22222-:22222 \
    -device virtio-net-pci,netdev=vnet \
    -drive file=haos_generic-aarch64-8.0.dev20220223.qcow2,format=qcow2,if=none,id=drive0,cache=writeback \
    -device virtio-blk,drive=drive0,bootindex=0 \
    -drive file=flash0.img,format=raw,if=pflash -drive file=flash1.img,format=raw,if=pflash
```

See also: https://futurewei-cloud.github.io/ARM-Datacenter/qemu/how-to-launch-aarch64-vm/

For those who want to give this a try already today, you can find development builds here: https://os-builds.home-assistant.io/8.0.dev20220223/.

closed 2022-02-23 by agners