home-assistant/operating-system #1551: HM-MOD-RPI-PCB not working, no /dev/ttyAMA0

Komet, 2021-09-20:
**Hardware Environment**

- [x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [x] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

## System Health

version | core-2021.9.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.17-v8
arch | aarch64
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 2923
Installed Version | 1.15.2
Stage | running
Available Repositories | 879
Installed Repositories | 1

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.4
-- | --
update_channel | stable
supervisor_version | supervisor-2021.09.0
docker_version | 20.10.7
disk_total | 57.8 GB
disk_used | 3.4 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.1.3), Samba share (9.5.1), File editor (5.3.3), deCONZ (6.10.0), HomeMatic CCU (11.3.0)

</details>

[... 693 lines cut ...]
[   68.785978] audit: type=1701 audit(1632164179.695:150): auid=4294967295 uid=0 gid=0 ses=4294967295 subj==unconfined pid=5084 comm="wmiir" exe="/usr/bin/wmiir" sig=11 res=1
[   68.822053] audit: type=1334 audit(1632164179.731:151): prog-id=15 op=LOAD
[   68.822787] audit: type=1334 audit(1632164179.731:152): prog-id=16 op=LOAD
[   69.125919] audit: type=1334 audit(1632164180.035:153): prog-id=16 op=UNLOAD
[   69.125940] audit: type=1334 audit(1632164180.035:154): prog-id=15 op=UNLOAD
[   74.374079] audit: type=1334 audit(1632164185.283:155): prog-id=12 op=UNLOAD
[   74.374102] audit: type=1334 audit(1632164185.283:156): prog-id=11 op=UNLOAD
[   74.413463] audit: type=1334 audit(1632164185.323:157): prog-id=14 op=UNLOAD
[   74.413484] audit: type=1334 audit(1632164185.323:158): prog-id=13 op=UNLOAD
```
</details>

**Description of problem:**

I'm trying to use the Homematic addon with a HM-MOD-RPI-PCB (which works fine with RaspberryMatic). I've done a fresh install of HAOS 6.4 64bit (also tried the 32 bit version), enabled these four lines in /mnt/boot/config.txt

```
enable_uart=1
dtparam=i2c_arm=on
dtoverlay=miniuart-bt
dtoverlay=rpi-rf-mod
```

but I don't get the /dev/ttyAMA0 device. Am I missing something?
If I don't enable the dtoverlay in config.txt I'm seeing /dev/ttyAMA0 but this obviously can't work with the addon.

agners, 2021-09-21:
@jens-maus haven't looked closer into it yet, maybe you have an idea what could be missing?

jens-maus, 2021-09-21:
@Komet Please note, that the `rpi-rf-mod` dtoverlay is not meant to provide a `/dev/ttyAMA0` device, but provide access to a `/dev/raw-uart` device for the required low latency communication with a `HM-MOD-RPI-PCB` or `RPI-RF-MOD`. So please try to use that instead.

In addition, also note, that the `config.txt` section you are referring to, is mostly meant to provide the necessary raw-uart device for the RaspberryMatic HA Add-on. So why are you not using the RaspberryMatic HA add-on (see https://github.com/jens-maus/RaspberryMatic/wiki/Installation-HomeAssistant) and refrain from using the older (and partly obsolete) Homematic Add-on?

Komet, 2021-09-21:
Oh wow, now I feel a little bit dumb! ;) I thought RaspberryMatic HA Addon was an old one which now has been integrated as Homematic Addon into the official addon repository. 

The documentation of the "official" addon states that you need to add `enable_uart=1 dtoverlay=miniuart-bt` in config.txt. But the default config.txt looks like
```
# Uncomment this to enable GPIO support for RPI-RF-MOD/HM-MOD-RPI-PCB
# enable_uart=1
# dtparam=i2c_arm=on
# dtoverlay=miniuart-bt
# dtoverlay=rpi-rf-mod
```
so I've enabled dtparam and the second dtoverlay.

I'm now using RaspberryMatic HA Addon and everything is working fine. Thank you very much for your prompt help!

closed 2021-09-21 by Komet

jens-maus, 2021-09-21:
> I thought RaspberryMatic HA Addon was an old one which now has been integrated as Homematic Addon into the official addon repository.

Nope, it is indeed the other way around and the RaspberryMatic HA Add-on overall is newer and has more functionality than the old Homematic Add-on. Dunno if the old one has still active users, so it might be perhaps time to think about abandoning it in favour of the RaspberryMatic Add-on. What do you think @pvizeli @agners ?