home-assistant/operating-system #1552: After OS upgrade from 6.2 to 6.4 Aeotec Z-Stick Gen5+ (ZW090) stopped working

Mpgod80, 2021-09-22:
Hi.

After OS upgrade from OS 6.2 to 6.4 Aeotec Z-Stick Gen5+ (ZW090) and Conbee stick stopped working.
I use an USB hub to connect them to my Odroid N2+ Home Assistant Blue Bundle Limited Edition.

From the integration "Z-wave JS" i get this error msg: "Retrying setup: Cannot connect to host a0d7b954-zwavejs2mqtt:3000 ssl:default [Connect call failed ('172.30.33.6', 3000)]"

If i go in to the Z-wave to JS settings i get this msg: 
"Driver: Failed to initialize the driver: Timeout while waiting for an ACK from the controller"
Serial port is set to: /dev/ttyACM0

I have restored to an earlier backup that worked, and reverted back to OS version 6.2. Then the Conbee stick started working again, but not the Aeotec Z-Stick Gen5+ (ZW090) stick. I have also done factory reset to the Aeotec Z-Stick, and no luck.

What should i do to get this working again?

After you removed the OS 6.3 and then released the 6.4 version i thought that you had fixed this issue with the usb stick problem in 6.3. But the problem is obvious still there. A friend of mine with an equal setup as mine also ran in to this problem when you released 6.3.

**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [X ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [X ] Updated from version 6.2
- Additional information (if accessible):

## System Health

version | core-2021.9.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.6
os_name | Linux
os_version | 5.10.53
arch | aarch64
timezone | Europe/Stockholm

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4935
Installed Version | 1.15.2
Stage | running
Available Repositories | 945
Installed Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 3 oktober 2021 2:00
relayer_connected | true
remote_enabled | false
remote_connected | true
alexa_enabled | false
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.2
-- | --
update_channel | stable
supervisor_version | supervisor-2021.09.0
docker_version | 20.10.6
disk_total | 113.9 GB
disk_used | 15.1 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | TellStick (2.1.0), Node-RED (10.0.0), Terminal & SSH (9.1.3), File editor (5.3.3), Visual Studio Code (3.6.2), Grafana (7.2.0), InfluxDB (4.2.1), AppDaemon 4 (0.7.0), Z-Wave JS to MQTT (0.24.1), Z-Wave JS (0.1.40)

</details>

<details><summary>Lovelace</summary>

dashboards | 6
-- | --
resources | 7
views | 7
mode | storage

</details>

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

agners, 2021-09-22:
The Z-Wave problem sounds like there is a problem communicating with the Add-On:

```
From the integration "Z-wave JS" i get this error msg: "Retrying setup: Cannot connect to host a0d7b954-zwavejs2mqtt:3000 ssl:default [Connect call failed ('172.30.33.6', 3000)]"
```

Is your add-on running? Any errors in the add-on log?

I'd first try to get Z-Wave running again on 6.2.

There was a reliability issue with certain USB to Serial based chips (see #1532), but I don't think that this should affect Connbee or Z-Wave sticks.

agners, 2021-09-22:
What also would be interesting are the kernel logs from Supervisor -> System -> Log Provider "Host"? Ideally from Version 6.2 and 6.4.

Mpgod80, 2021-09-23:
My add-on is running yes.

 Well the logs updating from 6.2 to 6.4 is not saved. But i am not the only one whit this problem. Se this thread created yesterday : https://community.home-assistant.io/t/failed-to-start-the-z-wave-js-add-on/340913.
He also run the Aeotec Z-Stick Gen5+ (ZW090).

I got this out from the Z-wave JS add on today: 

"Failed to initialize the driver: Timeout while waiting for an ACK from the controller
Error in driver ZWaveError: Failed to initialize the driver: Timeout while waiting for an ACK from the controller
    at Immediate.<anonymous> (/usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:793:6) {
  code: 5,
  context: undefined,
  transactionSource: undefined
}"

And in the link i posted that person gets the same error log as me......

HAPSagan, 2021-09-23:
I had similar same issue as Mpgod80 when I upgraded from 6.2 to 6.3. But first it was even worse – My Odroid refused to let me in at all. I don´t know If it was stuck in the boot up sequence or if it was something else. I had to unplug both the Aeotec Zwave stick gen 5 and my Conbee 2 and after that I had my Odroid unplugged for about half an hour. When I was able to log in at last neither the Zwave or Zigbee network was working. The Zigbee got up after a while by itself but not the Zwave. I tried everything I could think of but it said "Driver: Failed to initialize the driver: Timeout while waiting for an ACK from the controller". I did so much that I can´t tell what lastly got it working. If it was a reset of the Aotec stick, me restoring a backup, a combination of that or just luck…
Now I´m running OS 6.3 and feel I don’t dare to test the 6.4 when I see that the problem still exists. I thought this was solved when 6.3 was removed and you then released the 6.4, but obviously not.

Mpgod80, 2021-09-23:
After many hours of searching what´s wrong, i found what caused this huge problem.

I started to check what serialports the USB devices was using. The Aeotec Zwave stick gen 5+ should by default use /dev/ttyACM0.
But the other stick i use Conbee 2 had run over the Aeotec Zwave stick gen 5+ and snatched the serial port /dev/ttyACM0.
So when i tried to switch the port for the Zwave stick to /dev/ttyACM1 instead, it started working again!

So that is what happened when i upgraded the OS from 6.2 to 6.4. 

I guess that the Conbee 2 stick boots faster and therefore takes over the zwave sticks serialport in the boot sequence.
And all this has to do with the new function with Autodetect on the USB ports.

Is there a chance to add an option to turn off that function in HA in future updates?

pmoneill, 2021-09-23:
I reposting below the string to which Mpgod80 refers above.  Maybe I should have posted it here first.  I have not been able to fix it so my extensive Home Assistant system is useless.

On 22 Sept. 2021, after updating the Z-Wave JS add-on from 1.37 to 1.39 and then1.40, HA Core to 2021.9.7, and HA OS to 6.4 from what was current on 4 Sept. 2021, I lost all Z-Wave functionality getting the error message “Failed to start the Z-Wave JS add-on”.

Removing both the Z-Wave JS add-on and integration, then installing the integration with its automatic installation of the add-on gives the same error. Doing many variations/iterations of removing both Z-Wave components and rebooting, HA automatically discovers my Aetec Z-Stick Gen 5 and installs the integration but fails to start the add-on saying “Aborted. Failed to start the Z-Wave JS add-on”, then it automatically removes the integration. Restoring my 4 Sept. 2021 full backup does not fix this.

The Z-Wave log mentions the following driver issue but I don’t know how to provide the “valueDB” that apparently is missing:
“2021-09-23T00:32:25.297Z DRIVER Failed to initialize the driver: Cannot read property ‘valueDB’ of undefined
Error in driver ZWaveError: Failed to initialize the driver: Cannot read property ‘valueDB’ of undefined
at Immediate. (/usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:793:6) {
code: 5,
context: undefined,
transactionSource: undefined”

I simply did a routine update and lost Z-Wave. How do I restore Z-Wave JS functionality short of wiping my SSD and reinstalling everything?

I am running the standard HA OS installation on a NUC. Prior to these updates it was working fine except that after every reboot after an update I had to re-interview each Z-Wave device that I have added since switching from Classic Z-Wave.

sacofpotatoes, 2021-09-24:
I am having this same issue,  on a Raspberry Pi 4. If I revert to HAOS 6.3 Z-Wave starts working again.

pmoneill, 2021-09-24:
I reverted to HAOS 6.2 on my NUC and some Z-Wave devices work again.  Only the devices I added since I earlier switched to Z-Wave JS are available.  All of my devices added to my Aeotec Z-Stick Gen 5 under Classic Z-Wave (or openHAB) are unavailable and not in my device registry, however trying to include them again does not work, apparently because they are already in the stick.  Even the devices that are available have to be manually re-interviewed after a reboot to be made available.
We have a big mess here that may force me to factory-reset my Z-Stick and include my dozens of devices all over again.  I don't have the software skills to contribute to this so I made a monetary contribution.  In the meantime, the system that I rely on to run my house is dead.

sacofpotatoes, 2021-09-24:
> I reverted to HAOS 6.2 on my NUC and some Z-Wave devices work again. Only the devices I added since I earlier switched to Z-Wave JS are available. All of my devices added to my Aeotec Z-Stick Gen 5 [...]

I have noticed as well that many of my Z-Wave devices have disappeared after downgrading the OS. I am in the process of restoring a backup of home assistant, which I believe will bring all the devices back.... I don't know for sure yet if it will work, but it may be worth a shot if you have backups enabled.

pmoneill, 2021-09-24:
Sacofpotatoes:
The device and entity information for all devices, Z-Wave and others, seems to be stored by HA in its config/.storage directory in core.device_registry and core.entity_registry, respectively.  I think that these were part of the full minus HA Core backup that I restored.  However, I noticed that the corresponding files after this OS reversion and HA configuration restoration did not contain all of my prior devices and had a new format for the Z-Wave JS-driven devices, so these files must be built by HA during installation.  Copying the old registry files back in and restarting still only accessed my newer devices.

So now it looks like my only alternative is to reset my Z-Stick and include all my devices again under Z-Wave JS.  Dare I also move HA OS from 6.2 back up to 6.4, which started this fiasco?  Z-Wave has to be fundamental to most users' lighting, HVAC, and miscellaneous control so it's in the interest of the users and manufacturers of these devices to have good drivers.  We should all be supporting the Z-Wave driver developers any way we can.

agners, 2021-09-24:
> I started to check what serialports the USB devices was using. The Aeotec Zwave stick gen 5+ should by default use /dev/ttyACM0.
But the other stick i use Conbee 2 had run over the Aeotec Zwave stick gen 5+ and snatched the serial port /dev/ttyACM0.
So when i tried to switch the port for the Zwave stick to /dev/ttyACM1 instead, it started working again!

The order of devices is (and never was!) guaranteed. Every reboot, and of course also a change in OS version can shuffle the devices in a different order. That is just how Linux works.

For that reason, udev also creates aliases which are stable, based on the device VID/PID/serial number. You can find those under Supervisor -> System -> three dots -> Hardare, search for `tty`.

[image]

In this case I would use `/dev/serial/by-id/usb-0658_0200-if00` in the Add-On/Integration configuration.

FWIW, the current Z-Wave JS config flow at least offers the stable link by default:
[image]

Mpgod80, 2021-09-24:
I just tried to set that up in the Add-On/Integration configuration /dev/serial/by-id/usb-0658_0200-if00. 
But i had to try 4 times in row before is accepted the change.
Why is that?

My friend HAPSagan also tried to set this up and got this error code when trying "Driver: Error Resource temporarily unavailable Cannot lock port"

sacofpotatoes, 2021-09-24:
> Sacofpotatoes:
> The device and entity information for all devices, Z-Wave and others, seems to be stored by HA in its config/.storage directory in core.device_registry and core.entity_registry, [...]

After restoring an older version of HA I was still without the devices I was originally missing. So, like you, it seems  I will now need to manually repair all my devices. However, I was able to upgrade to 6.4 and Z-Wave worked this time... No clue what caused the issue to begin with.

pmoneill, 2021-09-25:
Sacofpotatoes:
I, too, was able to update to HA OS 6.4 and still have some Z-Wave functionality.  I spent most of today excluding all of my device from my Aeotec Z-Stick Gen 5, factory reseting the Z-Stick, and including everything I could back into the Z-Stick/HA.  Only about half of my devices would include.  12 Levitons, comprising 3 models succeeded.  4 Levitons comprising 2 of those same models (DZ15S-1BZ Decora Smart Switch,  DZ6HD-1BZ Decora 600W Smart Dimmer) failed to include, showing 3 red flashes at the end.  None of my 4 Aeotec ZW116 or ZW139 Nanoswitches nor my one ZW100 Multisensor gave any sign of attempting to pair but my two Aeotec ZW095 Home Energy Monitors paired right away.  Lots of network healings and restarts only gained one Leviton switch.  All three RCS/GoControl thermostats included easily, so it's really hit-or-miss now, when a week ago all this stuff was working fine.

agners, 2021-10-06:
The original poster @Mpgod80 reports that this was caused by the rename of the tty device. I suggest using `by-id` paths to avoid surprises in the future (see https://github.com/home-assistant/operating-system/issues/1552#issuecomment-926531072).

If anyone else can still reproduce issues caused by the OS upgrade, please open a new issue.

closed 2021-10-06 by agners