home-assistant/operating-system #1563: No logs after reboot

mattclar, 2021-10-03:
- [Pi4 ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
Hassos 6.4

- Home Assistant Frontend -> Configuration -> Info (use the copy button in the System Health block)
- Or use this command: `ha info`
-->

**Supervisor logs:**
NA

**Journal logs:**
See below

**Kernel logs:**
NA

**Description of problem:**
I have an intermittent freeze happening once every 24-48 hours for the last 2 weeks.
However when I reboot the logs are gone.
Even if I use journalctl -b -1 the logs only have entries from network manager.
I'm not sure if this is to do with whatever my error is or maybe it's a rpi thing as the internal clock isn't maintained on reboot there seem to be a few messages about setting the clock and that might be causing some logging to be lost?
It's kinda frustrating to not be able to find logs after a hard freeze like this makes instability hard to diagnose!

agners, 2021-10-03:
Is the above journal from a `journalctl` command? It seems to contain multiple boots, but they seem incomplete.

mattclar, 2021-10-04:
Yes this is my underlying issue, it seems as though the log is being
cleared somehow?
Yes the output is from journalctl

agners, 2021-10-04:
Home Assistant OS uses `systemd-journald` for logging, and its built-in log rotation/limit capability (see also https://www.freedesktop.org/software/systemd/man/journald.conf.html#SystemMaxUse=). However, with that you should have complete logs at for at least a couple of days.

Maybe the files get corrupted or similar? You can check the log files and print a list of log files using `journalctl`:

```
journalctl --verify
journalctl --header
```

mattclar, 2021-10-04:
Not sure if this is normal?

`# journalctl --verify
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@e1503ffe1dfd45ef902ca55910578ad3-0000000000000250-0005cc088aeac7b7.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@7b59e756c5524af6a19a0e686a86c5d4-000000000000049e-0005cd6dbbeb08a1.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@ea139bbc27274e9caa69aa4029c35ea3-000000000000027a-0005cbdef17fd198.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@72c048890ef04420996c63202e3ea049-000000000000096e-0005cb8b406ecc3a.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1003@c5144326b0a04543a46253c8bb8060b0-0000000000000456-0005cc6ab4a1ea4c.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@0005ccb858003716-dd5050988e183692.journal~
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1001@0005cbdef17e43aa-da820ed6f1cc0de2.journal~
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1001@0005cb91139c3e31-919c0155799976a4.journal~
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1001.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1001@a5923a38f1f0474e980f3cc97c39f151-00000000000004e4-0005cb6516a2f5ae.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@118b79832d6c4a9db879839482748927-0000000000000253-0005cbc8afa0c5ee.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1001@ba7400576ac74ec2800c7b100db3f8c8-0000000000000268-0005cbad8f5aec1b.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1003@d493e42edba54125a36cee0ad514a56a-0000000000000445-0005cbb102463f01.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@24ac17ca05344ae2aa4645dfd15e7951-000000000000024c-0005cbad8f544aae.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@f3187c02f5d345079072ec27fa154c23-0000000000000252-0005cc5153637fbe.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1003@20b921cfcc0c48e69358dc01dff9aa94-000000000000046b-0005cbe108ef8dee.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1003@8f70fceb97ad49179eac0e5e4ade01c1-0000000000000281-0005cc698c74304f.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1003@739d410f55d744538e13014882b4e49a-0000000000000680-0005cbb38611c87f.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@290df24beb97492b9c8cad56ebb4fb6e-0000000000000284-0005cc532621383a.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@72c048890ef04420996c63202e3ea049-00000000000004fe-0005cb88ef220305.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1001@0005ccb63a18f6fe-12982f804877e9b6.journal~
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@0005cd2fdb7bd006-c275d5cfaee21dfe.journal~
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1003@f1aeadea974848daa80a237ed362a10e-0000000000000288-0005cc5326213d61.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1003@0005cc698c7150b3-ab7154d5820d1a1f.journal~
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1003.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1003@71dbe95c1e6d4577b2b2e8107afd873d-00000000000006dc-0005cd6f70476d2c.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1001@98ad0334936f42068f10c5d485c620b5-0000000000000683-0005cbb38612b9bf.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@9e130e9bd2fc4a999a5c5d75ee8ccf72-0000000000000250-0005cd2fdb7b4863.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1001@0005cd710f6dfda6-d51353cf2f83d7a6.journal~
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@0005cd710f685989-fb7ad430bae149f2.journal~
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1003@bd63ed1afba7444aad86fd03893202c7-0000000000000b16-0005cbc7a1e212d1.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@4992b473d4b140e3a3f2584c878fe3dc-0000000000000254-0005cb91139855b6.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@ea139bbc27274e9caa69aa4029c35ea3-000000000000024c-0005cbdef17973a0.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1001@0f7e74e791164a8695d2acd5c939ee7c-0000000000000958-0005cb8b406d6c5e.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1001@0005cbc839a20e8f-9f2bf65b75e8cf1b.journal~
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1003@764be97208034df3bcbd3c82882cf571-00000000000004fa-0005cb88ef2161d5.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@0005cb6750ba64ec-dba9ab979f4a4d10.journal~
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@0005cc088aeb1249-e8e1098e373a6fa2.journal~
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@72c048890ef04420996c63202e3ea049-0000000000000736-0005cb8a450557ff.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1003@0005cd2fdb8036c9-99a3e9351bd733e7.journal~
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@c76b63e46f52415f87d4c18e2f04e5f5-0000000000000252-0005cd4b9ca73b98.journal
PASS: /var/log/journal/e458a7ce23954447b316934cba6c05e9/user-1004@24ac17ca05344ae2aa4645dfd15e7951-000000000000068c-0005cbb386141dce.journal
[... cut ...]
Rotate suggested: no
Head sequential number: 650 (28a)
Tail sequential number: 662 (296)
Head realtime timestamp: Fri 2021-10-01 14:45:18 UTC (5cd4b9cae8b42)
Tail realtime timestamp: Fri 2021-10-01 14:45:18 UTC (5cd4b9caf2def)
Tail monotonic timestamp: 6.003s (5b9b85)
Objects: 87
Entry objects: 4
Data objects: 35
Data hash table fill: 0.0%
Field objects: 24
Field hash table fill: 7.2%
Tag objects: 0
Entry array objects: 22
Deepest field hash chain: 0
Deepest data hash chain: 0
Disk usage: 8.0M
#`

agners, 2021-10-04:
Looks ok to me. I realized that `journalctl -b -1` to get the last boot seems not to work reliably at times. Not sure if that is a `journalctl` bug. But what seems to work for me is selecting a boot explicitly via:

```
journalctl --list-boots
```
Then select the hash of a boot and use

```
journalctl -b <hash>
```

mattclar, 2021-10-07:
Looks like your exactly correct! hopefully the fix flows downstream soon :)
[image]

mattclar, 2021-10-07:
also, thanks for looking into this :)