home-assistant/operating-system #1570: 6.4 won't boot on Tinkerboard S using stock image on emmc

Nachtmensch87, 2021-10-04:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [x] ASUS Tinker S
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [x] Fresh installation of release x.y
- [ ] Updated from version x.y

**Description of problem:**
Im attempting a fresh install of a default 6.4 on the tinkerboard S by flashing the emmc through balena. It flashes fine but upon trying to boot, nothing. The red LED stays on.  Both ethernet led's blink at random as well but the board never establishes a link to my DHCP server.

Setting jumper to maskrom and putting a SD with 6.4 in, it boots fine from SD.
So I believe its maybe again a Uboot issue but I can't be sure.

agners, 2021-10-04:
Just tested HAOS 6.4 here, dd'ed the latest version directly onto the eMMC. Booting seems to work fine.

Are you sure that you provide enough power to the board? E.g. typical USB micro power supplies are not enough.

How did you flash to the eMMC? Also make sure to remove the mask rom jumper (or set it to neutral) after flashing.

Nachtmensch87, 2021-10-05:
Thanks for your reply. Thats pretty strange. Flashed TinkerOS with Etcher in emmc and its booting fine (monitor plugged in).

So again flashed hass 6.4 os with Etcher. Same power supply, jumper is removed and nothing happens. Power supply output: 5V, 2.5 A

Well, when from your pov ist everthing fine, the issue can be closed...

agners, 2021-10-05:
Maybe its not properly uncompressing (it is a xz compressed  image, but Etcher should recognize that)?

Thanks for the update!