home-assistant/operating-system #1574: NIC issues

jraynes, 2021-10-06:
**Hardware Environment**
- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [X] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [ X] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**
error when downloading docker container "no supervisor internet"

**Description of problem:**
I am trying to install the latest version of Homeassistant on a BOXNUC7PJYH1. I download the image file and write it directly to a SATA SSD using balena etcher. When I plug the drive into the NUC and turn it on it boots fine but does not find the onboard wired NIC at all. When I look at the supervisor logs it appears to try and fetch the docker container but fails due to the supervisor not having a working internet connection.  Running network info at the CLI confirms that the system does not even see the NIC at all.

Looking at intel's website this model NUC appears to use Realtek NIC drivers. Can these be added to the generic x86 build?

agners, 2021-10-06:
@PatrickGlesner that is not the same issue. Please open a new issue.

closed 2021-10-20 by jraynes

teknology, 2023-01-13:
I have a similar issue but I'm unfamiliar with the framework of HAOS. If it's ubuntu i believe it a Realtek driver issue. If I can find out if I can run Ubuntu commands on HAOS, I can confirm the fix. Ubuntu has the same issue with Realtek Nic Cards.