home-assistant/operating-system #1598: Docker server error: "Client.Timeout exceeded while awaiting headers"

reidprichard, 2021-10-16:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [X] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [X] Updated from version 4.20
- Additional information (if accessible):

## System Health

version | core-2021.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.70
arch | x86_64
timezone | America/New_York

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4828
Installed Version | 1.15.2
Stage | running
Available Repositories | 889
Installed Repositories | 7

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 44

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.5
-- | --
update_channel | stable
supervisor_version | supervisor-2021.10.0
docker_version | 20.10.7
disk_total | 457.7 GB
disk_used | 16.2 GB
healthy | true
[... 1131 lines cut ...]
[ 1089.189913] scsi host6: usb-storage 2-1:1.0
[ 1090.241806] scsi 6:0:0:0: Direct-Access      USB      SanDisk 3.2Gen1 1.00 PQ: 0 ANSI: 6
[ 1090.242361] sd 6:0:0:0: Attached scsi generic sg3 type 0
[ 1090.244695] sd 6:0:0:0: [sdc] 240353280 512-byte logical blocks: (123 GB/115 GiB)
[ 1090.245782] sd 6:0:0:0: [sdc] Write Protect is off
[ 1090.245788] sd 6:0:0:0: [sdc] Mode Sense: 43 00 00 00
[ 1090.246781] sd 6:0:0:0: [sdc] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[ 1090.268417]  sdc: sdc1
[ 1090.272647] sd 6:0:0:0: [sdc] Attached SCSI removable disk
```
</details>

**Description of problem:**

When initially configuring Home Assistant a month or two ago, I could not get it working on anything more recent than OS version 4.20. Anything more recent would give me an error like the following:

Can't install homeassistant/amd64-addon-zwave_js:0.1.45: 500 Server Error
for http+docker://localhost/v1.41/images/create?tag=0.1.45&
fromImage=homeassistant%2Famd64-addon-zwave_js: Internal Server Error
("Get "https://registry-1.docker.io/v2/": net/http: request canceled while
waiting for connection (Client.Timeout exceeded while awaiting headers)")

At the time, I think the most recent version was 6.xx. I'm pretty sure I also tried versions in the 5.xx range, but I can't remember clearly. I don't recall exactly what the error would do, but I know it prevented me from getting Home Assistant working. 
Anyway, I've been cruising on 4.20 with no issues ever since, but Supervisor started giving me an error that the OS version was going to start causing problems - something about how it integrates with the OS. Anyway, I figured I'd bite the bullet and upgrade. The upgrade went smoothly. Immediately afterward, I went to update several HA addons. For whichever addon I try, I get a version of the above error.
I can still access HA and control things, both from the local network and externally, so I know there's not a network issue. Other than that, I don't even know where to begin to track down this error.

reidprichard, 2021-10-16:
Well damn. I must have beat my head against the wall for 2-3 hours trying to get this figured out when I set up the server, and now I find [the answer](https://www.reddit.com/r/homeassistant/comments/kgzd6p/first_time_install_issues_500_server_error_docker/) in 15 minutes of googling. Supervisor > System > "Host" pane > "IP Address" field > CHANGE button > expand IPv4, select "Static" radio button, and add ",8.8.8.8" in the DNS Servers field. Restart Core, and everything is working. 

Surely this is not classified as expected behavior, right?

agners, 2021-10-18:
By default the system should be using the DNS server which your Router provides through DHCP. It seems that whatever your router is providing is not working? Do you know what DNS server your Router is serving through DHCP?

reidprichard, 2022-01-15:
Sorry for not getting back for so long. I totally revamped my HA install (from bare metal to inside a Proxmox VM), which caused some issues with my SSH access into the machine. The in-browser terminal can't copy/paste, so it was impossible to capture the output of those commands. I was hoping to get SSH working again to track this down, but I've since made even MORE changes to my setup (moving my router to pfSense), and this issue seems to have disappeared. I'm guessing the root cause was just some funkiness with my previous router.

closed 2022-01-15 by reidprichard