home-assistant/operating-system #1602: HA Blue does not recognize zooz s2 USB stick

froz, 2021-10-20:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [x] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 2021.,10.5
- [ ] Additional information (if accessible):

## System Health

version | core-2021.10.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.70
arch | aarch64
timezone | America/Los_Angeles

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4857
Installed Version | 1.15.2
Stage | running
Available Repositories | 887
Installed Repositories | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.5
-- | --
update_channel | beta
supervisor_version | supervisor-2021.10.1
docker_version | 20.10.7
disk_total | 113.9 GB
disk_used | 10.0 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | MariaDB (2.4.0), RPC Shutdown (2.2), Samba share (9.5.1), Terminal & SSH (9.2.1), Visual Studio Code (3.6.2), Mosquitto broker (6.0.1), NGINX Home Assistant SSL proxy (3.0.2), ADB - Android Debug Bridge (0.8.0), AdGuard Home (4.1.9), Duck DNS (1.14.0), Home Assistant Google Drive Backup (0.105.2), Check Home Assistant configuration (3.9.0), Glances (0.13.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 5
views | 3
mode | storage

</details>

**Supervisor logs:**

nothing logged

**Journal logs:**

Nothing logged

**Kernel logs:**

dmesg | grep -i usb
[    0.068146] usbcore: registered new interface driver usbfs
[    0.068176] usbcore: registered new interface driver hub
[    0.068201] usbcore: registered new device driver usb
[    0.114326] gpio-447 (usb-hub-reset): hogged as output/high
[    1.191921] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    1.206895] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    1.222907] usbcore: registered new interface driver uas
[    1.227777] usbcore: registered new interface driver usb-storage
[    1.233755] usbcore: registered new interface driver usbserial_generic
[    1.240181] usbserial: USB Serial support registered for generic
[    1.323377] usbcore: registered new interface driver usbhid
[    1.328740] usbhid: USB HID core driver
[    1.419128] USB_PWR_EN: supplied by 5V
[    1.735702] dwc3-meson-g12a ffe09000.usb: USB2 ports: 2
[    1.740140] dwc3-meson-g12a ffe09000.usb: USB3 ports: 1
[    1.747838] dwc2 ff400000.usb: supply vusb_d not found, using dummy regulator
[    1.752421] dwc2 ff400000.usb: supply vusb_a not found, using dummy regulator
[    1.759552] dwc2 ff400000.usb: EPs: 7, dedicated fifos, 712 entries in SPRAM
[    1.771874] xhci-hcd xhci-hcd.4.auto: new USB bus registered, assigned bus number 1
[    1.800081] xhci-hcd xhci-hcd.4.auto: new USB bus registered, assigned bus number 2
[    1.807679] xhci-hcd xhci-hcd.4.auto: Host supports USB 3.0 SuperSpeed
[    1.814342] hub 1-0:1.0: USB hub found
[    1.821914] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    1.829983] hub 2-0:1.0: USB hub found
[    1.837582] dwc3-meson-g12a ffe09000.usb: switching to Device Mode
[    2.078519] usb 1-1: new high-speed USB device number 2 using xhci-hcd
[    2.294544] hub 1-1:1.0: USB hub found
[    2.358261] usb 2-1: new SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[    2.406424] hub 2-1:1.0: USB hub found
[   33.757751] USB_PWR_EN: disabling

**Description of problem:**

Zooz ZST10 is not recognized when plugged into any of the USB ports.  The device is not listed by lsusb and no errors or any other indication appears in any of the logs. Rebooting does not help.

The same device is recognized on an odroid C2 running dietPi as /dev/ttyUSB0 and the same using Linux on a chromebook.  Is /dev/ttyUSB* not available in the odroid kernel?

knwpsk, 2021-11-01:
Having same problem.
Forwarded link to this ticket to Zooz support so they know about the issue. Still working with them to try to troubleshoot.

agners, 2021-11-02:
From what I understand the device should appear as `/dev/ttyACMx` (see https://www.getzooz.com/downloads/zooz-z-wave-plus-radio-s2-usb-stick-zst10-manual.pdf).

@knwpsk if none of the USB port is working, it could be a hardware issue. There are similar reports on the ODROID forum: https://forum.odroid.com/viewforum.php?f=181&sid=777a8ae2aa4be255a9ef55c98e2d09f9

knwpsk, 2021-11-02:
@agners Agree, Zooz says it should show up as tty/ACMx. But it doesn't, not in any of the Blue's ports. 
I got it from Ameridroid, so I'm contacting them also. 
Very hard for me to tell what the issue is, or even where to start!

_(If anyone has had success with HA Blue and the Zooz stick, please let me know. 
If anyone has had success with HA Blue and another z-wave controller stick, please let me know.)_

knwpsk, 2021-11-02:
Update: I plugged a keyboard into the same USB port on the Blue where I had the zooz stick. (And connected a monitor via hdmi.) The keyboard works, and I can type on the console page, and the Blue's startup logging clearly recognized "Dell wired keyboard."
So, the physical port seems to be OK. That's not our problem.

basnagel, 2021-11-02:
The aluminum cover on the Blue is to wide for some usb connectors and they don't connect, (Conbee|| stick for example)

agners, 2021-11-02:
> The aluminum cover on the Blue is to wide for some usb connectors and they don't connect, (Conbee|| stick for example)

Hm, I think the power leads are slightly longer, so that would explain why the USB stick LEDs did light up.

@knwpsk try without the front plate or using a USB extension cord, this should avoid the problem @basnagel is mentioning.

Also, do you see something in the Supervisor Host logs? Those are the kernel logs and definitly should show something when a new USB device gets enumerated.

knwpsk, 2021-11-03:
I'll be damned. 
That worked!
@agners @basnagel you guys are my heroes for seeing this through. Thank you!!

Recap:
I can definitively confirm (by testing and contra-testing) that the problem I had with HA Blue detecting the Zooz S2 700 z-wave stick was that the Blue's silver faceplate is "too thick" and it doesn't let the stick plug all the way into the USB port. I removed the faceplate, plugged the stick in again, and it now appears in Blue's hardware list.

[image]
[image]
[image]
[image]

agners, 2021-11-04:
@knwpsk thanks for the update and details. Yeah that is unfortunate, I think its the second USB dongle we found now which causes problems :cry: 

@froz can you check if that is the problem in your case as well (pun not intended :sweat_smile: )

xeger, 2021-11-14:
FYI - I had this precise problem, and after a few hours of searching various forums, came across this issue. Verified that the Zooz stick is talking to my N2 after plugging it in via an extension cable. Thank you so much to the person who suggested this solution (and to the person who was thoughtful enough to post it for others to find.

(Addendum: for me, it shows up not as `/dev/ttyACM0` but as a `/dev/serial/by-id` dynamic device; no functional difference though. I'm guessing it's some devfs magic.)

agners, 2021-11-15:
@froz closing this issue since I assume your problem is also related to the case/distance. Test with an USB extension cable.

closed 2021-11-15 by agners

royliao1204, 2022-01-14:
Sorry to spam the thread, but just needed to say thanks!! I was tearing my hair out for the past hour trying to figure this out until I came across this thread. I have a machined aluminum case from ameridroid for the XU4 and the cause/solution was exactly the same!

numindast, 2022-01-27:
Came here to add my voice of support. The HA Blue custom (beautiful) case prevented fully inserting the Zooz ZST10 700 stick. I went on a hunt and found an extension and less than 5 minutes later it's up and running in HA.

AJolly, 2023-12-02:
Thanks! I thought my ha blue odroid N2+s usb ports were broken