home-assistant/operating-system #1624: This release is currently on the beta channel.

bkbartk, 2021-11-02:
**Hardware Environment**
Raspberry Pi 3
I added "gpu_mem=16" to my config.txt

I get a notification for updating the os to V6.6
but when I check the release notes is says. 
ℹ️ This release is currently on the beta channel. ℹ️

so, either the notes aren't updated correctly, or a beta release is published to the stable channel.

RCGrahamWilmott, 2021-11-02:
Same for me. Pi 4, and been asked to update to 6.6 (currently on 6.5). However the release notes say it's a beta release...

agners, 2021-11-02:
OS 6.6 is an official release. Official releases are first available through the beta channel, which was also the case with OS 6.6. It made it to the stable channel yesterday. I removed that note, thanks for the hint.

closed 2021-11-02 by agners