home-assistant/operating-system #1633: Detecting HardwareAction.REMOVE/ADD hardware /dev/bus/usb/002/XXX spamming logs

mbo18, 2021-11-05:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [x] Generic x86-64 -> Intel NUC NUC5i3RYK
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 6.5
- Additional information (if accessible):

## System Health

version | core-2021.11.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.75
arch | x86_64
timezone | Europe/Paris

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.6
-- | --
update_channel | stable
supervisor_version | supervisor-2021.10.8
docker_version | 20.10.8
disk_total | 109.3 GB
disk_used | 9.5 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), File editor (5.3.3), SSH & Web Terminal (9.0.1), Duck DNS (1.14.0), Visual Studio Code (3.6.2), Samba Backup (5.0.0), WireGuard (0.5.1), Log Viewer (0.12.0), ESPHome (2021.10.3), AdGuard Home (4.1.9), Network UPS Tools (0.8.0), chrony (2.2.1), SQLite Web (3.1.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
views | 6
mode | storage

</details>

**Supervisor logs:**

[... 110 lines cut ...]
BUSNUM: '002'
DEVNAME: /dev/bus/usb/002/098
DEVNUM: '098'
DEVPATH: /devices/pci0000:00/0000:00:14.0/usb2/2-7
DEVTYPE: usb_device
DRIVER: usb
ID_BUS: usb
ID_MODEL: 0a2a
ID_MODEL_ENC: 0a2a
ID_MODEL_ID: 0a2a
ID_PATH: pci-0000:00:14.0-usb-0:7
ID_PATH_TAG: pci-0000_00_14_0-usb-0_7
ID_REVISION: '0001'
ID_SERIAL: 8087_0a2a
ID_USB_INTERFACES: ':e00101:'
ID_VENDOR: '8087'
ID_VENDOR_ENC: '8087'
ID_VENDOR_ID: '8087'
MAJOR: '189'
MINOR: '225'
PRODUCT: 8087/a2a/1
SUBSYSTEM: usb
TYPE: 224/1/1
USEC_INITIALIZED: '1078621151'
```

agners, 2021-11-05:
It seems Linux has problems enumerating a particular device. `error -71` typically means it has problems communicating on a hardware level. Can you unplug USB devices to isolate which one it is? Maybe replace USB cables/Hubs if necessary.

mbo18, 2021-11-05:
Can it be an internal device?
All I currently have is a Conbee 2 (/dev/bus/usb/002/004), an RFLink (/dev/bus/usb/002/003) and a UPS (/dev/bus/usb/002/002). All are working fine.

agners, 2021-11-05:
One of the problematic devices seems to be `usb 2-7`. At one point it enumerates: `usb 2-7: New USB device found, idVendor=8087, idProduct=0a2a, bcdDevice= 0.01`. The vendor/product ID indicates that this is a Intel Bluetooth device:
https://devicehunt.com/view/type/usb/vendor/8087/device/0A2A

mbo18, 2021-11-05:
Ok so it is my NUC internal Bluetooth module as I did not add a USB Bluetooth module myself. What is strange is that this spamming seems to be here only since I updated to 6.6

agners, 2021-11-05:
Can you try downgrading to see if its really OS version related? Use the Terminal Add-on and the following command:

```
ha os update --version 6.5
```

jle1n0, 2021-11-06:
I lost ConBee 2 stick with my NUC platform when upgrading to version 6.6. It just complained about the missing serial device. Downgrading back to 6.5 made the system work again. With dmesg I could see the stick being recognized at least to some degree...

agners, 2021-11-08:
@jle1n0 do you get continuously ADD and REMOVE hardware log entries? If not, this is unlikely related to this issue and your problem requires a separate new issue.

mbo18, 2021-11-12:
> Can you try downgrading to see if its really OS version related? Use the Terminal Add-on and the following command:
> 
> ```
> ha os update --version 6.5
> ```

I have downgraded to 6.5 and logs are still there so not related to 6.6

agners, 2021-11-15:
I'd suspect hardware here. If you don't use the module, maybe simply removing it (if its removable :sweat_smile: ). Maybe testing with another distribution might be worth a try.

gieljnssns, 2021-11-16:
I'm also having problems with an Intel NUC and a Conbee 2.
From time to time when restarting HA
```
Logger: zigpy.application
Source: /usr/local/lib/python3.9/site-packages/zigpy/application.py:73
First occurred: 08:48:43 (1 occurrences)
Last logged: 08:48:43

Couldn't start application
```
```
Logger: homeassistant.components.zha.core.gateway
Source: components/zha/core/gateway.py:152
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 08:48:43 (1 occurrences)
Last logged: 08:48:43

Couldn't start deCONZ = dresden elektronik deCONZ protocol: ConBee I/II, RaspBee I/II coordinator
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 152, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 69, in new
    await app.startup(auto_form)
  File "/usr/local/lib/python3.9/site-packages/zigpy_deconz/zigbee/application.py", line 67, in startup
    self.version = await self._api.version()
  File "/usr/local/lib/python3.9/site-packages/zigpy_deconz/api.py", line 455, in version
    (self._proto_ver,) = await self[NetworkParameter.protocol_version]
  File "/usr/local/lib/python3.9/site-packages/zigpy_deconz/api.py", line 420, in read_parameter
    r = await self._command(Command.read_parameter, 1 + len(data), param, data)
  File "/usr/local/lib/python3.9/site-packages/zigpy_deconz/api.py", line 305, in _command
    return await asyncio.wait_for(fut, timeout=COMMAND_TIMEOUT)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```
Supervisor logs:
```
21-11-16 08:48:08 INFO (SyncWorker_9) [supervisor.docker.interface] Restarting ghcr.io/home-assistant/generic-x86-64-homeassistant
21-11-16 08:48:23 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
21-11-16 08:48:27 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-11-16 08:48:27 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-11-16 08:48:40 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2215018-if00
21-11-16 08:48:40 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/050 - None
21-11-16 08:48:43 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/051 - None
21-11-16 08:48:43 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2215018-if00
21-11-16 08:48:44 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2215018-if00
21-11-16 08:48:44 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/051 - None
21-11-16 08:48:52 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/053 - None
21-11-16 08:48:52 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2215018-if00
21-11-16 08:48:59 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
```
Host logs:
```
[492574.103114] iwlwifi 0000:00:14.3: No beacon heard and the time event is over already...
[492574.103192] wlp0s20f3: Connection to AP b8:e3:b1:81:ca:0d lost
[492575.908603] wlp0s20f3: authenticate with b8:e3:b1:81:ca:0d
[492575.914936] wlp0s20f3: send auth to b8:e3:b1:81:ca:0d (try 1/3)
[492575.968955] wlp0s20f3: authenticated
[492575.969238] wlp0s20f3: associate with b8:e3:b1:81:ca:0d (try 1/3)
[492575.999617] wlp0s20f3: RX AssocResp from b8:e3:b1:81:ca:0d (capab=0x31 status=0 aid=1)
[492576.010257] wlp0s20f3: associated
[492576.529148] iwlwifi 0000:00:14.3: No beacon heard and the time event is over already...
[492576.529224] wlp0s20f3: Connection to AP b8:e3:b1:81:ca:0d lost
[... 158 lines cut ...]
healthy | true
supported | failed to load: Unsupported
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), AdGuard Home (4.2.0), AppDaemon 4 (0.7.1), Caddy (1.6), Check Home Assistant configuration (3.9.0), Duck DNS (1.14.0), ESPHome (2021.10.3), Mosquitto broker (6.0.1), Studio Code Server (3.7.0), WireGuard (0.6.0), MariaDB (2.4.0), Samba Backup (5.0.0), InfluxDB (4.3.0), Grafana (7.3.1), Frigate NVR (2.3), SSH & Web Terminal (9.1.0), Metabase (0.3.0), Portainer (2.0.0), OwnTone (dev), Simple Scheduler (0.31), Spotify Connect (0.10.0), Glances (0.14.0), phpMyAdmin (0.6.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 18
views | 18
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

agners, 2021-11-18:
@gieljnssns hm, sudden disconnects, sometimes in a row. I'd guess something hardware related here. Maybe using a different port or/and a USB cable helps? There have been similar USB related issues reported lately with Intel NUCs, they turned out to be some type of hardware problem (see https://community.home-assistant.io/t/home-assistant-os-on-intel-nuc-no-usb-devices/355841/16).

gieljnssns, 2021-11-18:
I already tried to change USB port and extension cable, didn't help.

I can upgrade to OS 7.0.rc1, can this be helpful?

agners, 2021-11-18:
> I can upgrade to OS 7.0.rc1, can this be helpful?

Unlikely, but who knows :)

gieljnssns, 2021-11-18:
Are there any known issues with rc1?

agners, 2021-11-19:
Not that I am aware of so far.

jle1n0, 2021-12-16:
> @jle1n0 do you get continuously ADD and REMOVE hardware log entries? If not, this is unlikely related to this issue and your problem requires a separate new issue.

Ok. I just upgraded to 7.0 and faced the same issue. The system lost the whole Zigbee network behind Conbee II. I rebooted the system three times with the same result. Now I downgraded back to 6.5 and all is good again.

Shall I open a new issue or something? I can debug/collect logs tomorrow if needed.

agners, 2021-12-16:
> Now I downgraded back to 6.5 and all is good again.

So you only downgraded the OS and that made it work again?

> Shall I open a new issue or something? I can debug/collect logs tomorrow if needed.

Yes please.

jle1n0, 2021-12-17:
> > Now I downgraded back to 6.5 and all is good again.
> 
> So you only downgraded the OS and that made it work again?
> 
> > Shall I open a new issue or something? I can debug/collect logs tomorrow if needed.
> 
> Yes please.

Done: https://github.com/home-assistant/operating-system/issues/1680 I hope everything relevant is included there.

DanielTheCoder, 2021-12-20:
Experiencing the same issue as @jle1n0 with rbi

USB Cables and ports already changed.

```
21-12-20 08:59:51 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/038 - None
21-12-20 08:59:51 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2450573-if00
21-12-20 08:59:52 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2450573-if00
21-12-20 08:59:52 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/038 - None
21-12-20 09:00:03 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/039 - None
21-12-20 09:00:03 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2450573-if00
21-12-20 09:00:04 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/ttyACM0 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2450573-if00
21-12-20 09:00:04 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/039 - None
```

## System Health

version | core-2021.12.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.63-v7l
arch | armv7l
timezone | Europe/Vienna

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4802
Installed Version | 1.18.0
Stage | running
Available Repositories | 923
Installed Repositories | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.0
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 57.8 GB
disk_used | 11.8 GB
healthy | true
supported | failed to load: Unsupported
board | rpi4
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Git pull (7.13.1), Glances (0.14.0), Portainer (1.4.0), Check Home Assistant configuration (3.9.0), SQLite Web (3.2.0), RPC Shutdown (2.2), SSH & Web Terminal (10.0.0), Samba share (9.5.1), ESPHome (2021.12.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 6
-- | --
resources | 6
views | 30
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

mbo18, 2022-01-15:
Regarding my issue, I have disabled bluetooth in BIOS and logs are gone 👍 
Fortunately I don't use bluetooth

agners, 2022-01-17:
Thanks for the update. I'll close it then, if someone else encounters a similar problem, please open a new issue.

closed 2022-01-17 by agners