home-assistant/operating-system #1636: Can't make fresh install of HA: 'Updater.fetch_data' blocked from execution, no supervisor internet connection

martymarty004, 2021-11-08:
**Hardware Environment**

- [x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Appliance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
- [ ] Khadas VIM3

**Home Assistant OS release:**
- [x] Fresh installation of release 6.6
- [ ] Updated from version x.y
- Additional information (if accessible):

**Description of problem:**

I tried to make a fresh installation of Home Assistant OS 6.5 and 6.6 (x64) on my RPi 4, but I'm stuck at the "Preparing Home Assistant..." page.

The date/time doesn't seem to sync with the NTP servers, and the internet connection test fails.

I created a `CONFIG` USB key with `authorized_keys` file to enable SSH debugging, but it only worked on my current installation, on the new one the USB is detected, but nothing happens.

`time.cloudflare.com` is reachable both from my Windows device and the current HA installation (I couldn't try the new one since I don't have SSH debugging enabled).

My router (Fritz!Box 7590) is configured to use my RPi both for DHCP (its dhcp server is disabled) and DNS (I have the AdGuard Home addon), but I change those settings when I try to install Home Assistant (I have address reservation for my RPi and I use Cloudflare DNS). My provider is Dimensione (IT), it runs on the Open Fiber network and gives me IPv4 and IPv6.

Here's the full log from the web interface:

```txt
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing...
[14:44:40] INFO: Using udev information from host
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[14:44:41] INFO: Starting local supervisor watchdog...
21-08-06 14:44:45 INFO (MainThread) [__main__] Initializing Supervisor setup
21-08-06 14:44:45 INFO (MainThread) [supervisor.docker.network] Can't find Supervisor network, creating a new network
21-08-06 14:44:45 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-08-06 14:44:45 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4-64
21-08-06 14:44:45 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/aarch64-hassio-supervisor with version 2021.10.6
21-08-06 14:44:45 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
21-08-06 14:44:46 WARNING (SyncWorker_0) [supervisor.docker.interface] Can't find homeassistant/aarch64-hassio-supervisor for cleanup
21-08-06 14:44:46 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-08-06 14:44:46 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-08-06 14:44:46 INFO (MainThread) [__main__] Setting up Supervisor
21-08-06 14:44:51 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-08-06 14:44:51 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
21-08-06 14:44:51 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface io.hass.os
21-08-06 14:44:51 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.systemd1
21-08-06 14:44:51 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.login1
21-08-06 14:44:51 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.hostname1
21-08-06 14:44:51 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.timedate1
21-08-06 14:44:52 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.NetworkManager
21-08-06 14:45:55 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface de.pengutronix.rauc
21-08-06 14:45:55 INFO (MainThread) [supervisor.host.info] Updating local host information
21-08-06 14:45:56 INFO (MainThread) [supervisor.host.services] Updating service information
21-08-06 14:45:56 INFO (MainThread) [supervisor.host.network] Updating local network information
21-08-06 14:50:07 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-08-06 14:50:07 INFO (MainThread) [supervisor.host.manager] Host information reload completed
21-08-06 14:50:07 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
21-08-06 14:50:07 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
21-08-06 14:50:12 WARNING (MainThread) [supervisor.utils.whoami] Can't fetch Whoami data: Cannot connect to host whoami.home-assistant.io:443 ssl:default [Try again]
21-08-06 14:50:12 WARNING (MainThread) [supervisor.core] Can't adjust Time/Date settings: Can't fetch Whoami data: Cannot connect to host whoami.home-assistant.io:443 ssl:default [Try again]
21-08-06 14:50:12 INFO (SyncWorker_0) [supervisor.docker.interface] Found ghcr.io/home-assistant/aarch64-hassio-cli versions: [<AwesomeVersion CalVer '2021.09.0'>]
[... 87 lines cut ...]
21-08-06 14:53:56 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
21-08-06 14:53:56 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
21-08-06 14:54:26 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
21-08-06 14:54:27 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
21-08-06 14:54:57 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
21-08-06 14:54:57 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
21-08-06 14:55:27 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
21-08-06 14:55:27 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
21-08-06 14:55:57 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
21-08-06 14:55:57 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
21-08-06 14:56:27 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
21-08-06 14:56:27 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
21-08-06 14:56:57 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
21-08-06 14:56:57 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
21-08-06 14:57:27 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
21-08-06 14:57:27 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
21-08-06 14:57:57 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
21-08-06 14:57:57 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
21-08-06 14:58:27 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
21-08-06 14:58:27 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
[supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bsg/0:0:0:0 - None
21-08-06 15:03:53 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/sda1 - /dev/disk/by-id/usb-Wilk_USB_DISK_2.0_070106AB9CBF7187-0:0-part1
21-08-06 15:03:53 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/sda - /dev/disk/by-id/usb-Wilk_USB_DISK_2.0_070106AB9CBF7187-0:0
21-08-06 15:03:53 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/003 - None
```

agners, 2021-11-08:
> I created a CONFIG USB key with authorized_keys file to enable SSH debugging, but it only worked on my current installation, on the new one the USB is detected, but nothing happens.

That is strange, did you use the "Import from USB" command in the frontend? Is the Supervisor log saying something?

martymarty004, 2021-11-11:
I've finally found time to try your solution, and I'm writing this reply as a log of what I'm doing.

I've successfully gained SSH access using your command to import the config from the USB.
Furthermore, I tried to check the network configuration and I found the issue: even if I told my router not to use an IPv6 DNS server, it still advertised the RPi as one (and without AdGuard running, of course it wasn't working).

I've set the router to advertise itself as DNS for both IPv4 _and IPv6_.
I re-flashed the microSD to exclude any issues, and it should have connected properly now (the date/time is correct, yay!)

I reached the user creation page and I started importing the backup.
The backup has been imported correctly and everything is working fine now.

```txt
# ha info
arch: aarch64
channel: stable
docker: 20.10.8
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- os_agent
- haos
hassos: "6.6"
homeassistant: 2021.11.2
hostname: homeassistant
logging: info
machine: raspberrypi4-64
operating_system: Home Assistant OS 6.6
state: running
supervisor: 2021.10.8
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: Europe/Rome
```
I didn't think that the culprit could be my router, and without SSH access I couldn't actually verify the network configuration, but thanks to you, I was able to find the issue and fix it.

Now that I think about it, it would be nice if the welcome page could show the network configuration in one corner, for people like me who don't know their instance got the wrong configuration, and also for the people that connect only using homeassistant.local

closed 2021-11-11 by martymarty004

MysterHawk, 2024-04-16:
For future reference for people who will read this discussion, to enable ssh and import the public key in the ha console follow this

[https://blog.leandrotoledo.org/how-to-enable-ssh-on-home-assistant-os/](https://blog.leandrotoledo.org/how-to-enable-ssh-on-home-assistant-os/)

In tldr format a USB key in vfat with label CONFIG, and copy inside your ssh public key renamed `authorized_keys`

afterward, plug the USB key in your RPI and do a `os import`

now you should be able to connect to the homeassisant via ssh, doing:
`ssh root@homeassistant.local -p 22222`

MysterHawk, 2024-04-16:
also if you connect a monitor to the rpi4 and write inside the ha console `login` you will have a terminal available, just discovered this

MysterHawk, 2024-04-16:
I tried @martymarty004 solution but didn't work for me, you can change your network settings also using nmcli but the dns server are still not set at boot so the ntp services cannot be initialized, i give up :c

Davinatorr, 2024-04-21:
Hello! 

I just had the same issue with the information:
`[supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection`
I wanted to switch my Raspberry Pi 4 where my homeassistant was running to a Raspberry Pi 5, so i can use my RPI4 for my Klipper printer. (Initially I wanted to use the RPI5 for Klipper, but the Raspberry Pi 5 has problems with crowsnest and Camera usage, so i decided to switch the devices) 

I remembered, that i previously set username and password in the Raspberry Pi Imager, also i had SSH verification enabled. I reset everything in the Imager settings, because I don't need that for Home Assistant and installed Home Assistant fresh. Also I deleted the remembered devices in my Router Network Fritz!Box 7582, but I do not know if that made any difference. 

With the fresh installation everything works now as it is supposed to and Home Assistant has an internet connection. 

Maybe that helps in your case as well.