home-assistant/operating-system #1640: Pi Zero Image Fails to boot with pi zero w 2

mboroff, 2021-11-09:
### Describe the issue you are experiencing

A change needs to be made to the image or a different image must be created for Pi Zero W 2. None of the current pi images boot on a Pi Zero W2.

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

6.6

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. GET A PI ZERO w 2.
2. Burn the Pi 0 6.6 image
3. Burn the Pi 3 6.6 image
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
None
```

### Anything in the Host logs that might be useful for us?

```txt
None
```

### System Health information

None

### Additional information

None

agners, 2021-11-10:
Since this is a new device, this is essentially a feature request. Can you create one on https://community.home-assistant.io/c/feature-requests/13? Thanks.

pdesimonee, 2021-11-12:
indeed. New pi zero 2 won't boot the just downloaded OS with Raspberry imager. It cycles the boot phase as if had a problem on the sound card. I think a new release is needed

agners, 2021-12-29:
There is a feature request now:
https://community.home-assistant.io/t/support-raspberry-pi-zero-2-w/356814/25

Also, the `rpi3` and `rpi3-64` images of the [latest release](https://github.com/home-assistant/operating-system/releases) 7.1 (or newer) support Raspberry Pi Zero 2 W now.

closed 2021-12-29 by agners