home-assistant/operating-system #1646: Intermittently very slow, high cpu usage in Virtual box

vandad51, 2021-11-11:
### Describe the issue you are experiencing

The issue seemed to start once I installed the Nortek Husbzb combo stick (I had to install the expansion pack for Vbox). I can't confirm it was a direct cause though. Access to my HA instance from my ios app made me aware to the problem as accessibility to HA was hit and miss. As well as my internet being very slow as it's redirected through AdGaurd. I checked my Surface tablet running the VM, and it was at a steady 40-60% usage. Yet when I check the supervisor, the cpu usage charts report normal 2-10% usage. I can try to access backups with CLI but it's extremely slow [(I'll try to attach a video)](https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FATy-HtI6vv0TzIVSMIFAun2BCeZjAQC5HSTS-ycqCkcgaipa5vYgSc48%2F%24%7Bf%7D%3Fo%3DAoshrMW519hlIADDHS0uMEhPjheVW3njgh9fpSKcs08C%26v%3D1%26x%3D3%26a%3DCAogrXeDfJ_5JxrXvZRV_YBwNqVZM6wRbrxNxQTMBam2czsSdhDAje2G0S8YwJ3o2tovIgEAKgkC6AMA_0zVY9pSBIEJ5mNaBCBJzjxqJdVTr8dmfN_USECMlWumlYo5eDryAiIbLGG6qw-gEhSSvP7cLlJyJWYm25GlhGNYHcNvJSivF6a_gTBFYpXpirgccAn7uCr_l39J-Qk%26e%3D1639257345%26fl%3D%26r%3D14085061-6495-4CD3-B58B-5A2959EAEE39-1%26k%3D%24%7Buk%7D%26ckc%3Dcom.apple.largeattachment%26ckz%3D538DE0E2-391E-4C23-9197-3B0C0ADB61BB%26p%3D41%26s%3DX3aoG9QX5DhpgIbVBeKlL8rKloA&uk=sNZHsfO0g0cqaCuGjTOoRQ&f=IMG_8477.MOV&sz=141656223) Also some errors that came up while trying to restore a backup to a few days ago.
[image]

I'm running HAos 11.1, Can't see supervisor version right now as it's still stuck restoring a backup. VirtualBox 6.1.26. I should also say that i tried to update vbox a week ago and home assistant would not start up with vbox showing a error. I downgraded back to the current version that was working and it was back to normal.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

at this point, the problem is constant. Sometimes it cools off and the CPU load drops back to normal, but then within 5 minutes its back to 40-60% usage 

### Anything in the Supervisor logs that might be useful for us?

```txt
I can't access logs atm.
```

### Anything in the Host logs that might be useful for us?

```txt
I can't access logs atm.
```

### System Health information

I can't access logs atm.

agners, 2021-11-12:
> What operating system image do you use?
> 
> generic-x86-64 (Generic UEFI capable x86-64 systems)

That image is meant for bare-metal machines. Try using the OVA (open virtualization appliance) images.

vandad51, 2021-11-12:
Ok sorry. I see haos_ova-6.1.vdi is that what you’re referring to?

vandad51, 2021-11-12:
I did seem to find a correlation to what starts the high CPU usage. Below is my supervisor log. After doing a complete system reboot, it runs fines until it says "Starting system checks with state CoreState.RUNNING". It seems like the high usage starts as soon as the first check happens, and then it stays high. This is while my tablet is idle at home, me not tinkering on it.

21-11-12 11:52:47 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-11-12 11:52:47 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-11-12 11:53:18 INFO (MainThread) [supervisor.host.info] Updating local host information
21-11-12 11:53:19 INFO (MainThread) [supervisor.host.services] Updating service information
21-11-12 11:53:19 INFO (MainThread) [supervisor.host.network] Updating local network information
21-11-12 11:53:19 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-11-12 11:53:19 INFO (MainThread) [supervisor.host.manager] Host information reload completed
21-11-12 11:54:16 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-11-12 12:24:16 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-11-12 12:52:47 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
21-11-12 12:52:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
21-11-12 12:52:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.PLUGIN
21-11-12 12:52:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
21-11-12 12:52:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
21-11-12 12:52:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
21-11-12 12:52:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.CORE
21-11-12 12:52:47 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-11-12 12:52:47 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-11-12 12:52:47 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-11-12 12:52:47 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-11-12 12:52:47 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-11-12 12:53:05 INFO (MainThread) [supervisor.backups.manager] Found 11 backup files
21-11-12 12:53:07 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-11-12 12:54:17 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-11-12 13:24:18 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-11-12 13:52:47 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
21-11-12 13:52:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
21-11-12 13:52:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.PLUGIN
21-11-12 13:52:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
21-11-12 13:52:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
21-11-12 13:52:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
21-11-12 13:52:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.CORE
21-11-12 13:52:47 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-11-12 13:52:47 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-11-12 13:52:47 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-11-12 13:52:47 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-11-12 13:52:47 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-11-12 13:53:33 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
21-11-12 13:53:33 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
21-11-12 13:53:39 INFO (MainThread) [supervisor.store] Loading add-ons from store: 62 all - 0 new - 0 remove
21-11-12 13:53:39 INFO (MainThread) [supervisor.store] Loading add-ons from store: 62 all - 0 new - 0 remove
21-11-12 13:54:19 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-11-12 13:59:59 INFO (MainThread) [supervisor.host.info] Updating local host information
21-11-12 14:00:00 INFO (MainThread) [supervisor.host.services] Updating service information
21-11-12 14:00:00 INFO (MainThread) [supervisor.host.network] Updating local network information
21-11-12 14:00:00 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-11-12 14:00:00 INFO (MainThread) [supervisor.host.manager] Host information reload completed
[... cut ...]
21-11-12 14:52:48 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-11-12 14:52:48 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-11-12 14:52:48 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-11-12 14:53:07 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-11-12 14:54:21 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token

vandad51, 2021-11-13:
[image]. I did find this interesting though. Not sure if that’s home assistant using the CPU or Linux

agners, 2021-11-13:
> Ok sorry. I see haos_ova-6.1.vdi is that what you’re referring to?

My point is, you claim you are using `generic-x86-64`. You should not be using the `generic-x86-64` images for virtualization, but the `ova` images. The complete system health should tell what image you are exactly using.

vandad51, 2021-12-14:
My issue ended up being a Vm issue with hypervisor being enabled on my PC, and vbox out dated by one version. I disabled hypervisor and updated vbox and my problems have stayed away now for several weeks now.

closed 2021-12-14 by vandad51

agners, 2021-12-14:
Thanks for the update! I guess with hypervisor you mean Microsoft's Hyper-V?

vandad51, 2021-12-14:
Oh yes. Sorry, I’ve seen hypervisor around the web lately and forgot it’s called Hyper-V