home-assistant/operating-system #1647: pi3 HassOs. HA goes offline, SSH continues to work

gumileeev, 2021-11-11:
### Describe the issue you are experiencing

pi3 HassOs. HA goes offline, SSH continues to work
I have 2 boards pi3 b v1.2 and pi3 b +. HassOs 32bit.
I always install updates promptly. For the last month (possibly already 2), HA has been regularly going offline. This can happen every 30 minutes or work for about 6 days. At the same time, SSH continues to work.
I tried to take a new SD card and restore the backup. It did not help.
Command: ha sulervisor info
Result: state: started
Commands: supervisor repair, reload - do not help
The system comes back to life after the command: ha host reboot
From the RAM usage sensor, you can see that the amount of free memory before freezing becomes less than 10%. In my image, I marked 2 consecutive system failures with arrows.
[screen](https://skr.sh/sAyXHhulGUX)
I post "ha host logs" after a system crash: [code](https://pastebin.com/MixjmNeZ)
Here "ha host logs" after restarting the system with the command "ha host reboot": [code](https://pastebin.com/apy8NvtK)
I consulted in the chat [link telegram](https://t.me/SprutAI_HomeAssistant) and found out that at least 3 people have a similar problem on pi3 HassOs for about the last month.
I'm not a command line expert, but it seems to me that there might be a problem here:
```
[ 1275.666813] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=924adc                                                                                                                                                             80b7621de53e30c12d912fcc973f8933a00cb5e149bba86ae4a120a7ad,mems_allowed=0,global                                                                                                                                                             _oom,task_memcg=/docker/bcc6534bc3d3594f48b1ca01d78928ccef472934d67eb8fd68ea95bc                                                                                                                                                             825a7749,task=python3,pid=5172,uid=0
[ 1275.666981] Out of memory: Killed process 5172 (python3) total-vm:765708kB, a                                                                                                                                                             non-rss:499104kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:734kB oom_score_ad                                                                                                                                                             j:0
[ 1275.947066] oom_reaper: reaped process 5172 (python3), now anon-rss:0kB, file                                                                                                                                                             -rss:0kB, shmem-rss:0kB
[ 1279.837902] audit: type=1701 audit(1636631029.862:224): auid=4294967295 uid=0                                                                                                                                                              gid=0 ses=4294967295 subj==unconfined pid=114 comm="systemd-journal" exe="/usr/                                                                                                                                                             lib/systemd/systemd-journald" sig=6 res=1
[ 1280.788898] systemd-coredump[6820]: Process 114 (systemd-journal) of user 0 d                                                                                                                                                             umped core.
[ 1280.881327] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1280.882930] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1280.923177] systemd[1]: systemd-journald.service: Main process exited, code=d                                                                                                                                                             umped, status=6/ABRT
[ 1280.923643] systemd[1]: systemd-journald.service: Failed with result 'watchdo                                                                                                                                                             g'.
[ 1280.933461] systemd[1]: systemd-journald.service: Scheduled restart job, rest                                                                                                                                                             art counter is at 1.
[ 1280.944135] systemd[1]: Stopping Flush Journal to Persistent Storage...
[ 1281.022868] systemd[1]: systemd-journal-flush.service: Control process exited                                                                                                                                                             , code=exited, status=1/FAILURE
[ 1281.023435] systemd[1]: systemd-journal-flush.service: Failed with result 'ex                                                                                                                                                             it-code'.
[ 1281.027122] systemd[1]: Stopped Flush Journal to Persistent Storage.
[ 1281.029177] systemd[1]: Stopped Journal Service.
```

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

6.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

I cannot reproduce the problem, could not identify the dependency

### Anything in the Supervisor logs that might be useful for us?

```txt
В файлах home-assistant.log и home-assistant.log.1 проблема не видна
```

### Anything in the Host logs that might be useful for us?

```txt
https://pastebin.com/MixjmNeZ
```

### System Health information

## System Health

version | core-2021.11.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.17-v7
arch | armv7l
timezone | Europe/Moscow

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.16.0
Stage | running
Available Repositories | 901
Installed Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.6
-- | --
update_channel | stable
supervisor_version | supervisor-2021.10.8
docker_version | 20.10.8
disk_total | 28.6 GB
disk_used | 7.8 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), MariaDB (2.4.0), Home Assistant Google Drive Backup (0.105.2), File editor (5.3.3), Mosquitto broker (6.0.1), ESPHome (2021.10.3), Zigbee2mqtt (1.22.0-2), Terminal & SSH (9.2.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 1
views | 6
mode | yaml

</details>

agners, 2021-11-12:
It seems that the out-of-memory (OOM) killer kicks in on your system. Basically the system is running out of memory. This could be because you have too many Add-ons etc installed, which then at a certain peak load suddenly causes an issue. Or your system experiences some type of memory bug. In this case, I would check the memory usage in Supervisor and see if you can see a pattern for which part of the system memory usage grows.

gumileeev, 2021-11-12:
It seems to me that if PI3 is officially supported by HassOS, then the developer knows that there is only 1 gigabyte of memory in the system, and apparently it is assumed that it should work stably. Previously, there was no such problem and everything worked stably. I would be thinking about the hardware problem if I didn't have two almost identical computers, with similar configurations (2 different houses), and the problem appeared at about the same time.
I can't find a connection with the freezes, the system may work for a week, or it may not work for an hour. Moreover, one system is installed in my house, which I visit very rarely, nothing happens in it, and no one does anything on the server, but it still freezes. I would like to find a solution to the problem, for my part I am ready to carry out all the necessary tests.
Today I have access to statistics for 5 days. I marked the moments with arrows - when the system turns off: https://skr.sh/sAzv7pRe52l
It would be nice if the system at least started independently (ha core start), it would not be so painful, although this is not a solution to the problem.

dreimer1986, 2021-11-12:
Looks very much like a memory problem for me. It never goes higher than your arrows mark the reboots and always there is the problem. RPi 3 might be supported indeed, but the more it is logging and doing, the more RAM you need. The database does increase in size, too. Any and every update can cause these problems, either. OSes do not tend to become more lightweight these days. Platforms with 1G of RAM are limited and it's irrelevant if stuff happens or not, HA is logging and working with the input it gets.

agners, 2021-11-13:
> It would be nice if the system at least started independently (ha core start), it would not be so painful, although this is not a solution to the problem.

Currently there is a problem in Core where the exit code does not get passed back to the Supervisor. With this change the problem should be resolved: https://github.com/home-assistant/core/pull/59545

But in the end you should indeed make sure that OOM situations do not happen.

Can you correlate the time with something in your log? Maybe some specific automation which triggers at that time? I've heard from other instances where Home Assistant Google Drive Backup causes problems.

What @dreimer1986 says is of course true, systems tend to take more memory as they grow. So you may have to save memory somewhere else, or move to a system with more memory.

gumileeev, 2021-11-14:
> > 
> Can you correlate the time with something in your log? Maybe some specific automation which triggers at that time? I've heard from other instances where Home Assistant Google Drive Backup causes problems.
> 
I have been able to reproduce the problem several times. I have 258 objects, most of them are states of router integration devices. the main sensors are air quality, temperature, humidity, several lamps, sockets. About 20 automations associated with the presence and activation of heating, water supply start-up. I tried to download the "history" section 3 times with all the sensors in 3 days, each time it stopped the core. Moreover, the history of one day is loaded in less than 10 seconds, 2 days are loaded in about a minute. 3 days out of 3 attempts never loaded.

agners, 2021-11-15:
I'd guess data get prepared in memory, it seems that the amount of states your system has makes Core using too much memory and pushes the system over the edge. The above mentioned PR has been merged, so the next release should restart Core automatically.

I remember that my history was quite slow at one point, and I ended up using the `exclude` feature of the [recorder integration](https://www.home-assistant.io/integrations/recorder/). Just remove entities where you don't care about the history. Especially high frequent ones are good candidates to save storage and improve performance.

Maybe there can be improved something in Core to improve the situation, not sure. But this is the wrong place to discuss this, it would need to be discussed in the HA Core context.

closed 2021-11-15 by github-actions[bot]