home-assistant/operating-system #1650: generic x86-64 not working on vostro 3360 (ar8161)

Cyanogenbot, 2021-11-13:
### Describe the issue you are experiencing

This laptop uses an atheros ar8161 ethernet chipset along its i7 3417u which doesn't seem to be supported in the latest distribution, Is it possible for this chipset to be added to the next release? 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. install on a atheros ar8161
2. doesnt show up in router

### Anything in the Supervisor logs that might be useful for us?

```txt
No cant be added due to setup restrictions
```

### Anything in the Host logs that might be useful for us?

```txt
No cant be added due to setup restrictions
```

### System Health information

No cant be added due to setup restrictions

### Additional information

https://wiki.linuxfoundation.org/networking/alx

Cyanogenbot, 2021-11-15:
awesome, thanks!

agners, 2021-11-15:
@Cyanogenbot to test if that did the trick you can get a development build from https://os-builds.home-assistant.io/8.0.dev20211114/.

Cyanogenbot, 2021-11-15:
It works perfectly! thanks

closed 2021-11-15 by Cyanogenbot