home-assistant/operating-system #1655: iwlwifi firmware bug causing wifi dropouts

IceEyz, 2021-11-24:
### Describe the issue you are experiencing

There is a known Intel firmware bug that causes dropouts on Intel WiFi:
https://bugzilla.kernel.org/show_bug.cgi?id=203709

Arch made a patch that should workaround the Intel bug:
https://aur.archlinux.org/pkgbase/linux-beacon

Intel may or may not have fixed the bug in more recent firmware as well, but I wouldn't bet on it.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Connect to a WiFi infrastucture AP with a Intel WiFi NIC.
2. Use the connection for minutes, hours, or days and look for previously mentioned logs in dmesg.
3. If it does, only a reboot will fix the problem, reloading iw kernel modules does not.

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
[49031.488090] Intel(R) Wireless WiFi driver for Linux
[49031.508738] iwlwifi 0000:02:00.0: loaded firmware version 29.198743027.0 3168-29.ucode op_mode iwlmvm
[49031.528949] iwlwifi 0000:02:00.0: Detected Intel(R) Dual Band Wireless AC 3168, REV=0x220
[49031.576983] ieee80211 phy1: Selected rate control algorithm 'iwl-mvm-rs'
[49031.580702] thermal thermal_zone2: failed to read out thermal zone (-61)
[49031.581509] iwlwifi 0000:02:00.0 wlp2s0: renamed from wlan0
[49052.277325] wlp2s0: authenticate with 00:1d:c9:07:69:6f
[49052.281454] wlp2s0: send auth to 00:1d:c9:07:69:6f (try 1/3)
[49052.287161] wlp2s0: authenticated
[49052.287386] iwlwifi 0000:02:00.0 wlp2s0: disabling HT/VHT/HE as WMM/QoS is not supported by the AP
[49052.287494] wlp2s0: associate with 00:1d:c9:07:69:6f (try 1/3)
[49052.292418] wlp2s0: RX AssocResp from 00:1d:c9:07:69:6f (capab=0x11 status=0 aid=1)
[49052.293037] wlp2s0: associated
[49052.331686] IPv6: ADDRCONF(NETDEV_CHANGE): wlp2s0: link becomes ready
[49054.715196] wlp2s0: Connection to AP 00:1d:c9:07:69:6f lost
[49063.494074] wlp2s0: authenticate with 00:1d:c9:07:69:6f
[49063.497496] wlp2s0: send auth to 00:1d:c9:07:69:6f (try 1/3)
[49063.513898] wlp2s0: authenticated
[49063.514110] iwlwifi 0000:02:00.0 wlp2s0: disabling HT/VHT/HE as WMM/QoS is not supported by the AP
[49063.514325] wlp2s0: associate with 00:1d:c9:07:69:6f (try 1/3)
[49063.531797] wlp2s0: RX AssocResp from 00:1d:c9:07:69:6f (capab=0x11 status=0 aid=1)
[49063.532600] wlp2s0: associated
[49072.328009] wlp2s0: Connection to AP 00:1d:c9:07:69:6f lost
[49079.622427] wlp2s0: authenticate with 00:1d:c9:07:69:6f
[49079.625547] wlp2s0: send auth to 00:1d:c9:07:69:6f (try 1/3)
[49079.638530] wlp2s0: authenticated
[49079.638736] iwlwifi 0000:02:00.0 wlp2s0: disabling HT/VHT/HE as WMM/QoS is not supported by the AP
[49079.640082] wlp2s0: associate with 00:1d:c9:07:69:6f (try 1/3)
[49079.645977] wlp2s0: RX AssocResp from 00:1d:c9:07:69:6f (capab=0x11 status=0 aid=1)
[49079.646546] wlp2s0: associated
[49087.789862] wlp2s0: Connection to AP 00:1d:c9:07:69:6f lost
[49095.198314] wlp2s0: authenticate with 00:1d:c9:07:69:6f
[49095.201768] wlp2s0: send auth to 00:1d:c9:07:69:6f (try 1/3)
[49095.206143] wlp2s0: authenticated
[49095.206345] iwlwifi 0000:02:00.0 wlp2s0: disabling HT/VHT/HE as WMM/QoS is not supported by the AP
[49095.207816] wlp2s0: associate with 00:1d:c9:07:69:6f (try 1/3)
[49095.212403] wlp2s0: RX AssocResp from 00:1d:c9:07:69:6f (capab=0x11 status=0 aid=1)
[49095.213604] wlp2s0: associated
[49095.267269] wlp2s0: Connection to AP 00:1d:c9:07:69:6f lost
[49104.031043] wlp2s0: authenticate with 00:1d:c9:07:69:6f
[49104.035729] wlp2s0: send auth to 00:1d:c9:07:69:6f (try 1/3)
[49104.039616] wlp2s0: authenticated
[49104.039849] iwlwifi 0000:02:00.0 wlp2s0: disabling HT/VHT/HE as WMM/QoS is not supported by the AP
[49104.040726] wlp2s0: associate with 00:1d:c9:07:69:6f (try 1/3)
[49104.047101] wlp2s0: RX AssocResp from 00:1d:c9:07:69:6f (capab=0x11 status=0 aid=1)
[49104.047767] wlp2s0: associated
[49104.091625] IPv6: ADDRCONF(NETDEV_CHANGE): wlp2s0: link becomes ready
**[49104.649824] iwlwifi 0000:02:00.0: No beacon heard and the time event is over already...**
[49104.649848] wlp2s0: Connection to AP 00:1d:c9:07:69:6f lost
[49105.802195] wlp2s0: authenticate with 00:1d:c9:07:69:6f
[49105.805387] wlp2s0: send auth to 00:1d:c9:07:69:6f (try 1/3)
[49105.809951] wlp2s0: authenticated
[49105.810151] iwlwifi 0000:02:00.0 wlp2s0: disabling HT/VHT/HE as WMM/QoS is not supported by the AP
[49105.810663] wlp2s0: associate with 00:1d:c9:07:69:6f (try 1/3)
[49105.829809] wlp2s0: RX AssocResp from 00:1d:c9:07:69:6f (capab=0x11 status=0 aid=1)
[49105.830406] wlp2s0: associated
[49108.680065] wlp2s0: Connection to AP 00:1d:c9:07:69:6f lost
[49115.868650] wlp2s0: authenticate with 00:1d:c9:07:69:6f
[49115.871830] wlp2s0: send auth to 00:1d:c9:07:69:6f (try 1/3)
[49115.878156] wlp2s0: authenticated
```

### System Health information

## System Health

version | core-2021.11.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.75
arch | x86_64
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4982
Installed Version | 1.17.2
Stage | running
Available Repositories | 910
Installed Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.6
-- | --
update_channel | stable
supervisor_version | supervisor-2021.10.8
docker_version | 20.10.8
disk_total | 116.7 GB
disk_used | 36.5 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Check Home Assistant configuration (3.9.0), MariaDB (2.4.0), Mosquitto broker (6.0.1), InfluxDB (4.3.0), Grafana (7.3.1), Studio Code Server (3.7.0), Log Viewer (0.12.1), SSH & Web Terminal (9.1.1), Network UPS Tools (0.9.0), Let's Encrypt (4.11.0), UniFi Controller (0.19.3), Node-RED (10.1.1), File editor (5.3.3), Samba share (9.5.1), Zigbee2mqtt (1.22.0-3), PHEV (1.3)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 1
views | 14
mode | storage

</details>

agners, 2021-11-28:
Hm, this seems to be a work around for now by setting a kernel module parameter. You should be able to set that parameter by adding `iwlwifi.beacon_timeout=256` to `cmdline.txt`. The file is on the first partition of your installation media, or in `/mnt/boot/cmdline.txt` if you have shell access to the OS.

IceEyz, 2021-12-09:
Thanks for your reply and sorry for my delayed response. If I'm not mistaken, that would only work when this patch to the driver is made:

https://github.com/mikezackles/linux-beacon-pkgbuild/commit/8b6f0781a063405df78d6e31eabb12e60c51c814 (beacon_timeout.patch)

Could we apply that patch?

agners, 2021-12-09:
Hm, I see. Yeah should be possible.

From what I can tel its a workaround though. The hardcoded default today is 16, which sounds like a lot already. I wonder if increasing to 256 leads to anything reliable/useable.

IceEyz, 2021-12-09:
So do I, but there are quite a few positive reports. It's probably a workaround for some blocking issue in the driver that happens under rather specific circumstances. FYI, I would have tried to change wifi settings on the AP to see if shorter beacon intervals would make a difference. The AP is a car however, so there's nothing to configure.

agners, 2021-12-10:
> The AP is a car however, so there's nothing to configure.

:thinking:

agners, 2021-12-13:
@IceEyz the last nightly build comes with the patch included. You can test it by using the dev channel and updating to it:

```
ha su options --channel dev
ha su reload
ha os update
```

agners, 2022-01-12:
This has been backported to 7.1 as well. Closing.

closed 2022-01-12 by agners