home-assistant/operating-system #1656: UPS controlled via NUT crash RaspPi3 if connected when booting

MatteGary, 2021-11-29:
### Describe the issue you are experiencing

Hi, I'm experiencing the following issue: if my Raspberry Pi 3 boots up with the ups connected through USB, it will not boot completely and HomeAssistant will not start. If I then unplug USB and perform a power cycle, everything works fine.
What kind of info can I collect to help you in investigating this? How can I do that?
NUT Addon is collecting data correctly from the UPS.

System info:
Network UPS Tools v0.9.0
Home Assistant OS v 6.6
Core v core-2021.11.4
Supervisor v supervisor-2021.10.8

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 6.6

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Supervisor Addon Network UPS Tools (https://github.com/hassio-addons/addon-nut) installed
2. UPS connector via USB directly to Raspberry Pi3
3. Cut power to Raspberry (to simulate UPS running out of battery)
4. Restore Power
It may happen even if Raspberry is gracefully shutdown, haven't tried.

Result is: Home Assistant won't start, it is stuck somewhere. 

### Anything in the Supervisor logs that might be useful for us?

```txt
Cannot collect is since after a startup they are cleared.
```

### Anything in the Host logs that might be useful for us?

```txt
Cannot collect is since after a startup they are cleared.
```

### System Health information

## System Health

version | core-2021.11.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.17-v7
arch | armv7l
timezone | Europe/Rome

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4783
Installed Version | 1.17.2
Stage | running
Available Repositories | 919
Installed Repositories | 35

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.6
-- | --
update_channel | stable
supervisor_version | supervisor-2021.10.8
docker_version | 20.10.8
disk_total | 28.6 GB
disk_used | 9.4 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), AppDaemon (4.1.0), Auto Backup Hassio (1.0.2), Check Home Assistant configuration (3.9.0), File editor (5.3.3), Let's Encrypt (4.11.0), Mosquitto broker (6.0.1), Terminal & SSH (9.2.1), AppDaemon 4 (0.7.0), Network UPS Tools (0.9.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 5
-- | --
resources | 29
views | 20
mode | storage

</details>

agners, 2021-12-03:
Is there anything on the HDMI screen when the UPS is plugged in?

It could be that U-Boot has problems enumerating the device. I had this previously, especially for buggy USB devices. Can you try HAOS 7.0.rc1?

MatteGary, 2021-12-15:
I've just installed version 7.0 since it has been released, but I think the problem is still there.
I have to take a monitor and attach it to the raspberry in my garage, will do that as soon as possible.