home-assistant/operating-system #1659: HASOS install screen freezing in the eternity))

vpaulos, 2021-12-04:
### Describe the issue you are experiencing

1. install 7.0 release to intel nuc d53427
2. after reboot systemis in "preparing HA, this can take up to 20 minutes)" state
3. That`s all )) I was wait up to 4 hours, and nothing new.

in the terminal screen nothing happened too, - "ha>" state. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.0

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. install 7.0 release to intel nuc d53427
2. after reboot systemis in "preparing HA, this can take up to 20 minutes)" state
3. That`s all )) I was wait up to 4 hours, and nothing new.
4. in the terminal screen nothing happened too, - "ha>" state. 
[image]

### Anything in the Supervisor logs that might be useful for us?

```txt
cant access to supervisor menu or logs
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System Health information

[image]

vpaulos, 2021-12-04:
in 6.6 release - same problem

vpaulos, 2021-12-04:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing...
[07:43:35] INFO: Using udev information from host
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[07:43:35] INFO: Starting local supervisor watchdog...
21-12-04 07:43:37 INFO (MainThread) [__main__] Initializing Supervisor setup
21-12-04 07:43:37 INFO (MainThread) [supervisor.docker.network] Can't find Supervisor network, creating a new network
21-12-04 07:43:37 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-12-04 07:43:37 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: generic-x86-64
21-12-04 07:43:37 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/amd64-hassio-supervisor with version 2021.10.8
21-12-04 07:43:37 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
21-12-04 07:43:37 WARNING (SyncWorker_0) [supervisor.docker.interface] Can't find homeassistant/amd64-hassio-supervisor for cleanup
21-12-04 07:43:37 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-12-04 07:43:37 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-12-04 07:43:37 INFO (MainThread) [__main__] Setting up Supervisor
21-12-04 07:43:38 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-12-04 07:43:38 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
21-12-04 07:43:38 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface io.hass.os
21-12-04 07:43:38 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.systemd1
21-12-04 07:43:38 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.login1
21-12-04 07:43:38 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.hostname1
21-12-04 07:43:38 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.timedate1
21-12-04 07:43:38 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.NetworkManager
21-12-04 07:43:38 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface de.pengutronix.rauc
21-12-04 07:43:38 INFO (MainThread) [supervisor.host.info] Updating local host information
21-12-04 07:43:38 INFO (MainThread) [supervisor.host.services] Updating service information
21-12-04 07:43:38 INFO (MainThread) [supervisor.host.network] Updating local network information
21-12-04 07:43:38 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-12-04 07:43:38 INFO (MainThread) [supervisor.host.manager] Host information reload completed
21-12-04 07:43:38 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
21-12-04 10:43:39 INFO (SyncWorker_0) [supervisor.docker.interface] Found ghcr.io/home-assistant/amd64-hassio-cli versions: [<AwesomeVersion CalVer '2021.09.0'>]
21-12-04 10:43:39 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-cli with version 2021.09.0
21-12-04 10:43:39 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
21-12-04 10:43:40 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/amd64-hassio-cli with version 2021.09.0 - 172.30.32.5
21-12-04 10:43:40 INFO (SyncWorker_0) [supervisor.docker.interface] Found ghcr.io/home-assistant/amd64-hassio-dns versions: [<AwesomeVersion CalVer '2021.06.0'>]
21-12-04 10:43:40 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-dns with version 2021.06.0
21-12-04 10:43:40 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-12-04 10:43:40 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/amd64-hassio-dns with version 2021.06.0 - 172.30.32.3
21-12-04 10:43:40 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-12-04 10:43:40 INFO (SyncWorker_0) [supervisor.docker.interface] Found ghcr.io/home-assistant/amd64-hassio-audio versions: [<AwesomeVersion CalVer '2021.07.0'>]
21-12-04 10:43:40 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-audio with version 2021.07.0
21-12-04 10:43:40 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
21-12-04 10:43:41 INFO (SyncWorker_0) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/amd64-hassio-audio with version 2021.07.0 - 172.30.32.4
21-12-04 10:43:41 INFO (SyncWorker_0) [supervisor.docker.interface] Found ghcr.io/home-assistant/amd64-hassio-observer versions: [<AwesomeVersion CalVer '2021.10.0'>]
21-12-04 10:43:41 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-observer with version 2021.10.0
21-12-04 10:43:41 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
21-12-04 10:43:41 INFO (SyncWorker_0) [supervisor.docker.observer] Starting Observer ghcr.io/home-assistant/amd64-hassio-observer with version 2021.10.0 - 172.30.32.6
[... cut ...]
21-12-04 13:03:08 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-12-04 13:03:08 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-12-04 13:03:08 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-12-04 13:03:08 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-12-04 13:03:08 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
(this can take up to 20 minutes)

vpaulos, 2021-12-06:
After 5 installs system will work, but not properly!
1. cant install nothing from app store
2. system work very slowly

so what a fuck&??

agners, 2021-12-06:
@vpaulos on first boot the system downloads the Core container from the GitHub registry. That process can take a while. In the logs I don't see anything particularly special, it seems the process got started:

```
21-12-04 11:03:08 INFO (SyncWorker_0) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/generic-x86-64-homeassistant:landingpage to ghcr.io/home-assistant/generic-x86-64-homeassistant:2021.11.5
```

Since the container image is a bit more than 1GB, this process can take quite a while depending on your internet connection.

>  1.  cant install nothing from app store

I guess you mean Add-ons? This is probably related the the problem above. Is your internet connection to GitHub slow?

> 2. system work very slowly

This shouldn't be the case with an Intel machine. Can you be a bit more specific what is slow exactly? How is CPU load (in Supervisor -> System tab?)

closed 2022-03-13 by github-actions[bot]