home-assistant/operating-system #1660: Updating 6.6 to 7.0 on odroid N2+ breaks wifi

stlhrt, 2021-12-11:
### Describe the issue you are experiencing

After upgrading 6.6 to 7.0 on my Odroid N2+ my Odroid wifi dongle no longer works.
It's a RTL8811AU based device.

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

7.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. boot
2. list network adapters using  nmcli

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

agners, 2021-12-11:
> Anything in the Host logs that might be useful for us?
>
> No

At a minimum, there should be a USB device notification. If that is not the case, then your USB ports are broken.

stlhrt, 2021-12-11:
Rollback fixed them then ;-)

stlhrt, 2021-12-11:
lsusb listed device IDs without any names.

lodyh, 2021-12-11:
> […](#)

Restore from backup created during the update for me didn't fix it unfortunately - Host seems to still be at 7.0. My wlan0 is still MIA no matter what Ha OS i rollback to... did you do anything additional?

stlhrt, 2021-12-11:
No just regular ha os update --version 6.6 .
Lucky to be able connect via cable.

agners, 2021-12-29:
Home Assistant OS 7.1 which fixes this issue is now on the beta channel (and will likely be promoted to the stable channel in a few days). If you want to test today, you can update it using the following Terminal commands:

```
ha su options --channel beta
ha su reload
ha os update
```

closed 2021-12-29 by agners

stlhrt, 2022-01-03:
:+1:  works fine now.