home-assistant/operating-system #1661: Updating 6.6 to 7.0 on Intel NUC NUC6CAYH - device is offline

Fierys, 2021-12-11:
### Describe the issue you are experiencing

1. I had Home Assistant v. 6.6 working on Intel NUC6CAYH
2. After upgrading to 7.0 - I got a broken system.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. install Home Assistant in. 6.6 on Intel NUC6CAYH
2. Upgrade to 7.0
3. System not available
...

### Anything in the Supervisor logs that might be useful for us?

```txt
I cannot access the log files
```

### Anything in the Host logs that might be useful for us?

```txt
I cannot access the log files
```

Fierys, 2021-12-11:
Are there any chances to recover? or already get my Google Backup?

agners, 2021-12-12:
It depends what the exact reason is of the failure. Is there anything showing on the screen (connected to the device directly)?

Fierys, 2021-12-12:
I didn't have the opportunity to connect the system to a monitor to see what was displayed there. Just after the update, the network connection was lost (www, ssh, samba). I decided not to touch anything and wait, especially since I have backups.
Now I reseted the server and everything worked ...
For HA this is the first time this behavior during an update...

In general, the alarm turned out to be false, you need a little patience!
Thanks!

closed 2021-12-12 by Fierys