home-assistant/operating-system #1662: Latest update version 7.0 fails to update from 6.6

NVader2000, 2021-12-11:
### Describe the issue you are experiencing

When I click on update now it appears to go through the steps to update but never completes. Logs do not appear to show anything wrong.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Clicked Update now.
2. Waiting, waiting, waiting...

### Anything in the Supervisor logs that might be useful for us?

```txt
Not that I can see
```

### Anything in the Host logs that might be useful for us?

```txt
Not that I can see
```

### System Health information

  Not that I can see

### Additional information

  Not that I can see

vikdb, 2021-12-11:
Have the same problem, keeps failing to update and no errors in supervisor log.
( also cant disable the backup, i have a feeling its backup related )

drMaxxi, 2021-12-11:
Here too

jlt24, 2021-12-11:
When I try to install again I get the message
"Version 7.0 is already installed"

Tykwondo1, 2021-12-11:
After several try with my PC, I tried with my ipad app and it worked

NVader2000, 2021-12-11:
Tykwondo1, are you trying to update the mobile app or the Core OS for HA? People here are having issue with the Core OS. I tried to update the Core OS using my mobile device, same issue. Also, tried InPrivate and a different browser. Apparently, there is a bug.

NVader2000, 2021-12-11:
Apparently, this is not isolated to Pi. I see other post for other platforms having similar issue. However, some are successful, but breaks other stuff. Do not think that 7.0 is a good build and I will wait for the devs to fix this before installing.

obfri, 2021-12-11:
same here, backups were generated  but looks that update stops after backup. No entries in log showing download activity.

NVader2000, 2021-12-11:
Apparently, this is not isolated to Pi. I see other post for other platforms have similar issue. However, some are successful, but breaks other stuff. Do not think that 7.0 is a good build and I will wait for the devs to fix this before installing.

nunosousa76, 2021-12-11:
Same issue with upgrade except mine is from rc1

Tarabozo, 2021-12-11:
Same issues. VM with HA on Windows NUC. The update to 7.0 is automatically starts a backup first. This backup takes 40-45 minutes, but after that the update it's not done. I see no errors in the logs. I tried twice, same situation. I see 2 new backups, but the update is not done.
Maybe if we have the manual option to skip the backup, the update process will start.

agners, 2021-12-12:
Some browsers (especially Firefox) sometimes don't update the frontend correctly after the intsallation. Try Ctrl+F5 to force a refresh and check the installed version again.

If that does not help, then please share the Supervisor log so we can better analyze the problem.

obfri, 2021-12-12:
tried it again, now it works. Even core update to 12.0 and supervisor to 12.1 works. Only missing http://192.168.xxx.xxx:8123/hassio/dashboard ("Supervisor") at sidebar.

Xerionty, 2021-12-12:
> tried it again, now it works. Even core update to 12.0 and supervisor to 12.1 works. Only missing http://192.168.xxx.xxx:8123/hassio/dashboard ("Supervisor") at sidebar.

https://www.home-assistant.io/blog/2021/12/11/release-202112/#brand-new-configuration-panel

NVader2000, 2021-12-12:
Nope, still not work for me. Are the devs looking at this?

OrionVIII, 2021-12-12:
tried again, still the same (RPi3 with 64GB SD card). More than 40GB free space. The backup is created and then nothing happens, no hint in the logfile.

Sailman83, 2021-12-12:
Same here with Rasperry Pi 3B

aee74, 2021-12-12:
Same issue. Creating backup ...  waiting. Nothing

elcatedratico, 2021-12-12:
Same problem here... tried to upgrade from 6.2 to 7.0 several times. Its just creating a backup (can't unselect this option) an then it stops

Log:
21-12-12 19:17:18 INFO (SyncWorker_0) [supervisor.backups.backup] Backup folder homeassistant done
21-12-12 19:19:18 INFO (MainThread) [supervisor.backups.manager] Creating partial backup with slug 51335fdd completed
21-12-12 19:19:18 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-12-12 19:31:18 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
21-12-12 19:31:18 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
21-12-12 19:31:18 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.PLUGIN
21-12-12 19:31:18 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.CORE
21-12-12 19:31:18 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
21-12-12 19:31:18 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
21-12-12 19:31:18 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
21-12-12 19:31:18 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-12-12 19:31:18 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
21-12-12 19:31:18 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-12-12 19:31:18 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-12-12 19:31:18 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete

MB901, 2021-12-12:
Finally worked for me after several try and a host reboot ! (RPI4)

NVader2000, 2021-12-12:
Yup, got past this finally. Created full backup, wiped MicroSD, and reinstall HA version 7.0, restore from backup. Done!

mdehaas, 2021-12-12:
No issues with the update on the beta channel on HA Blue

agners, 2021-12-12:
@elcatedratico thanks for the logs!

Do you have maybe also a time stamp when the backup started? How long did the backup take?

agners, 2021-12-12:
It seems that the issue is related to the backup happening before the update.

If the update through the UI fails, can affected people try to use the Terminal Add-on and the command `ha os update --version 7.0` instead of the UI?

duncanbarnes, 2021-12-12:
> It seems that the issue is related to the backup happening before the update.
> 
> If the update through the UI fails, can affected people try to use the Terminal Add-on and the command `ha os update --version 7.0` instead of the UI?

This seems to have worked for me!

Tarabozo, 2021-12-12:
> It seems that the issue is related to the backup happening before the update.
> 
> If the update through the UI fails, can affected people try to use the Terminal Add-on and the command `ha os update --version 7.0` instead of the UI?

This worked for me also. Thank you.

johnjoemorgan, 2021-12-13:
> It seems that the issue is related to the backup happening before the update.
> 
> If the update through the UI fails, can affected people try to use the Terminal Add-on and the command `ha os update --version 7.0` instead of the UI?

This failed for me.  I'm on 

core-2021.12.0
supervisor-2021.12.1

Supervisor log shwows an end of file error:

`21-12-13 11:38:52 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/zerotier/armv7 with version 0.12.1
21-12-13 11:38:52 INFO (SyncWorker_7) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/log-viewer/armv7 with version 0.12.1
21-12-13 11:38:52 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-mosquitto with version 6.0.1
21-12-13 11:38:52 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-samba with version 9.5.1
21-12-13 11:38:52 INFO (SyncWorker_9) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/motioneye/armv7 with version 0.16.0
21-12-13 11:38:53 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/ssh/armv7 with version 9.1.1
21-12-13 11:38:53 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/grafana/armv7 with version 7.3.2
21-12-13 11:38:53 INFO (SyncWorker_8) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/influxdb/armv7 with version 4.3.0
21-12-13 11:38:53 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/phpmyadmin/armv7 with version 0.6.0
21-12-13 11:38:53 INFO (SyncWorker_10) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-mariadb with version 2.4.0
21-12-13 11:38:53 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to zigbee2mqtt/zigbee2mqtt-edge-armv7 with version edge
21-12-13 11:38:53 INFO (SyncWorker_11) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-configurator with version 5.3.3
21-12-13 11:38:53 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to sabeechen/hassio-google-drive-backup-armv7 with version 0.105.2
21-12-13 11:38:53 INFO (SyncWorker_7) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-check_config with version 3.9.0
21-12-13 11:38:53 INFO (MainThread) [supervisor.backups.manager] Found 3 backup files
21-12-13 11:38:53 ERROR (MainThread) [supervisor.backups.backup] Can't read backup tarfile /data/backup/33acc5f9.tar: unexpected end of data
21-12-13 11:38:53 INFO (MainThread) [supervisor.discovery] Loaded 1 messages
21-12-13 11:38:53 INFO (MainThread) [supervisor.ingress] Loaded 3 ingress sessions
21-12-13 11:38:53 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
21-12-13 11:38:53 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-12-13 11:38:53 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
21-12-13 11:38:53 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-12-13 11:38:53 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - CoreState.SETUP
21-12-13 11:38:53 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
21-12-13 11:38:53 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-12-13 11:38:53 INFO (MainThread) [__main__] Running Supervisor
21-12-13 11:38:53 INFO (MainThread) [supervisor.os.manager] Rauc: B - marked slot kernel.1 as good
21-12-13 11:38:53 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
21-12-13 11:38:53 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' starting 3 add-ons`

johnjoemorgan, 2021-12-13:
Sorry Code insert's not working correctly for me

Error line is

21-12-13 11:38:53 ERROR (MainThread) [supervisor.backups.backup] Can't read backup tarfile /data/backup/33acc5f9.tar: unexpected end of data

johnjoemorgan, 2021-12-13:
The CLI breaks my install and the restart fails even before I can get a SSL link up.  I have to disconnect power from the XU4 and power up again to get it to restart.

Supervisor and general log show no errors relating to the OS

elcatedratico, 2021-12-13:
Update by unsing the Terminal worked for me! (Had problems with the Terminal under MacOS and iOS, WinOS was fine)

vlad36N, 2021-12-13:
Same here, backups were generated but looks that update stops after backup. No entries in log showing download activity.

mrkaqz, 2021-12-13:
> ha os update --version 7.0

Yes, this is problem for me for awhile. 
As before I always skip backup and go ahead for an update in previous UI.
But after latest update it has no option to skip backup. 

I have tried many times but stuck at backup and gave error.

Finally using Terminal and update command to successfully update to 7.0

obfri, 2021-12-13:
does mandatory update make sense? Not for me - I'm running daily updates by "Home Assistant Google Drive Backup". And these mandatory updates are deleting other full updates ...

aee74, 2021-12-13:
> Yes, this is problem for me for awhile.
>
> As before I always skip backup and go ahead for an update in previous UI.
>
> But after latest update it has no option to skip backup.
>
>
>
> I have tried many [...]

How to update from terminal

agners, 2021-12-13:
> does mandatory update make sense? Not for me - I'm running daily updates by "Home Assistant Google Drive Backup". And these mandatory updates are deleting other full updates ...

Mandatory updates? I guess you meant mandatory backup.

That seems to be a problem in the frontend. It [just got fixed](https://github.com/home-assistant/frontend/pull/10886), so with the next Supervisor update it should no longer make a backup on OS updates.

agners, 2021-12-13:
> How to update from terminal

See https://github.com/home-assistant/operating-system/issues/1662#issuecomment-991976712

johnjoemorgan, 2021-12-13:
@agners They are referring to the the using command line code via a SSL terminal.  Explaining what that is and how to use it is not for here but you can look at official addons for the SSL && Web Terminal addon or the hit-hub page here: https://github.com/hassio-addons/addon-ssh

johnjoemorgan, 2021-12-13:
Manual update using terminal does not work (via the UI or Putty) for the Odroid XU4

phcella, 2021-12-13:
Update via terminal worked for me. I guess the problem is the mandatory backup. 
I have problems when I try to update core with backup. When I disable the backup, it updates normally

dgultig, 2021-12-13:
Update via terminal did the job for me as well.

Still missing the Supervisor icon in the sidebar though.

AlecRust, 2021-12-13:
> Still missing the Supervisor icon in the sidebar though.

https://www.home-assistant.io/blog/2021/12/11/release-202112/#brand-new-configuration-panel

dgultig, 2021-12-13:
My bad @AlecRust, thank you

OrionVIII, 2021-12-13:
Yeah, `ha os update --version 7.0` via ssh worked! Thank you.

nmsousa76, 2021-12-13:
a os update --version 7.0 via ssh worked for me as well

Vaargk, 2021-12-14:
For me the command does not work:
[image]
[image]
[image]

And indeed the url is wrong [url](https://github.com/home-assistant/operating-system/releases/download/7.0/haos_rpi0-w-7.0.raucb). Does someone knows how to give HA the correct update URL?

stigvi, 2021-12-14:
> For me the command does not work: [image] [image] [image]
> 
> And indeed the url is wrong [url](https://github.com/home-assistant/operating-system/releases/download/7.0/haos_rpi0-w-7.0.raucb). Does someone knows how to give HA the correct update URL?

Same happens with a smal system I have. It is running on a Rpi Zero W. It fails with "Error raised from OTA Webserver: 404"

NVader2000, 2021-12-14:
I have completely reflashed the HA OS 7.0 to my MicroSD card and started from scratch. Fortunately have all my stuff documented so I was able to manual rebuild everything without using backups.

However, I am finding a lot of issues/errors even with the supported integrations. HACS, is a nightmare to say the least. Almost, every HACS integration fails to show up as a card and/or entities. If you like your HACS integrations do not upgrade or rebuild to HA OS 7.0. Your mileage may vary.

colotroy1, 2021-12-20:
I'm not able to update OS.  I never have since I initially installed 6.2.  I've tried using the GUI and CLI ha update os and that doesn't work either.   I wonder if this is because I'm using a RP4 and external SSD drive rather than a micro mem card?

I guess my option is to pull the SSD, whipe it and install OS 7.0, but I've tried to upgrade the OS before and it appears to work but on reboot it's always the same.   Could this be specific to using an external SSD in the RP4?

colotroy1, 2021-12-20:
AH-HA!  Ok, No matter what I did I could not get an upgrade to work and when I'd login with SSH and enter ha os info it would tell me 6.2.  I starting thinking about the external SSD and I pulled the micro SSD and rebooted and it came up as 7.0!

So the upgrade worked!  However the core upgrade I'd done earlier, that worked, was now gone and I had to upgrade the core again.  It looks like there's an issue if you have the micro card in the RP4 AND are using an external SSD.  After pulling the micro card the upgrades worked for me.

gcoupe, 2021-12-21:
@colotroy1 - thanks for this tip. You solved a mystery for me as well.

agners, 2021-12-29:
The Frontend/Supervisor do not trigger a backup before update anymore, which should solve the problem for most of you. not 

The OP @NVader2000 solved it by reinstalling.

@colotroy1 are you using Firefox? The Frontend sometimes doesn't update properly with Firefox.

In any case, this wasn't a bug in OS itself and I consider the problem solved.

closed 2021-12-29 by agners