home-assistant/operating-system #1663: Upgrade from 6.6 to 7.0 on Odroid XU4 failed

ClearlyDazed, 2021-12-11:
### Describe the issue you are experiencing

Selected update. Performed backup. HA reports "Connection failed. Connecting". for extended time. Could not SSH into system. Powered off Odroid and rebooted into OS 6.6

### What operating system image do you use?

odroid-xu4 (Hardkernel ODROID-XU4)

### What version of Home Assistant Operating System is installed?

6.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Selected upgrade

### Anything in the Supervisor logs that might be useful for us?

```txt
21-12-11 10:03:44 INFO (MainThread) [supervisor.store] Loading add-ons from store: 66 all - 66 new - 0 remove
21-12-11 10:03:44 INFO (MainThread) [supervisor.addons] Found 8 installed add-ons
21-12-11 10:03:44 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-duckdns with version 1.14.0
21-12-11 10:03:44 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-mosquitto with version 6.0.1
21-12-11 10:03:44 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/tasmoadmin/armv7 with version 0.16.0
21-12-11 10:03:44 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/glances/armv7 with version 0.14.0
21-12-11 10:03:44 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to ikifar/remote-backup-armv7 with version 2021.11.2
21-12-11 10:03:44 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-check_config with version 3.9.0
21-12-11 10:03:44 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-ssh with version 9.2.1
21-12-11 10:03:44 INFO (SyncWorker_7) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-nginx_proxy with version 3.0.2
21-12-11 10:03:45 INFO (MainThread) [supervisor.backups.manager] Found 22 backup files
21-12-11 10:03:45 INFO (MainThread) [supervisor.discovery] Loaded 1 messages
21-12-11 10:03:45 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
21-12-11 10:03:45 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
21-12-11 10:03:45 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-12-11 10:03:45 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
21-12-11 10:03:45 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-12-11 10:03:45 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - CoreState.SETUP
21-12-11 10:03:45 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
21-12-11 10:03:45 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-12-11 10:03:45 INFO (MainThread) [__main__] Running Supervisor
21-12-11 10:03:46 INFO (MainThread) [supervisor.os.manager] Rauc: A - marked slot kernel.0 as good
21-12-11 10:03:46 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
21-12-11 10:03:46 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' starting 2 add-ons
21-12-11 10:03:46 INFO (SyncWorker_7) [supervisor.docker.interface] Cleaning addon_core_mosquitto application
21-12-11 10:03:53 INFO (SyncWorker_7) [supervisor.docker.addon] Starting Docker add-on homeassistant/armv7-addon-mosquitto with version 6.0.1
21-12-11 10:03:53 INFO (SyncWorker_9) [supervisor.docker.interface] Cleaning addon_a0d7b954_sonweb application
21-12-11 10:03:57 INFO (SyncWorker_9) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/tasmoadmin/armv7 with version 0.16.0
21-12-11 10:04:01 INFO (MainThread) [supervisor.services.modules.mqtt] Set core_mosquitto as service provider for mqtt
21-12-11 10:04:02 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' starting 3 add-ons
21-12-11 10:04:02 INFO (SyncWorker_2) [supervisor.docker.interface] Cleaning addon_core_ssh application
21-12-11 10:04:07 INFO (SyncWorker_2) [supervisor.docker.addon] Starting Docker add-on homeassistant/armv7-addon-ssh with version 9.2.1
21-12-11 10:04:07 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning addon_core_duckdns application
21-12-11 10:04:10 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on homeassistant/armv7-addon-duckdns with version 1.14.0
21-12-11 10:04:10 WARNING (SyncWorker_10) [supervisor.docker.addon] Glances running with disabled protected mode!
21-12-11 10:04:10 INFO (SyncWorker_10) [supervisor.docker.interface] Cleaning addon_a0d7b954_glances application
21-12-11 10:04:12 INFO (SyncWorker_10) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/glances/armv7 with version 0.14.0
21-12-11 10:04:17 INFO (MainThread) [supervisor.core] Start Home Assistant Core
21-12-11 10:04:17 INFO (SyncWorker_11) [supervisor.docker.interface] Starting homeassistant
21-12-11 10:04:18 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
21-12-11 10:04:40 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.STARTUP
21-12-11 10:04:40 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-12-11 10:04:47 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-12-11 10:05:59 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
21-12-11 10:05:59 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.APPLICATION' starting 1 add-ons
21-12-11 10:05:59 INFO (SyncWorker_10) [supervisor.docker.interface] Cleaning addon_core_nginx_proxy application
21-12-11 10:06:02 INFO (SyncWorker_10) [supervisor.docker.addon] Starting Docker add-on homeassistant/armv7-addon-nginx_proxy with version 3.0.2
[... 979 lines cut ...]
host_os | Home Assistant OS 6.6
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.1
docker_version | 20.10.8
disk_total | 56.6 GB
disk_used | 10.6 GB
healthy | true
supported | true
board | odroid-xu4
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.2.1), Check Home Assistant configuration (3.9.0), Mosquitto broker (6.0.1), Duck DNS (1.14.0), NGINX Home Assistant SSL proxy (3.0.2), Remote Backup (2021.11.2), Glances (0.14.0), TasmoAdmin (0.16.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 0
views | 7
mode | storage

</details>

agners, 2021-12-12:
Does powering it off and on again helps? Then its probably this bug: #1488.

ClearlyDazed, 2021-12-12:
The OS did not successfully update. A reboot was needed to get back to OS 6.6. Are you saying that the update lost ethernet connection and could not complete?

agners, 2021-12-12:
So in that bug, it was just Ethernet not working after update (which essentially makes the system not accessible on the network). However, after unplugging and replugging the device it booted with networking correctly working, and on the new OS release..

> A reboot was needed to get back to OS 6.6.

Usually a rollback to the old version only happens after 3 unsuccessful boot attempts.

ClearlyDazed, 2021-12-12:
Hi Stefan,

 

I tried numerous times to upgrade the OS, but it still will not happen. On
occasion HA would restart by itself but the OS would still be 6.6. Sometimes
the system would hang and by powering off and on the system would start but
still on OS 6.6.

 

Is there a trick to get it to follow through with the upgrade or is it by
chance and just keep on trying?

 

Thanks

Odroid XU4 failed (Issue #1663)

 

<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com
%2Fhome-assistant%2Foperating-system%2Fissues%2F1663%23issuecomment-99181208
5&data=04%7C01%7C%7C187e248cff404192051008d9bd0844b8%7C84df9e7fe9f640afb435a
aaaaaaaaaaa%7C1%7C0%7C637748665475445133%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4
wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=fsCwTEu
<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com
3HSEIA&data=04%7C01%7C%7C187e248cff404192051008d9bd0844b8%7C84df9e7fe9f640af
b435aaaaaaaaaaaa%7C1%7C0%7C637748665475455130%7CUnknown%7CTWFpbGZsb3d8eyJWIj
oiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=Je
VVtbiScZjUQJNPuWtoo1HHDsy8%2F%2B4sI5Nba9fyOPU%3D&reserved=0> .
<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fapps.apple
.com%2Fapp%2Fapple-store%2Fid1477376905%3Fct%3Dnotification-email%26mt%3D8%2
6pt%3D524675&data=04%7C01%7C%7C187e248cff404192051008d9bd0844b8%7C84df9e7fe9
f640afb435aaaaaaaaaaaa%7C1%7C0%7C637748665475455130%7CUnknown%7CTWFpbGZsb3d8
eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sd
ata=CdfZlpiJlZ9w4fWr9MMMsa8dTiEKxn%2Fu%2F5ImH4RZ0Nk%3D&reserved=0>  or
Android.
<https://github.com/notifications/beacon/AHUOKJTDFULHK6NFM5BLOPTUQPVXBA5CNFS
M5J3HSEIKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOHMO5T5I
.gif>

agners, 2021-12-12:
> Is there a trick to get it to follow through with the upgrade or is it by chance and just keep on trying?

It sounds that something is really wrong with the XU4. I need to check it on my device. Unfortunately, I will only be able to check my instance in ~2 weeks. Meanwhile, please just keep using 6.6. I'll update this post once I have more information.

johnjoemorgan, 2021-12-13:
@agners  I don't think its your XU4.  I have exactly a very similar problem.  Trying to update thru the UI the system backups up, reboots but still at 6.6.  If I run CLI as shown here 

https://github.com/home-assistant/operating-system/issues/1662#issuecomment-991984194

then my XU4 freezes on reboot and I need to do an "air gap" power cycle to get it to reboot to 6.6

As this problem seems to be happening on the Pi and XU4 and maybe others I'm going to wait for a update.

IN the meantime, if anyone wants me to try something to get this working or has suggestions I'm all ears!

johnjoemorgan, 2021-12-13:
Just to confirm ... Manual update using terminal does not work (via the UI or Putty) for the Odroid XU4

agners, 2021-12-13:
Ok, yeah there seems to be a problem with ODROID XU4 on OS 7. I'll check it next week.

axute, 2021-12-19:
any update here? have same issue on odroid hc1 (based on xu4).

agners, 2021-12-28:
I did some tests with ODROID XU4 booting from SD card, and it seems that 6.6 and 7.0 sometimes freezes on boot (but only sometimes, about every 3-5 boot attempts):

```
[    6.423161] printk: console [netcon0] enabled
[    6.426189] netconsole: network logging started
[    6.432619] input: gpio_keys as /devices/platform/gpio_keys/input/input0
[    6.570585] usb 3-1: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice=22.23
[    6.577645] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    6.584507] usb 3-1: Product: USB2.0 Hub
[    6.588425] usb 3-1: Manufacturer: GenesysLogic
[    6.651916] hub 3-1:1.0: USB hub found
[    6.654953] hub 3-1:1.0: 2 ports detected
[    6.905883] usb 4-1: new SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[    6.940139] usb 4-1: New USB device found, idVendor=05e3, idProduct=0616, bcdDevice=22.23
[    6.946989] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    6.954019] usb 4-1: Product: USB3.0 Hub
[    6.957948] usb 4-1: Manufacturer: GenesysLogic
[    6.987948] hub 4-1:1.0: USB hub found
[    6.990791] hub 4-1:1.0: 2 ports detected
[    7.386674] usb 6-1: new SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[    7.424137] usb 6-1: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=30.00
[    7.432267] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[    7.439097] usb 6-1: Product: USB 10/100/1000 LAN
[    7.443275] usb 6-1: Manufacturer: Realtek
[    7.447568] usb 6-1: SerialNumber: 000001000000
[    7.756217] usb 6-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[    7.878127] r8152 6-1:1.0: Direct firmware load for rtl_nic/rtl8153a-3.fw failed with error -2
[    7.888246] r8152 6-1:1.0: unable to load firmware patch rtl_nic/rtl8153a-3.fw (-2)
[    7.909797] r8152 6-1:1.0 eth0: v1.11.11
[   36.067506] random: crng init done
<freeze>
```

Compared to a successful boot:

<details>

```
[    6.458325] printk: console [netcon0] enabled
[    6.461290] netconsole: network logging started
[    6.468852] input: gpio_keys as /devices/platform/gpio_keys/input/input0
[    6.528758] ALSA device list:
[    6.530804]   #0: Odroid-XU4
[    6.531771] usb 3-1: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice=22.23
[    6.541635] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    6.541649] usb 3-1: Product: USB2.0 Hub
[    6.541661] usb 3-1: Manufacturer: GenesysLogic
[    6.590308] hub 3-1:1.0: USB hub found
[    6.593316] hub 3-1:1.0: 2 ports detected
[    6.676488] usb 4-1: new SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[    6.679172] VFS: Mounted root (squashfs filesystem) readonly on device 179:8.
[    6.707096] devtmpfs: mounted
[    6.711685] usb 4-1: New USB device found, idVendor=05e3, idProduct=0616, bcdDevice=22.23
[    6.715111] Freeing unused kernel memory: 1024K
[    6.718808] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    6.718836] usb 4-1: Product: USB3.0 Hub
[    6.718862] usb 4-1: Manufacturer: GenesysLogic
[    6.747469] Run /sbin/init as init process
[    6.753539] hub 4-1:1.0: USB hub found
[    6.757118] hub 4-1:1.0: 2 ports detected
[    7.368866] systemd[1]: System time before build time, advancing clock.
[    7.457005] usb 6-1: new SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[    7.495653] usb 6-1: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=30.00
[    7.503295] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[    7.510577] usb 6-1: Product: USB 10/100/1000 LAN
[    7.514833] usb 6-1: Manufacturer: Realtek
[    7.519200] usb 6-1: SerialNumber: 000001000000
[    7.672019] systemd[1]: systemd 247 running in system mode. (+PAM -AUDIT -SELINUX -IMA +APPARMOR -SMACK -SYSVINIT -UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS -ACL -XZ -LZ4 -ZSTD -SECCOMP +BLKID -ELFUTILS +KMOD -IDN2 -IDN +PCRE2 default-hierarchy=hybrid)
[    7.694373] systemd[1]: Detected architecture arm.

Welcome to Home Assistant OS 6.6!

[    7.799267] usb 6-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[    7.827123] systemd[1]: Set hostname to <homeassistant>.
[    7.835144] systemd[1]: Installed transient /etc/machine-id file.
[    7.902910] r8152 6-1:1.0: Direct firmware load for rtl_nic/rtl8153a-3.fw failed with error -2
[    7.911218] r8152 6-1:1.0: unable to load firmware patch rtl_nic/rtl8153a-3.fw (-2)
[    7.985857] r8152 6-1:1.0 eth0: v1.11.11                                                                                                                                                                                                                                                                                
[    8.509054] random: crng init done
```
</details>

Unfortunately also the latest stable kernel from kernel.org seems to cause the same problems here.

Now this begs the question: Is what I see the problem you are seeing? Does ODROID XU4 boot always reliably on 6.6 for you?

Do you use SD card or eMMC? I use a SD card, but I start to suspect that something is wrong with it. I'll need to do some more testing.

Also, make sure to only have plugged in a SD card or eMMC. Especially, having a old/different HAOS installation can confuse the boot loader/OS and mix up partitions between the two storage devices.

ClearlyDazed, 2021-12-29:
The XU4 uses an eMMC and there is no SD card. I haven't noticed it not booting up from any HA OS.

johnjoemorgan, 2021-12-29:
I use eMMC as well and have never had a boot problem ...  that I know of.  How can I get to the log file you show above?

agners, 2021-12-29:
The logs above are obtained using the [USB-UART Module Kit from Hardkernel](https://www.hardkernel.com/shop/usb-uart-module-kit/).

johnjoemorgan, 2021-12-29:
> The logs above are obtained using the [USB-UART Module Kit from Hardkernel](https://www.hardkernel.com/shop/usb-uart-module-kit/).

Ah ... one of the few pieces of kit I don't have :-(

Is there any other log I can get to you that might be of help?

agners, 2021-12-29:
What would be interesting is the output on HDMI when doing a 7.0 boot attempt.

johnjoemorgan, 2021-12-29:
> What would be interesting is the output on HDMI when doing a 7.0 boot attempt.

I knew that 5m HDMI cable I got 7 years ago would come in useful one day!

Not so sure the info helps.  The XU4 reboots and scrolls through a bunch of info and then halts at the screen shot attached.  Waits there for 30 seconds or so and then reboots again into 6.6

[image]

agners, 2021-12-29:
> I knew that 5m HDMI cable I got 7 years ago would come in useful one day!

:sweat_smile: 

> Waits there for 30 seconds or so and then reboots again into 6.6

Hm, that is different from what I see :worried: 

Do you remember what version of HAOS you initially installed? How did you flash the eMMC?

johnjoemorgan, 2021-12-29:
First install would have been mid-2018 on a Cubie Truc !!

But in I moved to the XU4 in March 2019 and 32GB eMMC using a manual flash as the HA image wasn't working.  See:

https://forum.odroid.com/viewtopic.php?p=249612&sid=fd9538b52899485898aa94c1890ff8ca#p249612

I then upgraded the eMMC to a 64GB in May 2020 using a restore from backup but then the initial HA install for that was a straight forward one as the XU4 image "just works" now.  

So the answer would be V2020.5

agners, 2021-12-29:
It's a bit inconclusive from your screenshot what is happening, but one thing which is problematic is that we don't update the bootloader present on the eMMC boot partition. This is especially problematic with OS 7, which needs a newish U-Boot to handle the new kernel squashfs image format (#1418).

With #1699 this will be done automatically. I'll backport this to release 7 so it will make it into 7.1. Let's see if that does the trick.

agners, 2021-12-29:
@johnjoemorgan can you test 7.1? It is on the beta channel. You can update it using the following Terminal commands:

```
ha su options --channel beta
ha su reload
ha os update
```

johnjoemorgan, 2021-12-30:
@agners @ClearlyDazed 

Thanks Stefan!  Worked a charm!  V7.1 installed. Have a great New Year and thank you for taking time to work this through.  Most appreciated.

closed 2021-12-30 by agners