home-assistant/operating-system #1664: QNAP Integration broken since Home Assistant OS 7.0!

carsten-h, 2021-12-11:
### Describe the issue you are experiencing

Since Home Assistant OS 7.0 the QNAP Integration is no longer working!

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade from Home Assistant OS 6.6 to 7.0
2. QNAP Integration is not working
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
This is interestingly from the Core Log:

During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 439, in send
    resp = conn.urlopen(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 755, in urlopen
    retries = retries.increment(
  File "/usr/local/lib/python3.9/site-packages/urllib3/util/retry.py", line 532, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/usr/local/lib/python3.9/site-packages/urllib3/packages/six.py", line 770, in reraise
    raise value
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 699, in urlopen
    httplib_response = self._make_request(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 447, in _make_request
    self._raise_timeout(err=e, url=url, timeout_value=read_timeout)
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 336, in _raise_timeout
    raise ReadTimeoutError(
urllib3.exceptions.ReadTimeoutError: HTTPConnectionPool(host='192.168.178.101', port=8080): Read timed out. (read timeout=5)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/qnap/sensor.py", line 297, in update
    self.data["system_health"] = self._api.get_system_health()
  File "/usr/local/lib/python3.9/site-packages/qnapstats/qnap_stats.py", line 130, in get_system_health
    resp = self._get_url("management/manaRequest.cgi?subfunc=sysinfo&sysHealth=1")
  File "/usr/local/lib/python3.9/site-packages/qnapstats/qnap_stats.py", line 76, in _get_url
    result = self._execute_get_url(url, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/qnapstats/qnap_stats.py", line 92, in _execute_get_url
    resp = self._session.get(url, timeout=self._timeout, verify=self._verify_ssl)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 555, in get
    return self.request('GET', url, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 529, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPConnectionPool(host='192.168.178.101', port=8080): Read timed out. (read timeout=5)
2021-12-11 21:12:10 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up qnap platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/components/qnap/sensor.py", line 245, in setup_platform
    for drive in config.get(CONF_DRIVES, api.data["smart_drive_health"])
KeyError: 'smart_drive_health'
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing that has to do with the QNAP Integration
```

### System Health information

## System Health

version | core-2021.11.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.63-v8
arch | aarch64
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4800
Installed Version | 1.18.0
Stage | running
Available Repositories | 974
Installed Repositories | 16

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.0
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.1
docker_version | 20.10.9
disk_total | 219.4 GB
disk_used | 20.2 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | AdGuard Home (4.2.0), AppDaemon 4 (0.7.1), Vaultwarden (Bitwarden) (0.14.0), Check Home Assistant configuration (3.9.0), Duck DNS (1.14.0), File editor (5.3.3), HDD Tools (0.51.0), MariaDB (2.4.0), Mosquitto broker (6.0.1), Nginx Proxy Manager (0.11.0), Samba share (9.5.1), TasmoAdmin (0.16.0), WireGuard (0.6.0), Studio Code Server (3.7.0), SSH & Web Terminal (9.1.1), RPi Temp Monitor (0.4), ArgonOne Active Cooling (24.1), Samba Backup (5.0.0), Emulated HUE (dev) (dev-0.0.6), Log Viewer (0.12.1), TasmoBackup (1.04.26)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 10
views | 2
mode | storage

</details>

### Additional information

Configuration for QNAP NAS:
```
sensor:
  - platform: qnap
    host: 192.168.178.101
    port: 8080
    ssl: false
    verify_ssl: false
    username: !secret qnap_username
    password: !secret qnap_password
    monitored_conditions:
      - cpu_temp
      - system_temp
      - drive_temp
      - drive_smart_status
      - status
      - volume_percentage_used
```

agners, 2021-12-12:
This does not seem like an OS issue. Are you sure you only updated the OS? Can you try downgrading it again to see if the error indeed disappears again? Use the following command in the Terminal Add-on to downgrade:

```
ha os update --version 6.6
```

carsten-h, 2021-12-12:
> Are you sure you only updated the OS?

Yes, it worked before and I only updated the OS!
After the update to Core 2021.12.0 there is no change!

I will try an OS downgrade.

carsten-h, 2021-12-12:
After downgrading nothing changed.
Then I restarted the NAS and it was working again. After upgrading to OS 7 it was still working.

But what is still not working is „nut“. The QNAP is the nut master and the Home Assistant Pi is the slave. But the connection is not working anymore. But it was also not working with OS 6.6.

closed 2021-12-12 by carsten-h

agners, 2021-12-13:
Ok, thanks for the update.