home-assistant/operating-system #1665: Upgrade from 6.6 to 7.0 - USB devices fail (zha stick and serial nuvo connection)

tws068, 2021-12-11:
### Describe the issue you are experiencing

Immediately following an upgrade to 7.0 I have lost access to the two USB devices connected to my ODRIOD N2 box. 

The first error appeared when Zigbe2MQTT wouldn't start properly:

`Zigbee2MQTT:error 2021-12-11 16:02:23: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-12-11 16:02:23: Failed to start zigbee
Zigbee2MQTT:error 2021-12-11 16:02:23: Exiting...
Zigbee2MQTT:error 2021-12-11 16:02:23: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)`

The second was failure of a custom integration to recognise a nuvo amplifier connected via USB/Serial:

`2021-12-11 16:13:25 ERROR (MainThread) [custom_components.nuvo_serial] Error connecting to Nuvo controller at /dev/ttyUSB0`

The devices do still seem to exist if I look at the hardware from within HA:

`ttyUSB0
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
ttyUSB1
/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_D-if00-port0`

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

7.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade from 6.6 to 7.0
2. Restart

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System Health information

## System Health

version | core-2021.11.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.83
arch | aarch64
timezone | Europe/London

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4979
Installed Version | 1.18.0
Stage | running
Available Repositories | 918
Installed Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | December 20, 2021, 12:00 AM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-west-2-0.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.0
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.1
docker_version | 20.10.9
disk_total | 28.0 GB
disk_used | 8.4 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | texecom2mqtt (1.1.7), Mosquitto broker (6.0.1), Samba share (9.5.1), Studio Code Server (3.7.0), Log Viewer (0.12.1), Node-RED (10.2.0), Terminal & SSH (9.2.1), Let's Encrypt (4.12.0), Zigbee2mqtt (1.18.1-1), Remote Backup (2021.11.2), Home Assistant Google Drive Backup (0.105.2)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 1
views | 1
mode | storage

</details>

jirimissbach, 2021-12-11:
just sniffing around, I'm having the same problem with Intel NUC since upgrading to version 7. My Google Coral (USB) is not being detected by Frigate. I have another HA running in VMWare and the connected Zigbee stick is working well, haven;t tried the Coral under the VM.

I'm going to roll back but happy to provide any logs if needed (no idea what ).

[2021-12-11 21:52:40] frigate.edgetpu                ERROR   : No EdgeTPU was detected. If you do not have a Coral device yet,

agners, 2021-12-12:
@tws068 this is probably caused by ttyUSB0/1 being enumerated differently than before. Linux does not guarantee the same USB device to have the same ttyUSBX device name between reboots or version updates. Try to use the `/dev/serial/by-id/...` names of the device instead of the `ttyUSBX` names.

@jirimissbach this is unlikely to be the same problem. Can you open a new issue with exact details? Also, can you try downgrading to see if that indeed resolves the problem again? (use `ha os update --version 6.6`).

tws068, 2021-12-12:
Fixed - @agners Thank you. I didn't know that Linux does not guarantee ttyUSBx device names after a version upgrade. This is exactly what happened. I updated the references in both integrations, rebooted and they now work. Much appreciated. I will also try your suggestion to use the full reference to avoid this in future.

Tom

closed 2021-12-12 by tws068