home-assistant/operating-system #1666: After upgrading from 6.6 to 7 - RPI4 Bodel B - OS will not boot with USB hard drive plugged in - Reboot Loop

brenle, 2021-12-11:
### Describe the issue you are experiencing

After upgrading from OS 6.6 to 7, if my USB hard drive (data drive) is plugged in, it goes into a reboot loop.  It will start, then show this:

`starting USB...`
`Bus xhci_pci: Register 5000420 NbrPorts 5`
`Starting the controller`
`USB XHCI 1.00`
`scanning bus xhci_Pci for devices... WARN halted endpoint, queueing URB anyway.`
`Unexpected XHCI event TRB, skipping... (3afefae0 00000004 13000000 03008401)`

If I unplug the USB hard drive, it will begin to boot and I can plug the hard drive in while booting and it starts up fine.  But, rebooting causes the same issue.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 7.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade to OS 7 with USB hard drive plugged in
2. Reboot after upgrade

### Anything in the Supervisor logs that might be useful for us?

```txt
All green INFO events
```

### Anything in the Host logs that might be useful for us?

```txt
Not that I see
```

### System Health information

## System Health

version | core-2021.11.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.63-v8
arch | aarch64
timezone | America/New_York

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | December 27, 2021, 7:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | false
alexa_enabled | false
google_enabled | true
remote_server | us-east-1-2.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | failed to load: timeout

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.0
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.1
docker_version | 20.10.9
disk_total | 234.7 GB
disk_used | 6.8 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | SQLite Web (3.1.0), Node-RED (10.0.1), Visual Studio Code (3.6.2), Z-Wave JS (0.1.45)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 0
views | 2
mode | storage

</details>

### Additional information

[image]

agners, 2021-12-12:
Do you have other USB devices connected besides the USB drive? If so, can you try without them connected?

robertdelpeut, 2021-12-12:
Same here, downgraded tot HA OS 6.6 for now.

brenle, 2021-12-12:
> Do you have other USB devices connected besides the USB drive? If so, can you try without them connected?

Sorry I just got time to try this @agners - Yes, I have a Nortek USB zwave stick. Unplugging it does not change the behavior though.  Only unplugging the USB Hard Drive will allow it to boot.  Also to note, this USB hard drive just has the HA data on it per the recommended guidance in the docs.  I believe it still boots from the SD card.

agners, 2021-12-13:
@brenle @robertdelpeut can you share a bit more details about the hard drive type you are using (drive and enclosure make & model)?

brenle, 2021-12-13:
> @brenle @robertdelpeut can you share a bit more details about the hard drive type you are using (drive and enclosure make & model)?

@agners here is what I am using:
- BENFEI SATA to USB cable: [https://www.amazon.com/gp/product/B07F7WDZGT/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1](https://www.amazon.com/gp/product/B07F7WDZGT/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1)
- ADATA SU800 256GB SATA SSD: [https://www.amazon.com/gp/product/B01K8A2A0E/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1](https://www.amazon.com/gp/product/B01K8A2A0E/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1)

luka6000, 2021-12-13:
It might be connected to UAS driver. Have you tried disabling it like this in the cmdline.txt? https://forums.raspberrypi.com/viewtopic.php?f=28&t=245931

My enclosure is not working with UAS so I used the quirks method and everything is fine on usb-storage driver up till OS6.6. Probably not as fast as it could but it's still much faster that sdcard. Seeing your thread I'm holding with my upgrade as usb storage is a must for me.

timstanley1985, 2021-12-13:
> It might be connected to UAS driver. Have you tried disabling it like this in the cmdline.txt? https://forums.raspberrypi.com/viewtopic.php?f=28&t=245931
>
> My enclosure is not working with UAS so I [...]

Good idea but I think this might be a red herring.  My SSD and enclosure have worked with UAS for a couple of years on raspbian before this issue cropped up with Hass operating system

robertdelpeut, 2021-12-14:
> @brenle @robertdelpeut can you share a bit more details about the hard drive type you are using (drive and enclosure make & model)?

Ofcourse. I used a Kingston SSD Now KC400 (SKC400S37/128G). The USB3.0 to SATA adapter is a generic one, I don't have a specific brand name of type.

agners, 2021-12-15:
The problem @brenie experience is caused by the boot loader U-Boot (the text on the screenshot are things printed by U-Boot).

With 7.0 U-Boot got upgraded from 2021.04 to 2021.10. It seems that this new U-Boot version crashes/causes problems when trying to enumerate this particular USB 3.0 S-ATA controller.

> It might be connected to UAS driver. Have you tried disabling it like this in the cmdline.txt? https://forums.raspberrypi.com/viewtopic.php?f=28&t=245931

Disabling in `cmdline.txt` is not going to help as `cmdline.txt` is only used by the Linux kernel, which is after U-Boot. The problem at hand requires a fix in the U-Boot.

That said, it might be that this new U-Boot has troubles with UAS... [This link](https://everprint.net/index.php?route=product/product&product_id=162123) seems to claim that @brenle 's adapter (BRENFEI) supports UAS. Or it could just be that particular controller.

I'll need to be able to reproduce the problem here. Unfortunately, I can't buy the particular BRENFEI adapter. I'll order a similar model from Startech with UAS support, maybe its even using the same chip (https://www.startech.com/en-us/hdd/usb3s2sat3cb).

@robertdelpeut @timstanley1985 are your screens showing the same issue "Unexpected XHCI event TRB, skipping...."? Can you check if your adapters support UAS? One way to find out is using a Linux OS (e.g. Raspbian from a SD card) and attach the disk. Then use `lsusb | grep uas`. If that returns something, your controller is using UAS.

timstanley1985, 2021-12-15:
My pi is hidden away in a boiler room so I can't easily plug in a monitor to check log messages.  I can confirm though that I am using the startech adapter you linked to.  It definitely supports UAS but I think I had to do a firmware upgrade when I first bought it a few years ago.

maikischa, 2021-12-15:
I have a very simmular problem. After the update the console get stuck in "Waiting for home assistant to start CLI"
If I disconnect and reconnect the USB-HDD during boot. It will start but with version 6.6. again

inverse, 2021-12-24:
Is this still an issue? I'm reluctant to upgrade for exactly this reason. What's the process for downgrading? Is it worth documenting this in the repo?

agners, 2021-12-24:
> Is this still an issue?

Since the issue is still open, I'd assume so. That said, there are 10'000 of Raspberry Pi running OS 7 already, and lots of them with USB SSD. There is not a general problem with "USB SSD". YMMV.

> I'm reluctant to upgrade for exactly this reason. What's the process for downgrading?

The best approach is to take a full backup **and** download it. Then you can reinstall from scratch if necessary.

agners, 2021-12-24:
Meanwhile I received the "Startech USB 3.0 to 2.5" SATA HDD/SSD Cable with UASP". However, I was not able to reproduce the problem. HAOS 7.0 boots and reboots just fine. I test with a Intel SSD drive attached to the adapter.

> But, rebooting causes the same issue.

Could it be a power issue on your end? It is weird that the new version causes the problem, but this could be due to timing or similar (e.g. maybe new U-Boot enumerates faster, at a point where the drive is not quite ready?).

inverse, 2021-12-30:
@agners just to confirm - upgrading to 7.0 worked for me

This SSD: https://www.amazon.de/exec/obidos/ASIN/B07G3KRZBX
This Adaptor: https://www.amazon.de/exec/obidos/ASIN/B06XCV1W97

brenle, 2022-01-03:
> Meanwhile I received the "Startech USB 3.0 to 2.5" SATA HDD/SSD Cable with UASP". However, I was not able to reproduce the problem. HAOS 7.0 boots and reboots just fine. I test with a Intel SSD drive attached to the adapter.

@agners do you think getting the Startech adapter would be the next best step since it works for you? Or am I stuck reloading/restoring from backup?

inverse, 2022-01-03:
@brenle Do you mean the one you have that's problematic? I think @agners received the Startech one already.

timstanley1985, 2022-01-03:
I had to reboot my pi yesterday and I had the same problem with a boot
loop.  I had to unplug the SSD and usb UART adapter (ebusd) for the first
10 secs after powering up to get it to boot.

Is there anything I can try on my end to debug this?

brenle, 2022-01-03:
> @brenle Do you mean the one you have that's problematic? I think @agners received the Startech one already.

@inverse Yeah, I was asking if he thinks it would make a difference if I just got the Startech one since he knows it works with his system.

thiscantbeserious, 2022-01-04:
> With 7.0 U-Boot got upgraded from 2021.04 to 2021.10. It seems that this new U-Boot version crashes/causes problems when trying to enumerate this particular USB 3.0 S-ATA controller.

Wouldn't it be the best to downgrade U-Boot then for the the time being going back to the last known stable - being 2021.04 until that issue has been solved then?

agners, 2022-01-04:
> I was asking if he thinks it would make a difference if I just got the Startech one since he knows it works with his system.

@brenle  It's kinda hard to tell if that would fix it, it might very well be. What would be interesting is what kind of controller the model you have has. Can you share the kernel logs when booting using OS 6.6 (Log Provider "Host" in the Supervisor System tab).

> Wouldn't it be the best to downgrade U-Boot then for the the time being going back to the last known stable - being 2021.04 until that issue has been solved then?

@thiscantbeserious Currently, there is not much downside to stay with 6.6 for affected systems. I'd rather prefer to move forward and try to get things fixed with U-Boot 2021.10.

agners, 2022-01-04:
> I had to reboot my pi yesterday and I had the same problem with a boot
loop.  I had to unplug the SSD and usb UART adapter (ebusd) for the first
10 secs after powering up to get it to boot.
> 
> Is there anything I can try on my end to debug this?

Is that with OS 7.0 only (e.g. does not happen with several reboots on 6.6?) Is it reproducible? It could also be power related: Unplugging and plugging in later does not cause such a big power spike for the power supplies.

brenle, 2022-01-04:
> Can you share the kernel logs when booting using OS 6.6 (Log Provider "Host" in the Supervisor System tab).

Sorry I still have OS 7.0 installed.  I will order a startech to test

brenle, 2022-01-08:
Update: Startech works fine, so looks like it was just some incompatibility with the usb adapter. Thanks everyone and especially @agners for the help

closed 2022-01-08 by brenle

acravo1, 2022-08-09:
Hello.

Not sure if it is the same problem but I was having the same error. I was able to get the screen to install the os but afterwards, the system would continuously reboot. I had the keyboard and mouse connected on yhe usb ports. Disconnecting them fixed the issue.

António