home-assistant/operating-system #1675: It is not possible to upgrade a Rpi Zero W to version 7

stigvi, 2021-12-14:
### Describe the issue you are experiencing

Looking at assets for the 7.0 release, the rpi zero w image is missing. Trying to upgrade a raspberry pi zero w always fails.

(It is also not possible to select rpi zero w when reporting this issue

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Try to upgrade a raspberry pi zero w
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
21-12-14 12:48:34 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/7.0/haos_rpi0-w-7.0.raucb
21-12-14 12:48:35 ERROR (MainThread) [supervisor.os.manager] Error raised from OTA Webserver: 404
```

### Anything in the Host logs that might be useful for us?

```txt
See supervisor log
```

stigvi, 2021-12-14:
I see that Raspberry Pi Zero W is not supported for version 7 and in that case I think this json data is wrong. It will make all HA installs on this board try to install a version that is not available.

https://version.home-assistant.io/stable.json

agners, 2021-12-14:
Yeah indeed, they should no longer be in the verison json's.

agners, 2021-12-29:
Unsupported boards have been removed. The Supervisor will write a log entry when the board can't be found in version file anymore (see https://github.com/home-assistant/supervisor/pull/3366).

closed 2021-12-29 by agners