home-assistant/operating-system #1677: Owner account created in onboarding can't log in

kamenomagic, 2021-12-14:
### Describe the issue you are experiencing

I have a brand new Raspberry Pi 4 (4 GB RAM, 32 GB SD). I followed the directions, flashed the SD card and created an account.

After completing the onboarding, you are already logged in and viewing the dashboard and can do everything.

The problem occurs when logging in from the iOS app (assuming all platforms exhibit this problem) or opening another browser tab and trying to log in there.

In each place it told me "incorrect username/password" even though I was sure it was correct.

To make sure, I reflashed the OS and started over completely, and the same exact issue occurred.

As a workaround, I found out that you can create another user (that works great) and everything works great. I then used the file editor add-on to edit config/.storage/auth to change the owner to the newly created and working account. Then I deleted the user that was created during onboarding, and everything works as expected now.

Expected: The user created during onboarding should be able to log in normally
Actual: The user created during onboarding is initially logged in, but can't log in again from anywhere.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

5.10.63-v8

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Flash SD card
2. Boot Pi
3. Fill out the onboarding information including creating the owner user
4. Try to login from the app/another tab or log out first and try logging in again

### Anything in the Supervisor logs that might be useful for us?

```txt
I don't have the original logs since I was just trying to get it working, but it did say "failed log in attempt" even though the username and password were correct (the login screen reported incorrect username and password)
```

### Anything in the Host logs that might be useful for us?

```txt
I don't have the original logs since I was just trying to get it working, but it did say "failed log in attempt" even though the username and password were correct (the login screen reported incorrect username and password)
```

kamenomagic, 2021-12-14:
FYI, I did get things working and am not stuck (see workaround in description above), however I confirmed this bug twice before getting things working and thought I would log it so that people were aware (and maybe I'll even be able to try to fix it myself and submit a PR....if I am able to find time)

agners, 2021-12-15:
The OS itself is not involved in the onboarding. It is Home Assistant Core itself which presents the interface to create an initial user.

closed 2021-12-15 by github-actions[bot]