home-assistant/operating-system #1679: New setup - no sound - Vbox on windows 10

vandad51, 2021-12-16:
### Describe the issue you are experiencing

After selecting Intel HD Audio as my audio source in VBOX, I am not getting any sound output to any of my devices upon triggering an example TTS sample. My logs do show something triggered it, but no result. 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.0

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

Using my  browser_mod media player entity, sent a TTS word.

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing relevant
```

### Anything in the Host logs that might be useful for us?

```txt
I think this came up when I trigged it just now?

[66671.118365] hrtimer: interrupt took 1814988 ns
```

### System Health information

## System Health

version | core-2021.12.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.83
arch | x86_64
timezone | America/Winnipeg

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4939
Installed Version | 1.18.0
Stage | running
Available Repositories | 974
Installed Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.0
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 30.8 GB
disk_used | 7.0 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Node-RED (10.2.0), Samba share (9.5.1), Studio Code Server (3.7.0), AdGuard Home (4.2.0), Mosquitto broker (6.0.1), Terminal & SSH (9.2.1), Glances (0.14.0), Duck DNS (1.14.0), File editor (5.3.3), Log Viewer (0.12.1), Z-Wave JS (0.1.48), Zigbee2mqtt (1.22.1-1), MariaDB (2.4.0), Nginx Proxy Manager (0.11.0), NGINX Home Assistant SSL proxy (3.0.2)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 5
views | 3
mode | storage

</details>

### Additional information

Here's what hasn't been bumped off my "Audio" log at this point

D: [pulseaudio] backend-native.c: Bluetooth Headset Backend API support using the native backend
D: [pulseaudio] backend-native.c: Registering Profile /Profile/HSPAGProfile 00001112-0000-1000-8000-00805f9b34fb
D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
D: [pulseaudio] bluez5-util.c: Unknown interface org.bluez.AgentManager1 found, skipping
D: [pulseaudio] bluez5-util.c: Unknown interface org.bluez.ProfileManager1 found, skipping
I: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
D: [pulseaudio] bluez5-util.c: oFono is running: no
D: [pulseaudio] backend-native.c: Registering Profile /Profile/HSPHSProfile 00001108-0000-1000-8000-00805f9b34fb
I: [pulseaudio] client.c: Created 0 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Client authenticated anonymously.
D: [pulseaudio] protocol-native.c: Protocol version: remote 34, local 34
D: [pulseaudio] protocol-native.c: SHM possible: no
D: [pulseaudio] protocol-native.c: Negotiated SHM: no
D: [pulseaudio] protocol-native.c: Disabling srbchannel, reason: No SHM support
E: [alsa-source-Generic Analog] alsa-source.c: ALSA woke us up to read new data from the device, but there was actually nothing to read.
E: [alsa-source-Generic Analog] alsa-source.c: Most likely this is a bug in the ALSA driver 'snd_hda_intel'. Please report this issue to the ALSA developers.
E: [alsa-source-Generic Analog] alsa-source.c: We were woken up with POLLIN set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
I: [pulseaudio] client.c: Freed 0 "supervisor"
I: [pulseaudio] protocol-native.c: Connection died.
E: [alsa-sink-Generic Analog] alsa-sink.c: ALSA woke us up to write new data to the device, but there was actually nothing to write.
E: [alsa-sink-Generic Analog] alsa-sink.c: Most likely this is a bug in the ALSA driver 'snd_hda_intel'. Please report this issue to the ALSA developers.
E: [alsa-sink-Generic Analog] alsa-sink.c: We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
I: [alsa-source-Generic Analog] ratelimit.c: 315 events suppressed
I: [alsa-source-Generic Analog] alsa-source.c: Overrun!
I: [alsa-sink-Generic Analog] alsa-sink.c: Underrun!
I: [pulseaudio] client.c: Created 1 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Client authenticated anonymously.
D: [pulseaudio] protocol-native.c: Protocol version: remote 34, local 34
D: [pulseaudio] protocol-native.c: SHM possible: no
D: [pulseaudio] protocol-native.c: Negotiated SHM: no
D: [pulseaudio] protocol-native.c: Disabling srbchannel, reason: No SHM support
I: [pulseaudio] client.c: Freed 1 "supervisor"
I: [pulseaudio] protocol-native.c: Connection died.
I: [pulseaudio] client.c: Created 2 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Client authenticated anonymously.
D: [pulseaudio] protocol-native.c: Protocol version: remote 34, local 34
D: [pulseaudio] protocol-native.c: SHM possible: no
D: [pulseaudio] protocol-native.c: Negotiated SHM: no
D: [pulseaudio] protocol-native.c: Disabling srbchannel, reason: No SHM support
I: [pulseaudio] client.c: Freed 2 "supervisor"
I: [pulseaudio] protocol-native.c: Connection died.
I: [pulseaudio] client.c: Created 3 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Client authenticated anonymously.
D: [pulseaudio] protocol-native.c: Protocol version: remote 34, local 34
D: [pulseaudio] protocol-native.c: SHM possible: no
D: [pulseaudio] protocol-native.c: Negotiated SHM: no
D: [pulseaudio] protocol-native.c: Disabling srbchannel, reason: No SHM support
I: [pulseaudio] client.c: Freed 3 "supervisor"
I: [pulseaudio] protocol-native.c: Connection died.
I: [pulseaudio] client.c: Created 4 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Client authenticated anonymously.
D: [pulseaudio] protocol-native.c: Protocol version: remote 34, local 34
D: [pulseaudio] protocol-native.c: SHM possible: no
D: [pulseaudio] protocol-native.c: Negotiated SHM: no
D: [pulseaudio] protocol-native.c: Disabling srbchannel, reason: No SHM support
I: [pulseaudio] client.c: Freed 4 "supervisor"
I: [pulseaudio] protocol-native.c: Connection died.
I: [alsa-source-Generic Analog] alsa-source.c: Overrun!
I: [pulseaudio] client.c: Created 5 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Client authenticated anonymously.
D: [pulseaudio] protocol-native.c: Protocol version: remote 34, local 34
D: [pulseaudio] protocol-native.c: SHM possible: no
D: [pulseaudio] protocol-native.c: Negotiated SHM: no
D: [pulseaudio] protocol-native.c: Disabling srbchannel, reason: No SHM support
I: [pulseaudio] client.c: Freed 5 "supervisor"
I: [pulseaudio] protocol-native.c: Connection died.
I: [pulseaudio] client.c: Created 6 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Client authenticated anonymously.
D: [pulseaudio] protocol-native.c: Protocol version: remote 34, local 34
D: [pulseaudio] protocol-native.c: SHM possible: no
D: [pulseaudio] protocol-native.c: Negotiated SHM: no
D: [pulseaudio] protocol-native.c: Disabling srbchannel, reason: No SHM support
I: [pulseaudio] client.c: Freed 6 "supervisor"
I: [pulseaudio] protocol-native.c: Connection died.
I: [alsa-source-Generic Analog] alsa-source.c: Overrun!
I: [alsa-sink-Generic Analog] alsa-sink.c: Underrun!
I: [pulseaudio] client.c: Created 7 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Client authenticated anonymously.
D: [pulseaudio] protocol-native.c: Protocol version: remote 34, local 34
D: [pulseaudio] protocol-native.c: SHM possible: no
D: [pulseaudio] protocol-native.c: Negotiated SHM: no
D: [pulseaudio] protocol-native.c: Disabling srbchannel, reason: No SHM support
I: [pulseaudio] client.c: Freed 7 "supervisor"
I: [pulseaudio] protocol-native.c: Connection died.

agners, 2021-12-16:
What Operating System are you running VirtualBox on? What "Host Audio Driver" are you using? Do you have virtualization extensions enabled (e.g.  Virtualization Technlogy setting in BIOS)?

vandad51, 2021-12-17:
- Windows 10 Pro 64 bit on a Surface Pro 1
- Windows DirectSound
- I'm not sure if the 1st Gen surface has the virtualization feature. When I open task manager, it does say in the performance tab "Virtualization: Enabled

agners, 2021-12-17:
Ok, that sounds like VT is enabled. Not sure, It does seem to work here using VirtualBox 6.1.30 with HAOS 7,0 and Intel HD Audio.

Maybe the VirtualBox folks have some ideas. The VirtualBox wiki has a [Investigating audio problems](https://www.virtualbox.org/wiki/AudioDebug) how-to. Otherwise maybe their forums can help.

vandad51, 2021-12-17:
Ok thanks, I'll try checking those logs and see if the error starts at Vbox. I'll report back what I find.

vandad51, 2021-12-18:
I was at Vbox 6.1.28, so i updated and it didn't make a difference. I tried enabling nested paging, but it didn't help and I can't enable Nested VT-X/AMD-V cause of Intel.  [This forum,](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1873641) and [this one,](https://askubuntu.com/questions/1230363/internal-speaker-audio-is-not-working-in-ubuntu-20-04-lts-after-upgrading-from) show very similar logs of what I'm experience, but not sure how to fire those commands in HA that they were talking about.

closed 2022-03-26 by github-actions[bot]