home-assistant/operating-system #1680: Lost Zigbee Network with Conbee II/NUC Setup When Upgrading from HA OS 6.5 to 6.6 or 7.0

jle1n0, 2021-12-17:
### Describe the issue you are experiencing

The whole Zigbee network behind Conbee II is lost when I upgraded Home Assistant OS 6.5 to 6.6. Downgrading back to 6.5 made the system work again. Later when I upgraded directly to 7.0 the same issue appeared again:
 
- restarting 7.0 version three times didn't change the situation
- downgrading back to 6.5 made the system work again

Conbee II seems not to appear in "Configuration -> Add-ons, Backups & Supervisor -> System -> Host -> ... -> Hardware" section in 7.0 whereas in 6.5 it is visible.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. upgrade from 6.5 to 7.0 (or 6.6)
2. check in the UI if Conbee II stick is recognized correctly or if the network works at all

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing special found.
```

### Anything in the Host logs that might be useful for us?

```txt
In 7.0

[   11.820500] usb 1-1: unable to read config index 0 descriptor/all
[   11.820523] usb 1-1: can't read configurations, error -110
[   11.820627] usb usb1-port1: attempt power cycle
[   12.200429] usb 1-1: new full-speed USB device number 4 using xhci_hcd
[   12.214417] usb 1-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[   12.214427] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   12.214432] usb 1-1: Product: ConBee II
[   12.214436] usb 1-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[   12.214440] usb 1-1: SerialNumber: DE2448122

In 6.5

[    0.749698] usb 1-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[    0.749705] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    0.749710] usb 1-1: Product: ConBee II 
[    0.749713] usb 1-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[    0.749717] usb 1-1: SerialNumber: DE2448122
```

### System Health information

## System Health

version | core-2021.12.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.83
arch | x86_64
timezone | Europe/Helsinki

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | failed to load: timeout

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.0
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 228.5 GB
disk_used | 7.8 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Check Home Assistant configuration (3.9.0), Samba share (9.5.1), File editor (5.3.3), Home Assistant Google Drive Backup (0.105.2), HA Command Center (0.6.0), Git pull (7.13.1), Grafana (7.4.0), InfluxDB (4.3.0), AppDaemon 4 (0.7.1), SSH & Web Terminal (10.0.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 5
-- | --
resources | 4
views | 12
mode | storage

</details>

agners, 2021-12-27:
This is most likely a Linux kernel issue. Between OS 6.5 and 6.6 we updated the Linux kernel from 5.10.70 to 5.10.75. 7.0 then comes with 5.10.78. It seems that this introduced the problem. 

In the latest development builds Can you try upgrading to the latest 5.10.88 kernel, maybe it has been recognized and fixed. Can you test the latest development build?

```
ha su options --channel dev
ha su reload
ha  os update
```

Then downgrade again using

```
ha su options --channel stable
ha su reload
ha  os update --version 6.5
```

This should be a safe operation, but a backup (along with downloading it) is still recommended :sweat_smile:

jle1n0, 2021-12-28:
Looks good with the new kernel:
```
[    0.000000] Linux version 5.10.88 (builder@edb79bd2cd8c) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g907739ed) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Mon Dec 27 15:05:18 UTC 2021
```
```
[    0.724684] usb 1-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[    0.724695] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    0.724700] usb 1-1: Product: ConBee II
[    0.724703] usb 1-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[    0.724707] usb 1-1: SerialNumber: DE2448122
```
And Zigbee network seems to work ok. I did three restarts and the system ran stable some hour. Roll back to 6.5 also went well.

I guess the conclusion for me is just to stick with the 6.5 version until the (>) 5.10.88 kernel appears in some future release?

agners, 2021-12-29:
Thanks for verifying. Home Assistant OS 7.1 is on the beta channel. You can update it using the following Terminal commands:

```
ha su options --channel beta
ha su reload
ha os update
```

closed 2021-12-29 by agners