home-assistant/operating-system #1685: Hardware logs full off ext4 filesystem being remounted at /run/systemd/unit-root/etc/

gieljnssns, 2021-12-20:
### Describe the issue you are experiencing

My Host logs are full off lines like this
```
[144507.543172] ext4 filesystem being remounted at /run/systemd/unit-root/etc/udev/rules.d supports timestamps until 2038 (0x7fffffff)
```

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 7.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Start HA
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
21-12-20 07:38:29 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-12-20 07:39:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/korylprince/hassio-caddy repository
21-12-20 07:39:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/thomasmauerer/hassio-addons repository
21-12-20 07:39:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
21-12-20 07:39:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/alexbelgium/hassio-addons repository
21-12-20 07:39:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sanderdw/hassio-addons repository
21-12-20 07:39:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/arthurdent75/SimpleScheduler repository
21-12-20 07:39:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/hassio repository
21-12-20 07:39:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
21-12-20 07:39:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/bleakvin/hass-guacamole repository
21-12-20 07:39:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/blakeblackshear/frigate-hass-addons repository
21-12-20 07:39:00 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/raph2i/hassio-addons repository
21-12-20 07:39:02 WARNING (MainThread) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate NVR (Full Access)
21-12-20 07:39:02 WARNING (MainThread) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate NVR (Full Access) Beta (0.10.0)
21-12-20 07:39:03 WARNING (MainThread) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate NVR (Full Access)
21-12-20 07:39:03 WARNING (MainThread) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate NVR (Full Access) Beta (0.10.0)
21-12-20 07:39:03 INFO (MainThread) [supervisor.store] Loading add-ons from store: 131 all - 0 new - 0 remove
21-12-20 07:39:03 INFO (MainThread) [supervisor.store] Loading add-ons from store: 131 all - 0 new - 0 remove
21-12-20 07:39:45 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'hass'
21-12-20 07:39:45 INFO (MainThread) [supervisor.auth] Successful login for 'hass'
21-12-20 07:45:00 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'hass'
21-12-20 07:45:00 INFO (MainThread) [supervisor.auth] Successful login for 'hass'
21-12-20 07:50:45 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'hass'
21-12-20 07:50:45 INFO (MainThread) [supervisor.auth] Successful login for 'hass'
21-12-20 07:56:00 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'hass'
21-12-20 07:56:00 INFO (MainThread) [supervisor.auth] Successful login for 'hass'
21-12-20 08:01:45 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'hass'
21-12-20 08:01:45 INFO (MainThread) [supervisor.auth] Successful login for 'hass'
21-12-20 08:07:00 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'hass'
21-12-20 08:07:00 INFO (MainThread) [supervisor.auth] Successful login for 'hass'
21-12-20 08:08:29 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-12-20 08:12:45 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'hass'
21-12-20 08:12:45 INFO (MainThread) [supervisor.auth] Successful login for 'hass'
21-12-20 08:18:00 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'hass'
21-12-20 08:18:00 INFO (MainThread) [supervisor.auth] Successful login for 'hass'
21-12-20 08:23:45 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'hass'
21-12-20 08:23:45 INFO (MainThread) [supervisor.auth] Successful login for 'hass'
21-12-20 08:29:00 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'hass'
21-12-20 08:29:00 INFO (MainThread) [supervisor.auth] Successful login for 'hass'
21-12-20 08:34:45 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'hass'
21-12-20 08:34:45 INFO (MainThread) [supervisor.auth] Successful login for 'hass'
[... 255 lines cut ...]
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), AdGuard Home (4.2.0), AppDaemon 4 (0.7.1), Caddy (1.6), Check Home Assistant configuration (3.9.0), Duck DNS (1.14.0), ESPHome (2021.12.1), Mosquitto broker (6.0.1), Studio Code Server (3.7.0), WireGuard (0.6.0), MariaDB (2.4.0), Samba Backup (5.0.0), InfluxDB (4.3.0), Grafana (7.4.0), Frigate NVR (2.3), SSH & Web Terminal (10.0.0), Glances (0.14.0), phpMyAdmin (0.6.0), Mealie (0.5.4), Arpspoof (1.0.0), Webtop Alpine (1.1), Guacamole (1.3.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 17
views | 19
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

agners, 2021-12-21:
Newer Linux kernels warn when a file system has only been created with 32-bit timestamp support. You can safely ignore the error, unless you plan to run your instance beyond 2038 :smile: 

If you plan to use this installation beyond 2038 (or you want to get rid of the message just because), you have to create a new installation using HAOS 7. You can use the backup and restore feature to do that. Unfortunately, I don't have another solution currently.

gieljnssns, 2021-12-22:
Thanks for your answer.

closed 2021-12-22 by gieljnssns

Megachip, 2022-01-27:
> Newer Linux kernels warn when a file system has only been created with 32-bit timestamp support. You can safely ignore the error, unless you plan to run your instance beyond 2038 😄
> 
> If you plan to use this installation beyond 2038 (or you want to get rid of the message just because), you have to create a new installation using HAOS 7. You can use the backup and restore feature to do that. Unfortunately, I don't have another solution currently.

These warnings also appear on HAOS 7.x

agners, 2022-01-27:
Yes, and they will appear in 8.x etc. etc. The file system has been formatted that way on first installation, I can't magically reformat your installation.

Please read the comment carefully. You can safely ignore it. Otherwise **you** will have to take actions if you want to get rid of it.